aceinnolab/Inkycal #1: Improvement Ideas

aceisace, 2018-09-08:
Here you have an option to chat about improvement ideas regarding the E-Paper Calendar.

As usual, nothing and nobody is perfect, but we can strive to get closer towards that. Please post your comments in a friendly way and remember to be nice. Thanks for your understanding.

All improvement ideas are welcome.

surak, 2018-09-27:
I'm making one. I had it ready, but updating raspbian messes up with the drivers.

That being said, I would like to be able to choose the orientation (I saw your comment on looking better this way). And instead of that huge month (who needs the biggest number on display to be the month?), how about the events of the day? Or the day/temperature? 

Just throwing some ideas. I will probably clone the thing when I have time, and request you some pulls.

aceisace, 2018-09-28:
@surak 
Hi Alexander. Thank you for your comment. I'm glad you liked the project and that you are giving improvement ideas :)

As for the RIBBA 21675 frame, I saw the picture you took from your other comment:
[image]
It does make a nice fit with the E-Paper Display. Are you planning to hang this on a wall by any chance?

Since I intended to place my Calendar on the table, I designed a case and a base to be printed with a 3D-printer (you can see the a few more pictures in the 'Gallery' folder). 

The inspiration for creating the E-Paper Calendar actually came from this picture:
[image]
As you can see, both versions look similar. The initial plan was actually to create something as close to this picture as possible. This is also the reason I chose the vertical orientation.

You have a good point regarding the orientation. I'd liked to create an option where you could just 'flip' the orientation but the main issue lies with the .bmp files. The E-Paper Calendar software is actually quite picky with the .bmp files (16-bit, no alpha-channel, only 3 colours). If I break down the investet time into this project, I'd say that 70% went to the .bmp files. In short, it will take a significant amount of time to create just the icons. 

As an alternative to the current 'Calendar' mode, I've been thinking of a 'Dashboard' mode which displays everything in much more details. Here is a current 'alpha' draft of the Dashboard:
[image]
Top to bottom; 
1) Weather details from openweathermap (including sunrise time, sunset time, rain chances, humidity, max and min. temperatur, wind speed, moon phase and weather icon)
2) Upcoming events of this month. This will show upcoming events and display a description of the event too.
3) Not too sure regarding this section. Currently, the idea was to use it for notes.

The best way to propose an idea for this project would be to post a 'draft' page (like above) on how and what you'd like your Calendar to display. That way, we can discuss the features and finalise the draft. After that, I can go on to creating the .bmp files and writing or adapting the code.

That being said, you are most welcome to clone and request pulls and I'm looking forward to them :).

aceisace, 2018-10-02:
## Important Notice:
A kind note to anyone using the software, please update to the latest version (early October 2018) as it contains a critical update which prevents ghosting and damage to the display from over-voltage in the long-term. It also contains full support for the 2-Colour E-Paper version. Refer to the wiki for updating instructions.

aceisace, 2018-10-03:
@surak 
I've finished a functional image converter in the [developers-only folder](https://github.com/aceisace/E-Paper-Calendar-with-iCal-sync-and-live-weather/blob/master/For-developers-only/2-Colour-converter.py). 

Now you can input png, jpeg and bmp files and the image converter will prepare them for the E-Paper display. It will reduce all colours of all images to simply pure black and pure white pixels. What's more, you can convert many files at once, without having to change the script.

It will hopefully help cutting down development time greatly for creating new icons. As for your idea of using a different orientation, a simple modification to the file will allow you to create icons for the horizontal mode. For horizontal orientation, remove this part `.rotate(-90, expand=True))` from lines 73, 97 and 102.

If something shouldn't work as expected, kindly let me know.

surak, 2018-10-04:
Thanks! I’ve been busy with other stuff, I need to get back to this :-)

aceisace, 2018-10-04:
@surak 
I've added the Gravit Designer file I used for creating the icons in the 'developers-only' folder. It's a good starting point for editing the current layout and creating icons.

aceisace, 2018-10-10:
@surak 
Are you using the latest version of the E-Paper Calendar software? If so, I wanted to ask you if you could post a gif or short clip of the display during updating. 
The new version (v1.2) uses a different code which should be able to update the screen faster for the 2-Colour display but I have no way of knowing if it actually works as I only have the 3-Colour display. Thanks in advance.

> Thanks! I’ve been busy with other stuff, I need to get back to this  :-)

Sure, I'm looking forward to it :)

surak, 2018-10-10:
Will do - I finally got back my notebook!

surak, 2018-10-10:
Hi. Indeed the update is WAY faster for screen refresh. But the rest of the code is still slow before that. 

Link to the video: https://youtu.be/HkM56hZw7kU

aceisace, 2018-10-11:
@surak 
Thanks for the update and the video, Alexander. I'm glad to see that the update for the 2-Colour version works as expected. During updating, there are no signs of ghosting and the image remains clear.

Indeed, the code does take a while to get everything ready. The communication and parsing of openweathermap api alone takes 20 seconds. It also takes a few seconds converting the finished picture to data the E-Paper can understand. It's difficult improving the loading time unless you have a more powerful CPU.

surak, 2018-10-11:
> @surak
> Thanks for the update and the video, Alexander. I'm glad to see that the update for the 2-Colour version works as expected. During updating, there are no signs of ghosting and the image remains clear.

Yes, that's great news. I am figuring out how to change the months thing into a more useful thing - like the weather for the day, for example, or a big clock. I still don't get how you map coordinates though - when I come back home, I'll show you what I mean.
 
> Indeed, the code does take a while to get everything ready. The communication and parsing of openweathermap api alone takes 20 seconds. It also takes a few seconds converting the finished picture to data the E-Paper can understand. 

Both things can be done asynchronously. One does not need to wait for the OWM reply to assemble the other parts of the bitmap, for example. I will play around with it and show you.

aceisace, 2018-10-11:
@surak 
Thanks, I'll soon add instructions in the wiki on how to set the correct coordinates for a bmp file. Once that's done, I'll notify you. 

> Both things can be done asynchronously. One does not need to wait for the OWM reply to assemble the other parts of the bitmap, for example. I will play around with it and show you.

I'll be looking forward to the multitasking part in python.

aceisace, 2018-10-14:
@surak 
I've updated the instructions in the [Wiki](https://github.com/aceisace/E-Paper-Calendar-with-iCal-sync-and-live-weather/wiki/Display-Layout-and-Bitmaps) with instructions on placing the images at the correct positions.  If there is something that's not clear, do let me know.

surak, 2018-10-24:
It's perfectly clear, thanks (I wonder if we have put the display upside down wouldn't it make it easier - the placement of images from the vertical standpoint I mean... but I digress).

The date at the top left is completely redundant - I just changed it for something useful, like... the hour :-)

I started making better use of the screen's real estate (let's just say that *EVERYBODY* who looked at the "Month" symbol though it was the hour, or the day. No one figure it was the month - HUGE usability clue...)

My plan now is to make the spacing between the calendar weeks smaller, and put the week's weather forecast somewhere (probably where the schedule is right now, and move the schedule to below the calendar):

[image]

aceisace, 2018-10-25:
@surak 
I'm glad to hear the instructions on the positions are clear in the wiki. You've done a neat job on utilising the empty space. Looks like you're getting the hang of using the E-Paper display. Btw, your E-Paper Calendar case looks good :)

You added a function to display the time. How often do you refresh the screen?

> My plan now is to make the spacing between the calendar weeks smaller, and put the week's weather forecast somewhere (probably where the schedule is right now, and move the schedule to below the calendar):

From what I understand, you're looking for a way to create space for more functions. Creating or Resizing icons is not difficult thanks to the converter but re-allocating different positions for the new icons will take a lot of time (calculating the positions mainly). You could probably remove the 'lines' or seperator and change the font size to make more space.

In the coming versions, I'll make the icon showing the name and number of the month smaller and add more weather details and if possible, an event list.

surak, 2018-10-26:
I refresh the time once a minute. But I only check for the weather once an hour. I'll merge your recent changes into my fork and commit those changes so you can have a look.

aceisace, 2018-10-27:
It's probably not a good idea to update the screen too often. I've attached a part of the E-Paper display just below:
[image]

As you can see, the panel's life is limited by either 1000000 refreshs or 5 years, whatever is reached first.
It comes down to : 1000000 / (365*5) = ~547 updates per day to reach the limit.
By updating the E-Paper 1440 times a day (minutes in each day), you surpass that limit and will decrease the lifetime of the display. 

I'll take a look at your fork once you have merged the changes. The features you are planning to implement are interesting and I'm looking forward to the changes in your fork.

surak, 2018-11-05:
Hi,

just to let you know, I found those open source icons: https://akveo.github.io/eva-icons/#/

aceisace, 2018-11-05:
Hi,
thank you for the link to the open-source icons. If I need any more icons, I'll look there as well. 
On the other hand, if you think one of the current icons could be replaced by a better one, I'd sure like to hear the suggestion.

surak, 2018-11-05:
Hi again!

I did put some thought on the whole thing recently from the standpoint of a designer, and tested with "users" (i.e. all my friends and visitors who saw the thing).

Turns out that:

1. the big month, as I suspected, is not just useless, it's misleading. Everyone knows in which  month we are, so they assume (wrongly) that the big number is either temperature or the current day.
2. what is actually better in a digital calendar than in a paper one? In our case, the weather and appointments. I'll extend this one:
    2.1 Weather: it's just *too* small. Every time someone looks at it, one has to go really close to see the actual data.
            2.1.1 Weather: current temperature/humidity is nice, but one wants to know what to wear for the day. Will it be cold? Will it rain? This is useful info.
    2.2: We spoke about the appointments before. As I live in Germany, for me the useful info there is the garbage calendar. What garbage container should I take outside for the next morning? That is a simple .ics file. Same for meetings, whatever.
3. The calendar itself is nice, but less useful than today's and tomorrow's info as I said above. It can occupy a smaller real estate of the screen (I have a full calendar at a 2.5" eink display)
4. The date in extended form is redundant. If you have a "Tuseday, May, 25th, 2018", you don't need a calendar at all. Or vice-versa.
5. The humidity runs off the screen when it's 100% (too wide). The temperature runs off the screen when it's minus double digits.

So, as soon as I come back home and have time, I will play around with those ideas. 

I'm not asking you to change your project, mind you. Please don't be offended by this.

aceisace, 2018-11-06:
Hi Alex, first of all thank you for taking the time to get more opinions from you colleges in regards to the current layout.

As you did mention quite a few things, I'll reply in the same order:
1) For the 'month' icon, I can make all of them smaller and omit the number of the month. This can be fixed relatively.

2) You mentioned the weather being __too__ small. Does this apply to the temperature and humidity or the icons or both? The section above the current seperator (red/black lines) can be modified easily. This also means that a small 'description' of the weather will be able to fit in there as well as more current-weather info (rain chances, wind speed, min. and max. temperature, sunset, sunrise etc.).

* In regards to the name of the events (like 'yellow garbage') etc. ,it'll probably be difficult to squeeze them on the display. Even if that is possible, the font will be too small. For example, someone might have 7 events in a day (highly unlikely, but still possible). Having to display all of these will be difficult in the limited space. In that case, it's not a very good idea to display the event names. 

* Also, due to privacy reasons, I intended to __not__ display any event names. If someone is using the Calendar in an office and everyone else can see his/her events, that's not so good either. Well, garbage is not so private but I'm sure there are certain others.

My suggestion is the following: For recurring events (yellow-, brown-, black-, paper- garbage), how about a custom icon for those event/events? For example, check if events today contain 'yellow garbage', if yes, then display a certain icon at either the day itself or another a given position.
[image]

* Alternatively, you could utilise the 'freed' space from the removed month number and display the icon there:

[image]
using a function like: If 'yellow garbage' in today's events, display yellow-garbage icon at pos1.
Since a picture/icons says more than a thousand words, I'd highly recommend this option as it saves space, makes it easier to understand and can be implemented/modified easier too.

3) If possible, I'd rather not change the 'days of the month' icons (1-30). It's really going to be a lot of work having to create icons, testing the layout, replacing all positions (about 42 possible locations) etc. Unless this is a __real__ problem, I'd like to avoid that. On the other hand, if you feel a dire need to change the Calendar layout, please do so in your own fork. 
Also, could you please elaborate 'but less useful than today's and tomorrow's info'? I think I did not understand this well enough to know what you meant.

4) The extended form of the date will be removed in the next version. That's relatively easy to do.

5) Could you please post a picture showing the digit's of the temperature/humidity not fitting on the display? I __did__ consider double minus digits as well as 100% humidity when creating the space for the temperature and humidity but it's possible I might have forgotten something. 
Just for reference, this is how it's displayed on my Calendar:
[image]

Feel free to test out your ideas once you're back and find the time. Also, I'm not offended at all by your comment. In fact, receiving an honest opinion (be it a suggestion or critics), is also precious :)

PS: Your garbage and meeting events are stored in .ics files, right? While there is an option to import events from an .ics file, you can import .ics files in your iCalendar too. That way, your iCalendar (url-based) will contain all events from meetings and for the garbage days too.

aceisace, 2018-11-11:
@surak 
Could you please review the suggestions I mentioned in the last comment. If we can finalise some of these suggestions, I can start with the implementation of those features. Thanks in advance.

surak, 2018-11-12:
Hi!

1. About month: It's, as I said, not just useful info, it's misleading. There's no reason for it to exist at all in the current incarnation. (The three-letters month name is fine).

2. About the weather, I will put it in huge letters. Like instead of the month now, or at the upper left corner where I put the clock, which will go. It's the single most important info before leaving the house.

*  About the events, one should know better what to put in an open calendar. It's absolutely easy to have multiple calendars, so I don't think it's a privacy issue.

*  About having many events, that is a problem in any calendar app where the screen space is not enough, no? For example, I am looking at my screen right now and I see this:

[image]

* The garbage icon is great, and I could do this for this specific case, but it's *TOO* specific. 

3. How slower is it to use truetype instead of loading every bitmap? I had no issues with that before. (I still need to help you with threading this code)

4. It's just better use of space.

5. I have to do it. We had some better days, but the 100% definitely runs out of my screen.

Sorry for the time again. Life is in a crazy rush here.

aceisace, 2018-11-16:
@surak 
Sorry for the late reply. I've been a bit busy this week.
I've reviewed a few things and experimented with a few others. To conclude, this is what's finalised so far:
1. Will be changed as you suggested. No additional discussion required.
2. Could you please clarify this point. It seems as I did not fully understand this part. A short sketch or photo would be nice.

* You have a point that the users themselves should decide what to put in an open calendar, so events will be shown with their names.

* The Calendar will display only the next upcoming event. Events within the same day but at a later time will be displayed as '2 more', just as this picture of yours suggests:
[image]

* I'll add an optional feature where users can use specific icons for recurring events, This feature will be disabled by default but the wiki will show in detail on how to do so.

3. I don't think using a font on the 2-colour display is an issue, but it definetely __is__ an issue with the 3-Colour one. The problem is that fonts are not made up of only black and white pixels, but rather quite a few grey-shades as well. All grey-coloured 'pixels' are translated to red by the 3-Colour display. This in turn makes it quite hard to read the letters on the 3-colour display and speeds up the ghosting effect. (Too much colour variation in too little space).

For example, this is how a word looks like normally:
[image]
And in magnification:
[image]

The above is 'translated' to the following, by reducing the colours to black and white, without grey:
[image]
It doesn't look too bad just yet, but see the zoomed-in version:
[image]
It doesn't look nice anymore, but still readable. The above happens on the 2-colour E-Paper display.

On the 3-Colour display, it looks a bit different:
normal:
[image] 
'translated': 
[image]
and zoomed-in after 'translation':
[image]

On a standard monitor, the differences seem minimal as a good lcd usually has very small pixels (high DPI value). But with this specific E-Paper display, the DPI rate is relatively low, which magnifies the problems by the colours.

By using an icon with 'customised' pixels, the translation step does not mess up with the colours themselves, ('translation'), leading to better readability and less ghosting. 

4. Will be changed as you suggested. No additional discussion required.

5. I'm not sure why the digits of the temp. and hum. are running off the screen. Did you maybe change the font size manually? In any case, this one is on hold for now as I do not seem to have issues with it. Please provide a picture showing the problem as I can't seem to replicate this issue.

aceisace, 2018-11-25:
@surak 
I've been trying out a few things here and there, trying  to implement all features you suggested, but it seems as though it's quite difficult to do so without changing the way on how a few things are handled. 

I can implement features that don't require changing a lot of the code itself as that would be too troublesome (e.g. positions). For example, the way things are now, I'd have to create new icons each time a change is suggested by a user. That's why, I'd say that in your case, you can test out the features and fixes you suggested and I'll take a look at them if you request a PR.

* The main problem is the conversion of pixels to data. for example, this is the section which handles the conversion:
```
def get_frame_buffer(self, image):
        buf = [0x00] * int(self.width * self.height / 4)
        # Set buffer to value of Python Imaging Library image.
        # Image must be in mode L.
        image_grayscale = image.convert('L')
        imwidth, imheight = image_grayscale.size
        if imwidth != self.width or imheight != self.height:
            raise ValueError('Image must be same dimensions as display \
                ({0}x{1}).' .format(self.width, self.height))

        pixels = image_grayscale.load()
        for y in range(self.height):
            for x in range(self.width):
                # Set the bits for the column of pixels at the current position.
                if pixels[x, y] < 64:           # black
                    buf[int((x + y * self.width) / 4)] &= ~(0xC0 >> (x % 4 * 2))
                elif pixels[x, y] < 192:     # convert gray to red
                    buf[int((x + y * self.width) / 4)] &= ~(0xC0 >> (x % 4 * 2))
                    buf[int((x + y * self.width) / 4)] |= 0x40 >> (x % 4 * 2)
                else:                           # white
                    buf[int((x + y * self.width) / 4)] |= 0xC0 >> (x % 4 * 2)
        return buf #due to python2 -> python3, int had to be added in 'get_frame
                   #_buffer 

```
I've been trying to find a way to adjust the values so that red remains red, black remains black and white remains white after the conversion. I did not have success on that, if you know a way, you're welocme.

* Another option is to create a new font, especially made for the 3-Colour and 2-Colour E-Paper displays, using only 2 colours, black and white, without grey-shades. This is quite troublesome and once again, if you know how to, please do help, But the advantage is that this would remove the need for icons (no random red pixels etc.). The disatvantage is that by using a font, there will be a lot more position-related issues, making it more complex

This is why I'd suggest that I/you create another page and let user choose which one they would like to use. If I provide instruction on how to design a new page, it should not be a problem creating a personalised page, just the way you want it.

For now, I'm going to add a few more weather details to display on the E-Paper calendar as well as a few more options I had in mind. Please do not mind that I couldn't implement all suggested features. Thank you. 

I will soon add a section containing a full set of instructions on creating your personalised page. When it's ready, I'll inform you.

surak, 2018-12-02:
Hi!!!!

I bought a smaller coloured e-ink, and I haven't had this problem you mentioned:
> 
> and zoomed-in after 'translation':
> 

And I don't see this problem.

[image]

So, I think we could use fonts for everything, if we fix this issue with font rendering at your screen.

surak, 2018-12-02:
Ps: I'm using the inkyphat library for this little one

aceisace, 2018-12-03:
Hi, thanks for your reply.

the picture clearly shows that the inkyphat has no 'mixing' of the colours around the text. This sounds really interesting. It seems as though the inkyphat library uses a different (and much better) code for converting the pixel colours to data. Thank you for sharing :+1: 

I am going to have a deeper look at the way the inkyphat library handles the conversion on the weekend and try implementing it on the software for this repo.

aceisace, 2018-12-09:
@surak 
I had a look at the inky library from pimoroni. Although the approach to turn the image into pixels is quite different (and slightly more time-consuming), it does fix the trouble of not being able to read text, especially when using small font sizes. You've probably realised this but the inky library actually does not convert rgb images to grayscale (which is waveshare's approach) as it only works with the 3 pre-defined colours. 

What I can do is fine-tuning the pixel conversion (see above snippet) so it can fix the problem with the fonts. Also, using thicker fonts and bigger font sizes helps reduce the issue too.

Even so, the bigger problem is the difficulty for rotating the text as the E-Paper uses the vertical layout instead of the default horizontal one. I'm having difficulties writing a function which requires these 3 things:
* space of the text box
* position of the text box
* Text with font name and font size

and then draws the text at the screen. Any help is welcome since I'm having a few diffculties with this.

Crickus, 2018-12-18:
Hi.
I was gifted with this thing
[https://www.waveshare.com/product/modules/oleds-lcds/e-paper/9.7inch-e-paper-hat.htm](url)
and have a great desire of applying your project on it.
As I understood all spaces and drawings conversion in project was counted for 7.5 screen. So I'll have a great headache stretching it. Am I right? Could you provide me with a peace of advise as a newcomer.
And another thing I want to change is make it in Cyrillic (Ukrainian or Russian).

aceisace, 2018-12-18:
@Crickus 
Hi, thanks for your interest in this project. As you know, this software was written for the 7.5" E-Paper Display from Waveshare/GoodDisplay, it will not work that easily with other E-Paper types due to the different resolution, (and pinout) etc. I did have interest in the 9.7" E-Paper from waveshare, but I think that it's still expensive (170USD) to prototype with. Sure, the refresh time is a lot faster, but it doesn't have colours. Maybe it's better suited for other tasks, like a E-book reader.

The good news is that this particular E-Paper (9,7" model) also works with SPI so you won't need to start from scratch to get this software up and running on your E-Paper display.

If you are an expierenced user, I'd recommend you take a look at this **[Gravit-Designer file](https://github.com/aceisace/E-Paper-Calendar-with-iCal-sync-and-live-weather/blob/master/For-developers-only/cal-v1.2-templates.gvdesign)** (open with Gravitdesigner) which contains nearly everything needed for the icons. If you adapt the current code (change the resolution, change icons (with bigger ones), font size, positions and a few more things), you can use this software on the 9.7" one. This is a big task, especially for such a large display, but I can help you with that. For starters, you can try using the Pillow library for python3 to display a square, then move on to text and more complex stuff.

If I did have the 9.7" E-Paper, I could do all the adjusting myself and publish a new repo for that display. But to be honest, there are bigger E-Paper displays for even better prices, like this one: [12,48" E-Paper from GoodDisplay](https://goodlcd.en.alibaba.com/product/60794581741-807540602/large_12_48_color_e_ink_display_1304_984_for_supermarket_electronic_price_tag.html?spm=a2700.icbuShop.84.12.43052e3biBeHc0), so I would prefer getting one of those for a bigger display. 

In regards to the language, so far, only 13 icons need to be changed for each language (month icons and the weekdays-icon). If you are interested, you could help translate them in Cyrillic.

aceisace, 2018-12-18:
@surak 
I've finally had the time to work on this project. Here is the picture of the E-Paper Calendar running on a beta version:
[image]
What do you think of this update? I've removed the date and added some more weather data, including a short description.

surak, 2018-12-18:
> @surak
> What do you think of this update? I've removed the date and added some more weather data, including a short description.

I like it! Right now I'm overseas for xmas vacation, in January I'll merge it into my fork. Did you do something with the drawing library?

Crickus, 2018-12-19:
> does the orange strip fold backwards?
Near the edge strip is more or less flexible, so I think it possible. I made few photos to show.
[image]
[image]

> I think that it's still expensive
I agree with you absolutely. Price is inadequate. But some of my friends decided what "more expensive" is synonym of "high quality". We'll see.

I think I'll try change sizes as you said to fit my resolution. I more or less common with this things. 
Maybe I'll try to make zone with nearest tasks as @surak did.
I planing to start on weekend.

aceisace, 2018-12-19:
@Crickus 
Good luck with the adaptation of this software for the 9.7" E-Paper Display from waveshare. If you need help, just drop me a message and I'll help as much as possible.

aceisace, 2018-12-19:
@surak 
I'm glad you like the new version (still in beta) of the software. Hopefully I can release it in the upcoming holidays. 

In regards to the Inky library from Pimoroni, I've had a quick look at their library. They use numpy to create a matrix filled with zeros in the size of the the resolution. After adding text with Pillow, the data overwrites the zeros with 0 for white, 1 for black and 2 for red. 

Due to this, they can do operations easily (like flipping the image etc.) but at the cost of a longer preparation time for each image, even without operations. Although this is an option, I'd rather not delay the preparation image for each image since it already is pretty long.

On the other hand, after looking into the inky library, I think I have found a better way to translate the pixels, especially for text. This removes the problem of the red border around each letter. Once I can get it working, I'll update you.

aceisace, 2018-12-24:
## Update
### The E-Paper Calendar software has been updated from v1.3 to v1.4 just recently (Late December).
It contains more features regarding the weather as well as some back-end improvements. If you are using an older version, don't forget to update.

**P.S.: If you notice any bugs, please let me know by opening an issue. Thanks in advance.**

aceisace, 2019-01-03:
@zAm1987 
Hi, thanks for your interest in this project. Also, nice to hear you're back to learning python :)

I haven't thought of using multiple ical files/urls so far and therefore did not implement this feature. As this seems like a good new feature, I'll update the software as soon as I find a way to implement multiple iCal files/ URLs. Hopefully in the next few days, this feature should be implemented. Thanks for the suggestion. :+1:

aceisace, 2019-01-03:
@zAm1987 
OK, I've updated the software and implemented the feature of multiple iCal URLs. To update, please follow the instructions on the README file (in the updating section). To use multiple iCal URLs, you can add them in the settings file. Instructions on how were added in the settings.py file. Although I have tested it, could you also provide some feedback after testing the update? Thanks in advance.

zAm1987, 2019-01-04:
Wow, that was fast :)
I'll have an look at it this weekend.

zAm1987, 2019-01-04:
I tried installing the updated version twice from scratch (new raspian stretch lite) with the installer WITH debug and something seems wrong, as the script doesn't start on reboot and stable.py cannot be started manually.

I got the following outputs during setup:
-----------------------------------------------
_Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-vrd4dwxf/Pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-tpjqezox-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-vrd4dwxf/Pillow/
[...]
Failed building wheel for Pillow
[...]
The headers or library files could not be found for jpeg,
  a required dependency when compiling Pillow from source.

  Please see the install instructions at:
     https://pillow.readthedocs.io/en/latest/installation.html
[...]
Traceback (most recent call last):
  File "/home/pi/E-Paper-Master/Calendar/monocolour-converter.py", line 37, in <module>
    from PIL import Image
ImportError: No module named 'PIL'_

and the following when trying to manually start stable.py after a reboot (as ist does not start automatically):
-----------------------------------------------
_Traceback (most recent call last):
  File "stable.py", line 13, in <module>
    from icon_positions_locations import *
  File "/home/pi/E-Paper-Master/Calendar/icon_positions_locations.py", line 3, in <module>
    from PIL import Image, ImageFont
ImportError: No module named 'PIL'_

Maybe some missing dependencies?

aceisace, 2019-01-04:
@zAm1987 
I'm very sorry for the inconvenience with the bug on the software. I'll test both installers on a fresh install on Raspbian stretch lite and fix the issue asap. When it's fixed, I'll notify you.

aceisace, 2019-01-05:
@zAm1987 
Hi, I've installed the E-Paper software on a fresh install of Raspbian Stretch (13-Nov-2018). Indeed, the Installer did output an error message regarding the PIL library, but the script itself was working fine:
[image]

The reason you encountered this error is because PIL was recently updated (1st Jan 2019) from 5.4.0 to 5.4.1. There was a bug with the latest release, leading to strange errors when installing. I've fixed the PIL issue by specifying the version (for now 5.3.0) in the Installers. 

The reason the auto-start at boot wasn't working is also due to the bug from PIL. This issue has also disappeared as the cause for this was the same as the PIL library.

Just to confirm, could you please test out the updated installer? Thanks in advance.

zAm1987, 2019-01-08:
Hi,

I just installed it again from scratch to test and the setup went through without errors.
The stable.py script also works fine now ;)

aceisace, 2019-01-08:
@zAm1987 
I'm glad to hear that the installer is working fine and without any issues :)

I have been using 2 urls since nearly a week without any errors. Before going deeper, could you try using the following list in the settings.py file:
```
ical_urls = [
"https://calendar.google.com/calendar/ical/en.usa%23holiday%40group.v.calendar.google.com/public/basic.ics",
"https://www.schulferien.org/media/ical/deutschland/ferien_baden-wuerttemberg_2019.ics?k=5dMdMMsZXnpA5bg9D0GY246EAIRqWAWv9fRtQ-ekYM1ADwSZeJpvo4fd-NhwlSxOPxm6BQRUgRvcQSKArfGUwZQLDfYfaiWnA1E9diCCB8s"
]
```
These are some generic iCal urls I used while testing and therefore are confirmed to work. If these do work, then the issue lies with the iCalendar. In that case, I'll look deeper into the issue.

zAm1987, 2019-01-08:
It is working fine with your example ical urls.

I tested my ical file and the one from my wife each on it's own and her file alone causes the errors, too.
I checked the urls twice and they are correct. As she uses the google calendar a) quite extensive and b) with special characters in event names it seems that the sheer number of events or the special characters are the problem. My file is not that huge and i do not use special characters normally.

zAm1987, 2019-01-08:
Yes it sends reminders by mail. I just had the same idea, because I cross checked our calendar settings and the reminders are the only difference I found in the settings.

Now I disabled all reminders in the google calendar web interface, rebooted the pi and startet stable.py again but the error still occurs.

aceisace, 2019-01-08:
@zAm1987 
If the error still persists, I can only think of 2 possibilities:
1) The changes in your iCalendar are not yet implemented on all events. To test if this is the case, try your the URL tomorrow or at least in a few hours.
2) It's possible some reminders (maybe past ones) have still got the Email-reminder in their details. If 1) did not fix the issue, export the iCalendar to a file, create a new iCal with a different name and lastly, import the file from before in the new one. Then, test it by using the new iCalendar's URL.

zAm1987, 2019-01-09:
Tested it again.

1. Waited a day and tried again. Also checked all upcoming events for event-specific email reminders. There are none. Maybe only last month and before. It still does not work.

2. Exported and imported into new calendar. Works fine, but I do not know why :D . If there are old events in the original calendar, which cause the error then they are there in the new one as well now...

aceisace, 2019-01-09:
@zAm1987 
I'm not exactly sure as to why the second method fixed the issue, but as long as it works, it's alright :)
I'll add the fix in the F.A.Q page of the wiki when I find out more about the issue. Thanks for your help in testing.

aceisace, 2019-01-09:
@surak 
@zAm1987 
@StevenSeifried 
I've implemented the feature of displaying events from the current day (currently in beta). By making the month icon smaller, the free space created is used to display the event name and time. For multiple events, the Calendar shows '+ x events' where x is the number of more events on the same day.  Could I have some feedback on this feature? Thanks in advance.
[image]

zAm1987, 2019-01-10:
I like it and would even display more than one event. My plan was - once I get the hang of python and your code - to display some events anyway.

While I think of it, maybe it ist more useful to fetch the events from the next 24 hours instead of today. Because when looking at the calendar at 23:00 for example, I am more interested in what ist coming next / after sleep instead of only the events that can happen in the last hour of the day only.

aceisace, 2019-01-10:
@zAm1987 
Thank you for your opinion. If you have any more improvement ideas, feel free to write them here as well.

I also think that displaying the events from the next 24 hours are more useful than the events from today. Since it's simpler to fetch events from a specific day (like today and tomorrow) compared to the next 24 hours, I'll use that option for now. I'll add another 'event' line with text for the next day's event, so it'll look like this on the E-Paper:
```
Today: Conference at 14:00   + 1 Event
Tomorrow: Electrician at 9:30   + 2 Events
```
Is this somewhat better?

zAm1987, 2019-01-13:
Sounds like a good compromise ;)

aceisace, 2019-01-13:
@zAm1987 
Thanks for your feedback. I've implemented the feature for the events from today and tomorrow. Once I test it thoroughly, I'll merge the changes in the public version. 
Preview of the new feature:
[image]

surak, 2019-01-16:
I am very happy with the development of this software. I've been through some rush here, so I'm unable to help lately. I will come back to you, and hopefully I can help with threading the code.

aceisace, 2019-01-16:
@surak 
Thank you for the compliment :). The development would not have been possible without the help from everyone, so thank you all as well. You don‘t need to worry about helping out. You can help whenever you have the time to do so.

smartroad, 2019-01-20:
I am loving the idea of this! Are there any plans for having it more like an adgenda? Rather than just display the calendar, actually list the upcoming events? Would help me out for my work schedule lol

aceisace, 2019-01-20:
@smartroad 
Hi, thanks for your interest in this project. 
I am testing a beta version of the software which displays events, when they start and how many events you have left today and tomorrow. This is a preview of the beta version: https://github.com/aceisace/E-Paper-Calendar-with-iCal-sync-and-live-weather/issues/1#issuecomment-453857058. 
So yes, it's work in progress and it will be implemented in the next release, but I can't exactly say when.

Ahbrown41, 2019-01-21:
Have you guys thought about using the Google calendar API as well?  Excited to test the version that displays the events, I would personally prefer a smaller calendar and a view of events and tasks :-). Perhaps I can help but Python is not my first language..

[17 more replies]

closed 2019-02-05 by aceisace