aceinnolab/Inkycal #4: Change color

startedfromtheb0ttom, 2018-09-23:
Hello

In the pictures some of the writing is red, but my E-Ink Display can not display this color. It can only show black. How can I change the corresponding color icons to black so that they are clearly visible on the display

aceisace, 2018-09-23:
@startedfromtheb0ttom 
If I've understood correctly, you're probably using the waveshare 7.5" 2-Colour Display, which does not support red. 
You did mention 'clearly visible' in your comment. Does that mean that the red parts are displayed in a different manner? Could you please post a picture of the display to show the issue?

The best and most permanent fix I can think of would be to reduce the colours of the icons from 3 (black, white & red) to 2 colours (black and white). I'll try to figure out something regarding that (maybe creating a python 'converter' or manually convert the icons to just 2 colours with GIMP 2).

startedfromtheb0ttom, 2018-09-24:
I use this 7,5" Waveshare E-Ink Display (https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT). It's not the 3 Color Version. The red icons are not clearly visible on the screen from a distance
[image]
[image]
[image]

aceisace, 2018-09-25:
@startedfromtheb0ttom 

Although this software wasn't originally intended to be used with the 2-colour E-Paper Display, I'm going to add full support for the 2-colour version soon. 

For now, I have a 'quick fix' which will allow you to use the software on the 2-colour E-Paper Display. I have tested it and confirm it works:
[image]

### Usage; 
**`curl -sSL https://raw.githubusercontent.com/aceisace/E-Paper-Calendar-with-iCal-sync-and-live-weather/master/2-Colour-support/setup | bash`**

Let me know if it works, thanks.

aceisace, 2018-09-25:
I'll close this issue for now then.

closed 2018-09-25 by aceisace

startedfromtheb0ttom, 2018-09-26:
Thank you! It's work's

aceisace, 2018-10-02:
## Important Notice:
A kind note to anyone using the software, please update to the latest version (early October 2018) as it contains a critical update which prevents ghosting and damage to the display from over-voltage in the long-term. It also contains full support for the 2-Colour E-Paper version. Refer to the wiki for updating instructions.