aceinnolab/Inkycal #6: Non-ASCII character

StrangeDaize, 2018-10-01:
A simple resolution for:
SyntaxError: Non-ASCII character '\xc2' in file stable.py on line 151, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details

Is to change from:
            temperature.text((2, 8), (Temperature + " °C"),  fill=0 ,font=font)
to:
            temperature.text((2, 8), (Temperature + u'\xb0' + "C"),  fill=0 ,font=font)

aceisace, 2018-10-02:
@StrangeDaize 
Thanks for opening your first ever issue and pointing that out. I ran into this error while I was developing this software with python2 but I couldn't manage to fix it at the time. 

After switching from python2 to python3 (mainly because some used packages won't be supported by python2 in the near future), this error was gone and hence, no longer an issue. 
Nonetheless, I will definetely add your fix to the stable.py file soon in case users want to use it with python2. Thanks for this improvement.

aceisace, 2018-10-02:
## Important Notice:
A kind note to anyone using the software, please update to the latest version (early October 2018) as it contains a critical update which prevents ghosting and damage to the display from over-voltage in the long-term. It also contains full support for the 2-Colour E-Paper version. Refer to the wiki for updating instructions.

aceisace, 2018-10-02:
@StrangeDaize 
I'm going to add this at the beginning to the file (stable.py) to fix the encoding. Is this all right? Or do you recommend changing the 'temperature' line?
#!/usr/bin/python
# -*- coding: utf-8 -*-

StrangeDaize, 2018-10-03:
I added just the encoding for the degree symbol, that way utf-8 isn't
forced universally.
This is a personal preference that allows me more flexibility.

aceisace, 2018-10-03:
@StrangeDaize 
I've added both implementations by adding the encoding at the beginning of the file and leaving an uncommented line with your fix.

> A simple resolution for:
> SyntaxError: Non-ASCII character '\xc2' in file stable.py on line 151, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details
>
> Is to change from: [...]

aceisace, 2018-10-03:
I'm closing this issue for now. If you think that this issue should remain open, please let me know.

closed 2018-10-03 by aceisace