aceinnolab/Inkycal #13: Calibration of black/white screens - was it updated?

surak, 2018-10-10:
Does it need an extra step of "red" calibration?

The code looks like it takes forever to go further.

aceisace, 2018-10-11:
@surak 
The red part does not need to be calibrated if the E-Paper display does not support red. It's only intended for the 3-Colour E-Paper display. 

I've updated the calibration file when I released v1.2. Here is the link to the [updated calibration file](https://github.com/aceisace/E-Paper-Calendar-with-iCal-sync-and-live-weather/blob/master/2-colour/calibration.py). It's now installed by default when running the new installer.

surak, 2018-10-11:
How about a setting "display=color| display=bw"? Something like that? That would eliminate those hacks and scripts, as both are supported at the same time.

aceisace, 2018-10-11:
That's a good idea. I'll add an option to choose the display version (bw/colour) in the main script and test it out. Once that is ready, I'll let you know.

aceisace, 2018-10-16:
@surak 
I've just released an updated installer for the next release (v1.3). It has more options in the settings.py file (including an option to choose the display) and doesn't require seperate folders for bmps either. 

The installer has been cleaned up greatly and is less complicated now.

aceisace, 2018-10-17:
I've tested the new version (v1.3) and it seems to work fine on my E-Paper Display. Can I ask you to review the new version before I officially release the new version? Thanks in advance

surak, 2018-10-20:
Sorry for the delay. Usually, I only have the weekends. I merged your master into my fork and it works wonderfully. I am still trying to remove the bitmaps for the month and change it into a short list of the next day's appointments. 

About the month: I managed to write it using TrueType fonts instead of bitmaps - it's just a big number after all. What do you think?

aceisace, 2018-10-20:
@surak 
Thanks for the reply. I'll wait for your replies on weekends then. I'm glad to hear the new Installer and version (1.3) are working fine. 

Showing upcoming events on the display is a good idea. If you need help with extracting the events from the iCal, I can provide you with an example code. Let me know if you need it.

As for replacing the months bmp with writing using ttf, I have a tip: Do take a look at the temperature, humidity and date writing functions in the main script (stable.py). All of these functions create a box first, add writing, rotate the box by 90 degrees clockwise and place it on the given position.

Please keep in mind that this method is a lot more complicated than simply pasting an image with fixed dimensions at a given place. For reference, a more detailed explanation of the date writing function at the top left corner:

* option 1:
`# date writing function`
`space1=Image.new('1', (115,25), color=255)` ->define a box with the size: 115x25px in white
`date = ImageDraw.Draw(space1)` -> define drawing that box
`date.text((2, 3), (time.strftime('%a %-d %b %y')),  font=font, fill=0)` -> write the date in the box with an x-offset of 2 and y-offset of 3 using the font defined before in black fill
`rotate1 = space1.rotate(270,  expand=1)` -> rotate that box 90° clockwise, expanding it to prevent stretched or squashed images.
`image.paste(rotate1, (595,20))` -> paste the image on the display at position 595(x) and 20(y)

But placing an image is very simple:
* option 2
`draw = (ImageDraw.Draw(image)).bitmap` -> describe what 'draw' means
`icon = open(path_to_icon.bmp)` ->define name and path of bmp
`place = (340,280)` -> x and y co-ordinates on where to place the icon
`draw(place, icon)` -> draw the icon at the place

For static content which doesn't change, I'd recommend the latter option. I'm not sure if it's possible but if you know of a way to create a function that does all the steps from the first option, it'll make writing text a lot easier. Then you won't have to write all of this each time you want to write text on the display.

aceisace, 2018-10-31:
I'm closing this issue for now. If you think that this issue should remain open, please let me know.

closed 2018-10-31 by aceisace