aceinnolab/Inkycal #15: Wrong text on main page

FrederikLauber, 2018-11-08:
Hi,

the mainpage states:
"Altough mainly intended for the Raspberry Pi Zero W, mainly since it only requires 1W per hour, it works on the Rapsberry Pi 2/3 as well.".

My grievance is with the "1W per hour". Watt is already a measurement of energy per time making the statement physically nonsensical. It should either read "1W" or something like "1VAs per hour".

Also, (just noticing while writing) Altough is writen Although I think,
Cheers,
Frederik

aceisace, 2018-11-08:
@FrederikLauber 
Hi Frederik. Thank you for pointing that out. I've changed the following:
`Altough mainly intended for the Raspberry Pi Zero W, mainly since it only requires 1W per hour`
with:
`Altough mainly intended for the Raspberry Pi Zero W, mainly since it requires very little power (1W)`
Does this sound correct now?

FrederikLauber, 2018-11-08:
Sounds perfect! Thanks for changing it so quickly!

closed 2018-11-08 by FrederikLauber

aceisace, 2018-11-08:
You're welcome. Thanks for closing this issue.