aceinnolab/Inkycal #16: Typo in file name

surak, 2019-01-03:
Separator is written seperator in calendar/other/en

surak, 2019-01-03:
And the January month bitmap says "Janruary"

aceisace, 2019-01-03:
@surak 
OK, the typos have been fixed. Could you please confirm this? Thanks

surak, 2019-01-04:
I just tested the stable(beta).py - it still calls seperator FYI

aceisace, 2019-01-04:
> Separator is written seperator in calendar/other/en

It's possible I misunderstood something, but I did change seperator to separator on my repo:
[image]

aceisace, 2019-01-05:
@surak 
OK, I've fixed the typo in the stable(beta).py as well. Are there any more typos which need fixing?

aceisace, 2019-01-06:
@surak 
> And the January month bitmap says "Janruary"

Is it fixed now?

surak, 2019-01-07:
Sorry, back to work, didn't have any time anymore. If there's something else, I will reopen this one. Meanwhile, I close it!

closed 2019-01-07 by surak