aceinnolab/Inkycal #18: Uninstaller script or description how to uninstall

schneekluth, 2019-01-16:
Please extend the installer script with an option to uninstall your programm. Or maybe just add a step-by-step guide on how to uninstall it.

aceisace, 2019-01-16:
@schneekluth 
Hi, I'll add an option to do so in the next few days. Just out of curiousity, is there any specific reason you want to uninstall? Maybe I could change something if there is room for improvement.

aceisace, 2019-01-16:
@schneekluth 
I'm glad to hear you aren't discontent with the E-Paper Calendar software.
And yes, this software uses supervisor to handle starting the script at each reboot and not system.d. The reason is very simple, you just need very little configuration (4 lines are enough for supervisor) and it's easy to change debugging options, too. 
As for your Nextcloud iCalendar, are you using email triggers for events by any chance? If so, this might be the reason as for why the iCalendar isn't working and I can provide you with a temporary fix. Otherwise, if you could past the error code, I'll be able to help you more specifically.
The task list feature from iCalendars is something I'm also planning to implement soon. But it seems that the current free space is limited for more text-related features like displaying tasks. I'd have to create a new page and display them on the empty page. So yes, it's work in progress, but I can't say just yet how long it'll take to complete.
For the uninstalling option, I've created a temporary (not yet tested) solution in the [wiki](https://github.com/aceisace/E-Paper-Calendar-with-iCal-sync-and-live-weather/wiki/Uninstalling). It works by uninstalling the libraries, removing the configurations and removing the folder with the software. It should work, but if something isn't right, please let me know here. Thanks in advance.

aceisace, 2019-01-20:
@schneekluth 
I'm not really familiar with Nextcloud and have never used it myself either so I can't really say anything too specific about nextcloud. I did a small search on how to export a calendar in nextcloud and it seems as though it is possible.

There should generally be 2 ways to export the calendar. The first is as a .ics file, which after downloading can be imported to Google Calendar. The second option is to export via URL. In this case, you can copy the link from your nextcloud calendar and import it in Google calendar.

schneekluth, 2019-01-20:
I tried it with the URL-option but it did not work. Importing to google is something I don't want. I use Nextcloud to stay as far away from google as possible :) But anyway, thanks for your research. Nextcloud integration would be very nice in your project. Especially with your announced task feature.

schneekluth, 2019-01-20:
No, I mean a list of tasks. For every caldav synced calendar you can assign a set of tasks to do. It has nothing to do with the event in the calendar itself. Think about it more as a to do list that is automatically synced with the calendar. I use it with this [app](https://play.google.com/store/apps/details?id=org.dmfs.tasks&hl=de).

aceisace, 2019-01-20:
Now I understand what you meant by the task feature. But with the current free space left on the E-Paper, it seems almost impossible to add more text (for example tasks) without large modifications of the layout.

I'll see if I can do something about it by doing a more thorough research about iCalendars and tasks in the next few days.

aceisace, 2019-02-03:
@schneekluth 
Hi, very sorry for the late reply. I've been ill so I couldn't work as well this week. 
Please take a look at the new installer ([Installer with debug](https://github.com/aceisace/E-Paper-Calendar-with-iCal-sync-and-live-weather/blob/master/Installer-with-debug))
I've implemented the feature of updating, installing and uninstalling the E-Paper Calendar software. 

**Good news:** Today, I've also managed to release v1.5, which contains a lot of improvements and more features. If you don't mind, maybe you could test and review the software? Also, thanks for pointing out the uninstalling feature :+1: 

I'm also closing this issue as the uninstalling function has been implemented as of today. If you feel a need to re-open this issue, please feel free to do so. Many thanks.

P.S.: You can dicuss the task feature of the iCalendar in #1 now.

closed 2019-02-03 by aceisace