aceinnolab/Inkycal #19: Remove hardcoded directory

surak, 2019-02-05:
Since the directory cloned is "E-Paper-Calendar-with-iCal-sync-and-live-weather", it makes no sense for it to expect this to be at "/home/pi/E-Paper-Master".

One can refer to files from his own directory and relative paths.

aceisace, 2019-02-05:
You do have a point. Could you provide a suitable suggestion then? Thanks

aceisace, 2019-02-12:
For now, I think I'll leave it as it is. Otherwise things might get a bit overly complicated. In the next release, I'll try doing something about it. 

I'll be closing this issue for now. If you need to re-open it, please feel free to do so.

closed 2019-02-12 by aceisace

aceisace, 2019-05-21:
@surak 
I've implemented the relative-path feature and removed the hardcoded directory with the new release.

surak, 2019-05-22:
Cool! I have to merge your stuff! Thanks!