aceinnolab/Inkycal #20: Support, Discussion, Suggestions, Critics for v1.5

aceisace, 2019-02-05:
Here you have an option to chat about improvement ideas regarding v1.5 of the E-Paper Calendar software.

As usual, nothing and nobody is perfect, but we can strive to get closer towards that. Please post your comments in a friendly way and remember to be nice. Thanks for your understanding.

All improvement ideas are welcome.

zAm1987, 2019-02-10:
I tried a fresh install of raspian stretch lite with the installer (option 2) for the new version.
After the first try somehow the E-Paper-Master folder was empty and so there was no script to start.

During my second try I watched the installer and it worked well. Only the description says "nano ... settings.py", but the filename is "settings.py.sample".

Also I tried a manuel calibration, because the display was offline for a while and the icons where somewhat distorted after running the new version for the first time. During the calibration the following error occured (I have the 2-color version)

`pi@raspberrypi:~/E-Paper-Master/Calendar $ python3.5 calibration.py
calibrating black...
Traceback (most recent call last):
  File "calibration.py", line 40, in <module>
    main()
  File "calibration.py", line 37, in main
    calibration()
  File "calibration.py", line 27, in calibration
    epd.display_frame(epd.get_frame_buffer(red))
UnboundLocalError: local variable 'red' referenced before assignment`

aceisace, 2019-02-10:
@zAm1987 
Hi, it seems as though there were some conflicts with the Installer and the new version. 
The error above seems to be caused by 2 different versions (v1.4 and v1.5) running in the same environment. 

The best way to fix this is by backing up modified files form the E-Paper-Master directory (should normally only be the settings.py file) to the home directory. Then, please re-run the installer to uninstall the software and lastly, use the installer to install the software again. Your issue should be fixed after a reboot. Should something still go wrong, please paste the error here.

The missing settings.py file was a bug in the Installer. I've just now fixed it so a settings.py file exists when choosing the install option form the installer.

Thank you for pointing out the bug :+1:

zAm1987, 2019-02-11:
@aceisace

The question is how the two different versions came into place, because as I described it was a completely fresh Installation from an untouched raapian stretch lite image.

Anyway, I will try the uninstalling and reinstalling option and report whether it works or not ;)

aceisace, 2019-02-12:
@zAm1987 
I‘ve found the cause for this error. In settings.py, line 27 should have been indented but it wasn‘t. I‘ve just now fixed and tested the calibration module for the 2-colour display. It should work just fine now.

zAm1987, 2019-02-12:
@aceisace 

I used the uninstall option and installed again. I can confirm that the calibration now works as it should. 

The settings.py file however is still missing after the install option.

And again, the Icons and the font looks somewhat distorted. I am not sure if this was the case with the older version, but it seems that everything is pixelated. Maybe the script still thinks it's a 3-color-version and so the missing red causes the jagged edges? It could be the voltage, too, because my actual "test bench" is fed via an usb-hub and the e-paper-display is known for having problems with slightly different voltages...

[image]

aceisace, 2019-02-12:
@zAm1987 
It's good to hear that the calibration module works as expected.
I'll look into the missing settings.py file again. 
For now, please type in the following commands in the terminal:
```
cd /home/pi/E-Paper-Master/Calendar
cp settings.py.sample settings.py
```

In regards to the slightly disorted icons, this is a non-desired side-effect caused by switching from specially formatted .BMP icons to standard .JPEG icons. Up to v1.4, all icons were rotated, inverted and in .BMP format. The main problem was that the time taken for making even a single 'perfect' icon (in BMP) was too long for a project in development phase (adding more features etc.), so I'm using JPEG icons now. If you want to know more about it, please see this comment in issue1 [(image to data conversion)](https://github.com/aceisace/E-Paper-Calendar-with-iCal-sync-and-live-weather/issues/1#issuecomment-439564160)

You can try reducing the disortion by editing this section in `home/pi/E-Paper-Master/Calendar/epd7in5.py`. 

```
for y in range(self.height):
            for x in range(self.width):
                # Set the bits for the column of pixels at the current position.
                if pixels[x, y] != 0: # this part
                    buf[int((x + y * self.width) / 8)] |= 0x80 >> (x % 8)
        return buf
```
Currently, anything that isn't fully white, is converted to shades of grey. I did try to modify these values, but the results aren't as good yet. Once I've fixed a few iCalendar-related issues, I'll work on improving this section.

So the issue is not related to the fact that the main script thinks you're using a black-white-red display, nor is it related to a different voltage level caused by using usb-hub.

zAm1987, 2019-02-12:
@aceisace

Ah okay, somehow i missed the switch to the jpg format. Sounds reasonable and I can understand it. Thanks for the explanation. :)
I will try to play around with the icons, when I finally have an bit more time.

For the missing settings file I already nano'ed into the sample file and saved / renamed it in order to fix it ;)

aceisace, 2019-02-12:
@zAm1987 
Sound good :)
Keep me updated when you make some progess with the icons. Thanks.

aceisace, 2019-02-16:
For anyone interested, here is a breakdown on how long the main programm takes for each section:
* Preparing monthly Calendar template: **<1s**
* Connecting to openweathermap api servers and fetching data: **17 seconds**
* Filtering upcoming events: **<3s**
* Converting Image to Data and sending it to the display: **60-70 seconds**
* Refreshing E-Paper-Display: **6s** for black-white, **16s** for black-white-red E-Paper display
* Calibration time for  black-white-red E-Paper-display: **443s** 
* Calibration time for  black-white E-Paper-display: **247s**

jankosso, 2019-02-18:
The Alert issue is quite annoying, because all my calenders use them.

Can you elaborates a little bit, what causes this issue?

aceisace, 2019-02-18:
@jankosso 
The Alert issue is supposed to have been fixed in the last few days. It consists of this section:
```
for icalendars in ical_urls:
                decode = str(urlopen(icalendars).read().decode())
                #fix a bug related to Alarm action by replacing parts of the icalendar
                fix_e = decode.replace('BEGIN:VALARM\r\nACTION:NONE','BEGIN:VALARM\r\nACTION:DISPLAY\r\nDESCRIPTION:')
                #uncomment line below to display your calendar in ical format
                #print(fix_e)
                ical = Calendar(fix_e)
```
The fix_e (fix error) line is meant to fix alert-based errors, at least for Google Calendar. I am using this myself and can confirm that it works with Google (i)Calendar.

The main issue lies with the ics.py python library [(see here)](https://github.com/C4ptainCrunch/ics.py). The ics.py library is a bit picky with the formatting of the iCalendar. Although there is a standard for the iCalendar format, not every provider is using the latest version. Due to the differences in formatting, errors are caused. For example, in the iCalendar from google, the `Alert=None` part causes the issue. Ics.py does not support `Action=None` and throws an Alert-error. The above snippet replaces the `None` with `Display` so ics.py does not throw the error. You can let python print your iCalendar in the iCalendar format by uncommenting the line `print(fix_e)`
I did open an issue at the ics.py repo to report this issue and another one, but it seems as though it didn't really get any attention from the developer.

As every iCalendar (from different providors) varies slightly in format, it's difficult to implement the support for every single format. As Google Calendar is the most used one, it has been implemented first.

7n2jju, 2019-02-19:
Hi, I just report problem when week_starts_on = "Sunday" is used and fixed it.

This causes an error like:
```
Traceback (most recent call last):
  File "E-Paper.py", line 282, in <module>
    main()
  File "E-Paper.py", line 82, in main
    draw(weekplace, weeksun)
  File "/usr/local/lib/python3.5/dist-packages/PIL/ImageDraw.py", line 135, in bitmap
    self.draw.draw_bitmap(xy, bitmap.im, ink)
ValueError: bad transparency mask
```
So, I just removed line 82 in E-Paper.py that calls `draw(weekplace, weeksun)` following to the case Monday and it worked fine when week starts from Sunday.

aceisace, 2019-02-19:
@7n2jju 
Hi, thanks for finding the bug. I've just now removed line 82 from the E-Paper.py file on this repo to make sure it doesn't happen again.

jankosso, 2019-02-20:
Regarding the Google calendar issue:

@aceisace thanks for the info, i dug a little deeper into that lib and i think i found the issue.
Your workarround only fixes Alert=None issue, but the library also doesn't work with Alert=Email.
In the alarm.py you can read, that is still a ToDo ;-)

`def get_type_from_action(cls, action_type):
        # TODO: Implement EMAIL action
        if action_type == 'DISPLAY':
            return DisplayAlarm
        elif action_type == 'AUDIO':
            return AudioAlarm
        elif action_type == 'NONE':
            return None

        raise ValueError('Invalid alarm action')`

I just changed it to return None, when the action is Email and this fixed it:

`def get_type_from_action(cls, action_type):
        # TODO: Implement EMAIL action
        if action_type == 'DISPLAY':
            return DisplayAlarm
        elif action_type == 'AUDIO':
            return AudioAlarm
        elif action_type == 'NONE':
            return None
        elif action_type == 'EMAIL':
            return None

        raise ValueError('Invalid alarm action')`

So for a workarround without patching the lib, you can also replace the ACTION:EMAIL 
Like so:
fix_e = fix_e.replace('BEGIN:VALARM\r\nACTION:EMAIL','BEGIN:VALARM\r\nACTION:DISPLAY\r\nDESCRIPTION:')

aceisace, 2019-02-20:
@jankosso 
Thank you very much for taking your time to research about the iCalendar issue and finding a way to solve the issue :)  :100: 
I'll test out the recommended change tomorrow and will implement it as soon as I can confirm that it works.

aceisace, 2019-02-21:
@jankosso 
I've tested out your suggested fix for the `raise ValueError('Invalid alarm action')` and have implemented the change on the master branch. Thanks for contributing :+1:

aceisace, 2019-02-25:
**Quick update:**
The software was recently updated. Some of the main improvements are:
* Added events_max_range parameter in the settings.py file. See the list in the README for more info
* The space below the Calendar is used more efficiently. The Display will now display all events that can fit below the monthly Calendar. Depending on how much space there is, there may be 4 or 6. displayed events
* Improved the calibration time by opening images instead of generating images.

If you want to switch to the latest version, please re-run the installer, selecting the 'update' option. 
Should you notice a bug, please let me know either by mail or by opening an issue. Thanks.

Clickbaitcake, 2019-02-27:
Is there any possibility of getting a _Day View_ similar to that in Outlook? This would really help me keep on top of my day. The moving bar shows the current time and how far into the next meeting block I should be. 
Example screenshot attached. 

What do you think? 

[image]

multimedial, 2019-02-27:
Hello, 

I am from Duesseldorf, and found your project via the Waveshare page when trying to test my e-Ink Paper display. 

I got an issue after installing your code and editing the settings.py file accordingly - the free API key provided from OpenWeatherMap is apparently not entitled to pull down the current weather forecast.

I was also wondering if I could discuss a project I am working on with you via email. You can contact me at info@multimedial.de. 

Looking forward to your answer, and thanks for the code.

aceisace, 2019-02-27:
@Clickbaitcake 
Hi Secprentice, thanks for suggesting a new feature. The daily view is certainly possible but will require some time to finish. I'm estimating it'll take at least a few weeks, but it might be finished faster (or slower) than that.
As the space on the Calendar is almost used up, I can only implement the daily view by replacing the monthly Calendar. Does this sound like a good compromise?
I can also implement the moving bar which shows the current time on the daily view, but it might not be such an good idea for an E-Paper display as more frequent display updates are not good for the Display. If you don't mind an update interval of 15 minutes, I can implement it. But I wonder if it's useful if the time is updated every 15 minutes. If you do more frequent updates than 15 minutes, the Raspberry will be under pretty heavy load (nearly) all the time. Let me know about your opinion.
Also, I've not heard or come across 'exchange connectivity'. Do you mind elaborating a bit more about this? Thanks in advance.

aceisace, 2019-02-27:
@multimedial 
Hi, thanks for your interest in the E-Paper software, for your comment and your compliment :)
You did mention about an issue where your key was not entitled to get weather _forecasts_ .

As there is no error message attached, it's a bit difficult finding out where exactly the issue lies. But these are the tips I can give you for now:

* Make sure that your api-key is in string format in the settings.py file, for example:
`api_key = "somesecretopenweathermapapikey"`
* Free api-keys are not entitled to get weather forecasts, only the current weather. Previously, free users could get that access but openweathermap stopped it some time ago. If you're not trying to get the forecast or if you've not modified the E-Paper.py file, most likely it's the next point.
* If you try to request the current weather data too many times in a short period of time, the server will temporarlily block the access. Usually, trying again in a few hours (or the next day) resolves the issue and 
everything will start to work like normal again.

If the error still persists, please paste the error message in your next comment. Then I can try to debug the issue more specifically.

Sorry I had to edit your comment. It's usually not a good idea adding personal details in a comment since it's available to everyone. But sure, feel free to contact me for discussing your project via Email (you can find the email address on the README page).

Also, thanks for your opinion. I'll be working on the daily view next. When I make some progress, I'll let you know.

multimedial, 2019-02-27:
@aceisace 
Thanks for getting back to me. The API key issue has vanished in the meantime, I think the key was not active yet when I started my experiments, I got everything to work now and am already tinkering with the code.

I would like to suggest to make the panel modular. There is a similar project, the magic mirror one (https://magicmirror.builders/) where the content to be displayed can be chosen from a variety of modules. 

I will send you an email to your email address.

aceisace, 2019-02-27:
@multimedial 
Glad to hear your issue was fxed :)
Thank you for the suggestion about the MagicMirror2 project. I am familiar with the MagicMirror and use the software on a 42" LCD display at home. I really like the project, the repo and the modular software of the MagicMirror.

I'm also been thinking that it's about time to make the software modular, but unfortunately, there are still ways to go for me since I'm not too familiar with writing classes and modules just yet.

On the other hand, there is a very passionate forker of this repo who is working hard to make the software modular. You may see the progress to his repo here: https://github.com/mgfcf/E-Paper-Calendar-with-iCal-sync-and-live-weather. Once he's finished implementing the modules of this software, I'll be implementing them on the main repo.

Sure, looking forward forward to your mail.

aceisace, 2019-03-04:
@zAm1987 
Hi, it's been a while. I've updated the software so that it can display the generated image much better now. 
By evaluating each pixel of the generated image before sending it to the E-Paper, it's now possible to give each pixel a value of pure white, pure black (and pure red if using the 3-colour display), hence solving the problem with the 'jagged edges'.
I have tested it myself and have pushed the changes to the main repo. May I ask you to test the updated software and give some feedback regarding the displayed image? Thanks in advance.

zAm1987, 2019-03-05:
@aceisace 

You're really putting effort into this project, keep up the good work! ;)

I just used the "update"-method of the installer, modified the settings file again and tried to run E-Paper.py. Somehow it failed, while the old version worked fine with the same settings (albeit the last time I tested it was some weeks ago). Still not working after a reboot, too.

The script output says:
```
`Fetching events from your calendar

Traceback (most recent call last):
  File "E-Paper.py", line 357, in <module>
    main()
  File "E-Paper.py", line 210, in main
    rule=rrulestr(r_string,dtstart=parse(r_start.group(1)))
AttributeError: 'NoneType' object has no attribute 'group'`
```

I'll try a fresh install in the next days to see if maybe the updating process mixed something up...

aceisace, 2019-03-05:
@zAm1987 
Hi, thanks for your compliment and for testing out the new software :)
Are you, by any chance, using a non-google Calendar? The error you're getting seems to be related to recurring events. As this feature was just recently (a few days ago) implemented, it's possible it might not be working as expected. I'd like to apologise for the inconvience. 

In the meanwhile, you can bypass the Error by uncommenting/deleting lines 203-218 and dedenting 219-223 in the E-Paper file so that the section from line 202 (for events in ical.events:) looks like this:
```
                for events in ical.events:
                    if events.begin.date().month == today.month:
                        if int((events.begin).format('D')) not in events_this_month:
                            events_this_month.append(int((events.begin).format('D')))
                    if today <= events.begin.date() <= time_span:
                        upcoming.append({'date':events.begin.format('YYYY MM DD'), 'event':events.name})
```
Or, as an easier alternative, you can just use the attached 'E-Paper-mod.py' file, unzip it and replace it with the existing one (in /home/pi/E-Paper.Master/Calendar/). Don't forget to change the name of the file to E-Paper.py.

**Attached file:**
[E-Paper-mod.zip](https://github.com/aceisace/E-Paper-Calendar-with-iCal-sync-and-live-weather/files/2933423/E-Paper-mod.zip)

zAm1987, 2019-03-06:
@aceisace
I'm using Google Calendar, so this should not be the problem.

I deleted the specified lines in the E-Paper.py file and now it works. The text and the icons are crisp and sharp, I think even sharper than before the switch to the jpgs. Looks great!

aceisace, 2019-03-06:
@zAm1987 
I'll take a deeper look at the cause of the error and for a way to fix it. But at least you can use the new software now.
I'm glad to hear that the icons and text look much more readable now :)
Now that this problem is solved, I can finish the RSS-feedparser in the next few days. You can look forward to that as well.

arustleund, 2019-03-11:
Hi there! I've encountered another bug in the ics library... I have an alarm trigger like this:
`TRIGGER:P0D`
Apparently the ics library can't handle this, it expects a + or a - in front of the P. I ran this through an ICS validator and it passed, so I think this is valid iCal.
Since we don't really need the alarms anyway, could we just remove all alarms from the calendar before sending it on to the ics library? This would circumvent any potential VALARM bugs in the ics library.

aceisace, 2019-03-11:
@arustleund 
Hi arustleund. Thank you for your interest in this software and for providing the bug-fix for invalid alarm triggers (for the ics.py library). As alarms are not supported by this software anyway, it would make sense to remove them from the parsed iCalendar if they are causing bugs.

Are you getting this error while using Google Calendar? I've not seen this type of error so far so it would be helpful with debugging.

I'll take a look at the code you sugegsted and will implement it as soon as I can. It will take at least some days until I can work on this as I'm finishing off the RSS-feed parser.

arustleund, 2019-03-11:
Yes, I’m using Google Calendar. I also confirmed by reading the specification that this is a valid duration value.

aceisace, 2019-03-11:
@arustleund 
Thanks for the info. I'll start with the implementation soon.

arustleund, 2019-03-11:
@aceisace No problem. I just realized a slight optimization of my code would be to do the second (END) "find" call starting at beginAlarmIndex. Also up to you if you want to be more defensive about the possibility of a missing END:VALARM (which would be invalid ics).

Clickbaitcake, 2019-03-13:
It might be a good idea to start breaking out issues into different threads. I have never seen a github repo with multiple issues inside one giant issue. What do you guys think?

aceisace, 2019-03-13:
@Clickbaitcake 
Hi, thanks for your suggestion. I don't mind anyone opening up an issue for any reason. As you can see, there were several other issues which were closed after being resolved.

The reason I created this 'giant' issue to provide an option to formally share suggestions, discuss features and give feedback to the author (myself) about the software. Compared to the 'normal' issues you know from other Github repos, I wanted users to be actively involved in the development of the software by writing comments in this issue.

If you'd like, you can open up another issue for the 'Day View' feature you suggested.

aceisace, 2019-03-17:
@arustleund 
Hi, I've implemented the bugfix you suggested in your comment earlier along with some other improvements. As I did not originally have any issues due to `TRIGGER:P0D` in my iCalendar, may I ask you to test the new script to see if it works as expected? Thanks in advance.

arustleund, 2019-03-17:
Looks good, although your fix_e_1 fix_e_2 bug fix lines are no longer necessary and can be removed.

Ahbrown41, 2019-03-17:
New code is working well with my calendars!  The only issue is my screen appears like it is ghosted and the lines are not fully drawn..  Hope it is not suddenly my ePaper, I upgraded using the shell script method.  Any ideas?

aceisace, 2019-03-17:
@arustleund 
Thanks for testing, for your code form earlier and for your feedback. I've uncommented those lines in question (fix_e_1 & fix_e_2) in the E-Paper.py file. If you have any more suggestions, feel free to mention it here :)

aceisace, 2019-03-17:
@Ahbrown41 
Hi Alex, thank you very much for testing out the latest version. I just realised a bug this morning in the E-Paper file where the timer was not working as expected. I apologise for that. I've fixed it earlier today so that may be the cause for the ghosting. You can run just the calibration module a few times to fix the ghosting like this: `python3.5 /home/pi/E-Paper-Master/Calendar/calibration.py` and update the software so the bug is gone.

aceisace, 2019-03-19:
@multimedial 
@Clickbaitcake 
Hi, I've not yet had the chance to start with the 'Daily View' feature just yet, but I'll start with that soon. In the meanwhile, if you want to use the Daily view feature asap, you may want to check out the [fork from mgfcf](https://github.com/mgfcf/E-Paper-Calendar), where this feature has been implemented. On this fork, I'm estimating that it'll take a few weeks at least until this feature has been implemented.

mauricecyril, 2019-03-19:
Hi aceisace, thank you so much for making this available. I tried mgfcf's fork of your repository using the Portrait Day List view. The display starts to render however the loop gets the following error:

'''

_________Starting new loop___________
Date: Tue 19 Mar 19, time: 00:22

Fetching weather information from open weather map
Fetching events from your calendar
Fetching posts from your rss-feeds

Starting to render
Converting image to data and sending it to the display
This may take a while...

Data sent successfully
Powering off the E-Paper until the next loop

1 of 1 rendered
=> Finished rendering

Traceback (most recent call last):
  File "/home/pi/E-Paper-Master/Calendar/E-Paper.py", line 94, in <module>
    main()
  File "/home/pi/E-Paper-Master/Calendar/E-Paper.py", line 87, in main
    sleep_time = loop_timer.time_until_next()
  File "/home/pi/E-Paper-Master/Calendar/LoopTimer.py", line 32, in time_until_next
    interval_duration = timedelta(minutes=self.interval)
TypeError: unsupported type for timedelta minutes component: str''

aceisace, 2019-03-19:
@mauricecyril 
Hi, since the fork of @mgfcf is a lot different from this one, it's difficult for me to debug any errors. From what I understand about the error, it seems that the file 'LoopTimer.py' in the Calendar folder, which handles update-timings requires integers, not strings. You may want to give the author of that fork (mgfcf) a shout about this error. 

To make it easier to seperate issues between the 2 versions (this one and the fork), I'll open an issue just for that fork.

aceisace, 2019-03-20:
Since the development in the last few weeks was quite fast, I thought it would be nice to share the updates with everyone:

* **Improved image quality:** Before the image is sent to the E-Paper display, a set of operations are done to the generated image to ensure it can be displayed correctly. The result is an image with crisp colours. Despite the improvement, it takes less than a second to process the entire image.
* **RSS-Feeds:** RSS-feeds are now supported! You can now enter URLs of RSS-feeds in the settings.py file and display them on the E-Paper.
* **Dynamic content below the Calendar:** The space below the monthly calendar is utilised more efficiently. On months where there is more space available below the calendar, more feeds/events will be displayed.
* **Custom Display Update Intervals:** You can now specify how frequent you want to update the E-Paper Display. Options include updates every 15 minutes, 20 Minutes, 30 mins or  1 hour. Please keep in mind with more frequent updates, the chances of ghosting increase as well.
* **Too long event names:** Previously, too long event names caused errors in the software. This has been fixed recently by chopping off letters from the end of the full event name until it fits. (Credit to Hubert)
* **Speed improvements:** Previously, it took around 20 seconds to generate the image. It now takes less than 5 seconds to generate the full image. The time taken to display the image (which takes nearly 1 minute) is still the same, but there is development ongoing on this part.
* **Optional features:** You can now choose what to display below the monthly calendar. Currently there are 2 options, event names and RSS-feeds.
* **Calibration hours:** An option has been added to the settings.py file to execute the calibration of the Display at given hours. You can now choose hours that you find most convenient.

There are also a lot more improvements than mentioned here, particularly in the back-end. But since these are less significant for normal users, they have been left out. 

If you want to use the latest version of the software, you can re-run the Installer from the README and select the update option. More details on that can be found in the README.

Should you encounter any bugs, please report them (via an issue or email) so they can be fixed. Thanks for your help :)

mauricecyril, 2019-03-20:
Amazing update aceisace. Would it be possible to have an option for a pure dynamic agenda view? For example:  https://goo.gl/images/XmBieY

aceisace, 2019-03-21:
@mauricecyril 
Hi, I'm glad to hear you like the updates :)

A dynamic agenda is certainly possible, but will take a lot of time to develop. It's definitely a feature I'd like to implement as soon as I can, but there are some issues I'm working on.

Agenda views and other frequently updated views (for example a Day View) require several updates per hour (a good value would be every 5 minutes). With the current conditions, the Raspberry Pi will be under a lot of stress nearly all the time, which can lead to software crashes. The main reason is that the script from waveshare, that converts the image to data for the E-Paper, is very slow and requires high CPU usage. Basically, it's not really optimised for speed and is inefficient.

To 'unlock' more frequent updates without too heavy CPU load, I'm working on re-writing those functions. As soon as I'm done with that, I'll start working on different views, including the Agenda View.

mauricecyril, 2019-03-21:
Thanks so much aceisace for this amazing project and for all the features you've planned :) 

For the Agenda view would it still be able to pull only once every hour (to preserve the life of the screen) instead of 5 mins? 

For the eink screen the less frequent refreshes that still provide a view of the upcoming agenda items would still work. Ie. Agenda items refreshed once an hour but populate any updates between the last refresh and fill the screen with as many agenda items as possible (ie. Maximize screen realestate).

aceisace, 2019-03-21:
@mauricecyril 
You're welcome :)

Yes, it should be possible to update the Display once an hour, instead of 5 minutes (by specifying the update-interval in the settings.py file). But according to the Datasheet of the manufacturer, (see here: #1 ), the lifetime of the E-Paper Display is 5 years or 1 000 000 updates, whichever is reached first. It basically comes down to one update every 5 minutes, so it shouldn't affect the lifetime of the E-Paper.

I've thought of an agenda view design which utilises a large proportion of the screen and therefore does not need to be refreshed that often, but for the Day View feature suggested by multimedial and Clickbaitcake and some other users, more frequent updates are required.

Once I have designed the templates for the new views and tested them, I'll post an update.

Clickbaitcake, 2019-03-21:
@aceisace I think that a 15 or 30min refresh for day view would be more than enough, at least in my environment. I'm sure nobody's calendar moves around that much?  

Earlier when I talked about Exchange I ment connectivity to a Microsoft Outlook mail server backend with thier calendar system. They use a service called A[ctiveSync ](https://docs.microsoft.com/en-us/previous-versions/office/developer/exchange-server-interoperability-guidance/hh352638(v%3Dexchg.140))to pull calendar info down to iPhones etc. [Its implemented quite well in a similar project to yours](https://github.com/danxfisher/MeetEasier).  Is this something that could be made possible in your system?

Lastly, thank you very much for the hard work here, I haven't seen this many projects with such a dedicated developer. Do you take any donations?

aceisace, 2019-03-21:
@Clickbaitcake 
@multimedial 
Hi, the next feature I'm working on is a more detailed view of the Calendar (for example a Day View or as suggested by @mauricecyril, an Agenda View). Both of these are pretty similar in the sense that they show event times and names. In my opinion, the suggestion from the Agenda View is slightly better as it utilises space a bit more efficiently (which is spare on the E-Paper) and does not require too frequent updates either. If possible, I'd like to stick to one View, but if there is a need for both views, I'll do both. Before I start, I wanted to get your opinions on this matter. Which View do you think it is more suitable for the E-Paper?

aceisace, 2019-03-21:
@Clickbaitcake 
Thanks for your opinion on the update-interval matter. I'll leave the udate-interval as it is then (ranging from 15 mins to 1 hour). Manual updates are not not yet implemented, but if there are more users that find this useful, I'll implement this feature as well.

In regards to the Exchange connectivity, I've done a small research about it and now have an idea of what it is. The project that you mentioned, which handles this very well, is written mainly in Javascript, while this software is based purely on Python(3.x). As a result, it seems difficult to use that software for fetching, but I have some good news too.

From what I understood so far, you're trying to sync your Calendar from your Exchange account. If this is the only thing you need, you can get create an export link for your Calendar, which will be in iCal format and is therefore 'readable' by the software. [These instructions ](https://help.emplify.com/hc/en-us/articles/229406887-How-do-I-get-an-iCal-feed-from-Microsoft-Exchange-) show how to export the Calendar from Microsoft exchange.

If you need to do some more complex and advanced operations related to Microsoft Exchange, there are some good python based libraries which handle these: [exchangelib](https://github.com/ecederstrand/exchangelib) and [pyActiveSync](https://github.com/solbirn/pyActiveSync). Does this answer your question?

Lastly, thanks for your compliment :). I'm glad to hear you like the software and its development.
Although I don't ask for donations, of course you can give a donation which would gladly be appreciated. Please use this link if you want to donate: [image]

aceisace, 2019-03-22:
@Ahbrown41 
Hi, I've got some good news. I found the bug that caused the problem with incorrectly rendered images on 3-colour E-Paper displays. It was a typo made a few commits ago that caused this error. This has been fixed now and the code should now work as expected. Could you please test the new update to confirm this? Thanks in advance,

Ahbrown41, 2019-03-24:
Cool thanks!  I will try it this weekend :-)

aceisace, 2019-03-24:
@Ahbrown41 
Thanks. If there are any issues, please let me know so I can fix them.

Jooosch, 2019-03-25:
Hi,
I've got the following errors after installing:
sudo python3.5 /home/pi/E-Paper-Calendar/Calendar/E-Paper.py
Traceback (most recent call last):
  File "/home/pi/E-Paper-Calendar/Calendar/E-Paper.py", line 33, in <module>
    from icon_positions_locations import *
  File "/home/pi/E-Paper-Calendar/Calendar/icon_positions_locations.py", line 19, in <module>
    weekday = im_open(opath+'weekday.png')
  File "/usr/lib/python3/dist-packages/PIL/Image.py", line 2312, in open
    fp = builtins.open(filename, "rb")
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/E-Paper-Master/Calendar/other/weekday.png'

aceisace, 2019-03-25:
@Jooosch 
Hi, thank you for pointing this out. I must have forgotton to change the path in the icons_positions_locations.py file. I'm sorry for that. You can either update using the installer or just change this line in the file ` icon_positions_locations.py`, locacted in `/home/pi/E-Paper-Calendar/Calendar:

`path = '/home/pi/E-Paper-Master/Calendar/'`
to 
`path = '/home/pi/E-Paper-Calendar/Calendar/'`

If there are any more issues, please feel free to report them :+1:

Jooosch, 2019-03-25:
Thanks for the fast reply 👍 Problem Solved

aceisace, 2019-03-25:
@Jooosch 
You're welcome :)

Jooosch, 2019-03-30:
Now I'm getting another error in the E-Paper-err.log file:
Traceback (most recent call last):
  File "/home/pi/E-Paper-Calendar/Calendar/E-Paper.py", line 36, in <module>
    import pyowm
ImportError: No module named 'pyowm'

But pyowm seems to be installed:

sudo pip install pyowm
Requirement already satisfied: pyowm in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: geojson<2.4,>=2.3.0 in /usr/local/lib/python2.7/dist-packages (from pyowm)
Requirement already satisfied: requests<3,>=2.18.2 in /usr/local/lib/python2.7/dist-packages (from pyowm)
Requirement already satisfied: urllib3<1.25,>=1.21.1 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.18.2->pyowm)
Requirement already satisfied: idna<2.9,>=2.5 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.18.2->pyowm)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.18.2->pyowm)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.18.2->pyowm)

Could you please help me again?

aceisace, 2019-03-30:
@jankosso 
Hi Jooosch, sure, I'll gladly help you again :)

I'm not sure why this exact error message is appearing in your E-Paper-err.log file as this type of error should not occur under normal circumstances (i.e. running a non-modified version of the E-Paper.py file and using the SLI (single line installer)).

For starters, please delete the log files like this:
`cd /home/pi/E-Paper-Calendar &&  sudo rm *.log` to delete any outdated errors and logs. Then, please use Python3.x (recommended is 3.5) to run the E-Paper.py script, like this:
`python3.5  /home/pi/E-Paper-Calendar/Calendar/E-Paper.py`.

Since this software is only meant to run on python3.x, the required libraries are installed only for python3.x (with pip3 ), for both, the user pi and sudo. As a result, python2.x will not be able to run the script.

Please try the above suggestions and let me know if it worked or not.

[62 more replies]

closed 2019-05-18 by aceisace