aceinnolab/Inkycal #21: Invalid Alarm Action

Ahbrown41, 2019-02-09:
Hello,
I have the newest software, the new version looks great, I love that I can now see the upcoming calendar items and the new layout is nice!

I am however struggling with getting the iCal feeds to work.  Any of my private ones appear to generate messages like below.  Do you have any ideas on how to stop these?

If I grep the ICS I see the following actions:

ACTION:AUDIO
ACTION:DISPLAY
ACTION:EMAIL
**ACTION:NONE**

Looking here, it looks like there is a workaround that is non ideal but possible.  Appears to not work at first try but will experiment.

https://github.com/C4ptainCrunch/ics.py/issues/127

Thoughts?

_________Starting new loop___________

Date: Sat 9 Feb 19, time: 07:32

Connecting to Openweathermap API servers...
weather data:
Temperature: 24 °C
Humidity: 45%
Icon code: 01d
weather-icon name: wi-day-sunny
Wind speed: 2km/h
Sunrise-time: 7:00
Sunset time: 17:29
Cloudiness: 1%
Weather description: Clear

Fetching events from your calendar

Traceback (most recent call last):
  File "/home/pi/E-Paper-Master/Calendar/E-Paper.py", line 282, in <module>
    main()
  File "/home/pi/E-Paper-Master/Calendar/E-Paper.py", line 196, in main
    ical = Calendar(urlopen(icalendars).read().decode())
  File "/home/pi/.local/lib/python3.5/site-packages/ics/icalendar.py", line 69, in __init__
    self._populate(container[0])  # Use first calendar
  File "/home/pi/.local/lib/python3.5/site-packages/ics/component.py", line 60, in _populate
    extractor.function(self, lines)  # Send a list or empty list
  File "/home/pi/.local/lib/python3.5/site-packages/ics/icalendar.py", line 214, in events
    calendar.events = list(map(event_factory, lines))
  File "/home/pi/.local/lib/python3.5/site-packages/ics/icalendar.py", line 213, in event_factory
    return Event._from_container(x, tz=calendar._timezones)
  File "/home/pi/.local/lib/python3.5/site-packages/ics/component.py", line 32, in _from_container
    k._populate(container)
  File "/home/pi/.local/lib/python3.5/site-packages/ics/component.py", line 60, in _populate
    extractor.function(self, lines)  # Send a list or empty list
  File "/home/pi/.local/lib/python3.5/site-packages/ics/event.py", line 497, in alarms
    event.alarms = list(map(alarm_factory, lines))
  File "/home/pi/.local/lib/python3.5/site-packages/ics/event.py", line 495, in alarm_factory
    af = AlarmFactory.get_type_from_container(x)
  File "/home/pi/.local/lib/python3.5/site-packages/ics/alarm.py", line 45, in get_type_from_container
    return AlarmFactory.get_type_from_action(action_type.value)
  File "/home/pi/.local/lib/python3.5/site-packages/ics/alarm.py", line 36, in get_type_from_action
    raise ValueError('Invalid alarm action')
ValueError: Invalid alarm action

aceisace, 2019-02-09:
@Ahbrown41
Hi Alex,
thank you for your interest in this software and for the compliments of the new version :)

I need to take a deeper look to find out where exactly the issue lies. I‘ll get back to this on Monday as I‘m not at home at the moment. On Monday, I‘ll let you know more about it. 

For now, could you please tell me what action you‘re using as triggers for events? Thanks in advance

Ahbrown41, 2019-02-10:
It appears  NONE throws it off.  I will look for some examples that are not private :-)

aceisace, 2019-02-10:
@Ahbrown41 
Thanks for your help. It seems there are quite a few issues with ics.py parsing iCalendar URLs. Although there are fixes available, it would maybe be better to switch to a updated or more advanced library, like the icalendar one ([see here](https://github.com/collective/icalendar)). What's your opinion?

aceisace, 2019-02-11:
@Ahbrown41 
I've more or less understood where the errors are coming from. I'll try finding a fix for this issue and add it in the E-Paper.py file. On the other hand, if you manage to find a way that prevents these errors, please share it here. Thanks in advance

aceisace, 2019-02-11:
@Ahbrown41 
OK, I've tested out a few things. For now, this is the solution that works:
Could you please try to see if the error from your icalendar persists when using the below snippet? Thanks.
```
from ics import Calendar, Event
from urllib.requests import urlopen

ical_urls = [
"iCalendar-url"
]

elist = [] #create empty event list
for icalendars in ical_urls:
    decode = str(urlopen(icalendars).read().decode()) #create a string first
    #fix errors related to invalid alarm action
    fix_e = decode.replace('BEGIN:VALARM\r\nACTION:NONE','BEGIN:VALARM\r\nACTION:DISPLAY\r\nDESCRIPTION:')
    ical = Calendar(fix_e)
    for events in ical.events:
        if(time.now().strftime('%Y-%-m') == (events.begin).format('YYYY-M')):
            elist.append(events.name+" on " +events.begin.format('D MMM YYYY'))
print(elist)
```

aceisace, 2019-02-14:
I've tested out the snippet from above which 'bypasses' the invalid Alarm action. I've tested the new snippet for a few days and so far, no errors have shown up.  Therefore, I've replaced the old snippet with the new one (from the above comment).

As this issue has been fixed, I'll be closing it now. Should you feel a need to re-open the issue, feel free to do so.

closed 2019-02-14 by aceisace