aceinnolab/Inkycal #23: Error after installing the software

multimedial, 2019-03-12:
Did I do something wrong?

I ran the script from the repo page: 

```
...
The install was successful
The programm is set to start at every boot.
Please enter your details in the file settings.py.
If this file is not modified, the programm will not start
To modify the settings file, enter:
nano /home/pi/E-Paper-Master/Calendar/settings.py
You can test if the programm works by typing:
python3.5 /home/pi/E-Paper-Master/Calendar/E-Paper.py
pi@raspberrypi:~ $ python3 E-Paper-Master/Calendar/E-Paper.py
Please install numpy with: sudo apt-get install python-numpy
Traceback (most recent call last):
  File "E-Paper-Master/Calendar/E-Paper.py", line 44, in <module>
    import epd7in5b
  File "/home/pi/E-Paper-Master/Calendar/epd7in5b.py", line 1, in <module>
    import epdif
  File "/home/pi/E-Paper-Master/Calendar/epdif.py", line 12, in <module>
    SPI = spidev.SpiDev(0, 0)
FileNotFoundError: [Errno 2] No such file or directory
pi@raspberrypi:~ $ sudo apt-get install python-numpy
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-numpy is already the newest version (1:1.12.1-3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@raspberrypi:~ $
```

multimedial, 2019-03-12:
That's ok, just wanted to know if I had missed anything.-------- Originalnachricht --------Betreff: Re: [aceisace/E-Paper-Calendar-with-iCal-sync-and-live-weather] Error after installing the software (#23)Von: Ace An: aceisace/E-Paper-Calendar-with-iCal-sync-and-live-weather Cc: multimedial ,Mention @multimedial
Hi. thanks for pointing this out. I'll look into this asap and will fix this. As soon as I find out the cause, I'll let you know.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/aceisace/E-Paper-Calendar-with-iCal-sync-and-live-weather/issues/23#issuecomment-472061888",
"url": "https://github.com/aceisace/E-Paper-Calendar-with-iCal-sync-and-live-weather/issues/23#issuecomment-472061888",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

aceisace, 2019-03-12:
@multimedial 
You didn't miss anything. It was a bug in the installer that caused this type of error. Basically, it should have been `sudo apt-get install python3-numpy`. I've just now fixed all bugs in the installer I could find by testing it myself on Raspbain Stretch Lite. If you select the update or install option in the installer now, it should work as expected.

If you find any more bugs, feel free to report them here. Thanks in advance.

multimedial, 2019-03-12:
Will check. Thanks for your work!

multimedial, 2019-03-12:
Sorry, there is still an issue...

`The install was successful
The programm is set to start at every boot.
Please enter your details in the file settings.py.
If this file is not modified, the programm will not start
To modify the settings file, enter:
nano /home/pi/E-Paper-Master/Calendar/settings.py
You can test if the programm works by typing:
python3.5 /home/pi/E-Paper-Master/Calendar/E-Paper.py
pi@raspberrypi:~ $ cd E-Paper-Master/
pi@raspberrypi:~/E-Paper-Master $ cd Calendar/
pi@raspberrypi:~/E-Paper-Master/Calendar $ python3 E-Paper.py
Traceback (most recent call last):
  File "E-Paper.py", line 44, in <module>
    import epd7in5b
  File "/home/pi/E-Paper-Master/Calendar/epd7in5b.py", line 1, in <module>
    import epdif
  File "/home/pi/E-Paper-Master/Calendar/epdif.py", line 12, in <module>
    SPI = spidev.SpiDev(0, 0)
FileNotFoundError: [Errno 2] No such file or directory
pi@raspberrypi:~/E-Paper-Master/Calendar $ sudo pip3 install spidev
Requirement already satisfied: spidev in /usr/lib/python3/dist-packages
pi@raspberrypi:~/E-Paper-Master/Calendar $`

multimedial, 2019-03-12:
Sorry, there seems still to be an issue...

`
The install was successful
The programm is set to start at every boot.
Please enter your details in the file settings.py.
If this file is not modified, the programm will not start
To modify the settings file, enter:
nano /home/pi/E-Paper-Master/Calendar/settings.py
You can test if the programm works by typing:
python3.5 /home/pi/E-Paper-Master/Calendar/E-Paper.py
pi@raspberrypi:~ $ cd E-Paper-Master/
pi@raspberrypi:~/E-Paper-Master $ cd Calendar/
pi@raspberrypi:~/E-Paper-Master/Calendar $ python3 E-Paper.py
Traceback (most recent call last):
  File "E-Paper.py", line 44, in <module>
    import epd7in5b
  File "/home/pi/E-Paper-Master/Calendar/epd7in5b.py", line 1, in <module>
    import epdif
  File "/home/pi/E-Paper-Master/Calendar/epdif.py", line 12, in <module>
    SPI = spidev.SpiDev(0, 0)
FileNotFoundError: [Errno 2] No such file or directory
pi@raspberrypi:~/E-Paper-Master/Calendar $ sudo pip3 install spidev
Requirement already satisfied: spidev in /usr/lib/python3/dist-packages
pi@raspberrypi:~/E-Paper-Master/Calendar $

`

multimedial, 2019-03-12:
I'd like to point out however that I haven't adapted the settings.py file just yet...

multimedial, 2019-03-12:
Sorry, this isn't an issue any more, I had forgotten to install all libraries for the paper as pointed out py Waveshare.

closed 2019-03-12 by multimedial

aceisace, 2019-03-12:
@multimedial 
I'm glad you could fix the issue yourself. Normally, this software is independet form waveshare so conflicts of that sort should not even exist.
I think in your case, the error:
```
SPI = spidev.SpiDev(0, 0)
FileNotFoundError: [Errno 2] No such file or directory
```
is caused due to the SPI not being activated. For reference purposes, I'll write the fix here anyway:
`sudo sed -i s/#dtparam=spi=on/dtparam=spi=on/ /boot/config.txt`
Running the above command will activate the SPI on the Raspberry and the problem should be fixed.