aceinnolab/Inkycal #24: E-Paper Calendar fork from mgfcf

aceisace, 2019-03-19:
### mgfcf's E-Paper Calendar fork

Amongst the forkers of this software,, mgfcf (Max G.) has been working hard on the E-Paper software refactoring. Apart from some more design options, it's also modular so it's easier to modify.

I am very thankful to Max G. for taking the time on refactoring the E-Paper software. But since there is hardly any 'perfect' software, bugs and errors can still occur. The aim of this issue is to provide a way to interface with mgfcf (for bugs, errros, etc.).

As usual, nothing and nobody is perfect, but we can strive to get closer towards that. Please post your comments in a friendly way and remember to be nice. Thanks for your understanding.

mauricecyril, 2019-03-19:
Thanks Aceisace. Hi Max G.

I tried the Portrait Day List view and display starts to render however the loop gets the following error which stops the loop:

'''
Starting new loop__
Date: Tue 19 Mar 19, time: 00:22

Fetching weather information from open weather map
Fetching events from your calendar
Fetching posts from your rss-feeds

Starting to render
Converting image to data and sending it to the display
This may take a while...

Data sent successfully
Powering off the E-Paper until the next loop

1 of 1 rendered
=> Finished rendering

Traceback (most recent call last):
File "/home/pi/E-Paper-Master/Calendar/E-Paper.py", line 94, in 
main()
File "/home/pi/E-Paper-Master/Calendar/E-Paper.py", line 87, in main
sleep_time = loop_timer.time_until_next()
File "/home/pi/E-Paper-Master/Calendar/LoopTimer.py", line 32, in time_until_next
interval_duration = timedelta(minutes=self.interval)
TypeError: unsupported type for timedelta minutes component: str

'''

mgfcf, 2019-03-19:
Hey mauricecyril,

as aceisace already pointed out, it seems to be an issue with the type of the update_interval. The LoopTimer works with integers. It seems, that you were trying to use a string, which lead to some type errors.

You could just removed the quotations from your update_interval (e.g. "15" => 15) or use the updated version, in which I provided a solution, that converts the update_interval to a string and than back to an integer, to make it independent of the given type.

If you ancounter any more issues just let me know.

mauricecyril, 2019-03-20:
Thank you so much. I removed the quotations in the settings.py for the duration and calendar loop works now.

bbence84, 2019-05-03:
This is a really an awesome project, I have already set the hardware and the software up, and it works perfectly! I have couple of questions though:
1. I would like to translate the interface to my language (Hungarian). Are the texts in the mgfcf fork coming from system texts? I have tried to set the language to 'hu', and even edited the TextFormatter.py to add a new item ('hu') to the two list of strings. But still does not work. What I am doing wrong?

2. Anyone has experience running this software with a Pi Zero W on a battery? How long it's supposed to last on an average power bank?

(I had to also replace the font, because the provided otf font was missing a few special characters in our language, e.g. ű).

Thanks!

aceisace, 2019-05-03:
@bbence84 
Thanks for your interest in this project :).

About your questions, @mgfcf can answer your first question in much more detail than me. If you are interested, you can help with the translation on the main branch by providing the following:
* Full month names, 2-/3-letter abbreviations for weekdays.

About the second question, it's a bit hard to give a strict value as there are many factors to consider, so this is just based on my own tests and expierences.

Assuming you're Zero W is running headless and nothing else (USB, mouse, keyboard, monitor) is connected, then a 10000mah power bank can power it for about 20 hours.

mgfcf, 2019-05-03:
Hey @bbence84 ,
welcome to this project :D

The text is not to 100% system text, but most of it. I was already planning to work on that and am able to do it the next days.
You are missing the necessary datetime-locale, which is defined in the Assets.py-file. There you find a dictionary with the language related to the locale. I am now going to add most of the missing locales, so that an update will fix that issue. Otherwise you should be able to add the following item to the dictionary:
`"hu" : "hu.UTF-8"`
The resulting dictionary would look something like this:
`datetime_locals = {
    "de" : "de_DE.UTF-8",
    "en" : "en_US.UTF-8",
    "zh_TW" : "zh_TW.UTF-8",
    "hu" : "hu.UTF-8"
}`
It may happen, that some text or letters will be cut of somehow or simply be to big for certain text fields. Feel free to tell me, so that I can adjust the settings necessary to display all texts for all fonts correctly.

I have experience in running the Pi with just a battery, tho as @aceisace mentioned already, it probably wont be long. You can increase the `update_interval` in the settings.py-file, so that your calendar is updated less frequently, and remove some hours of the `calibrate_hours` list, so that it does not calibrate so often. But bare in mind, that the calibration is there to prevent ghosting, that the image is burning in. That may cause some shadow effects, but a calibration a day will probably still be enough.
If you want to set the `update_interval` bigger than 60 min., you need the set the `run_on_hour` option in the E-Paper.py-file of the loop timer to `False`. You will find that in line 51 of the E-Paper.py-file. Just change the line to:
`loop_timer = LoopTimer(update_interval, run_on_hour=False)`
That option forces the calendar to update on every full hour. Normally it is used to 'align' the updates to full hours.

Should your problem not be fixed or you have other problems or just questions, please post about it in an issue. :)

bbence84, 2019-05-05:
Thanks! Based on your suggestions, I managed to get it to work, with the following observations and limitations:

- In EventListDesign.py, there's a dictionary more_texts that also contains translated texts, so I added 'hu' there too
- The setlocale line in E-Paper.py does not work with 'hu', only if I changed it to:
locale.setlocale(locale.LC_ALL, '') (I also changed the environment variables to hu_HU.utf8 based on https://medium.com/@mohammedhammoud/fix-locale-error-unsupported-locale-setting-f9849eec41f1). I was always getting "locale.Error: unsupported locale setting" if I left it as it is. Seems that the hu locale is not working.
- I also tried to specify the language in OwmForecasts.py: 
self.api = pyowm.OWM..., I have added the language=xxx param to the API initialization, however due to the bug(?) in pyowm, this does not seem to change anything, e.g. I am still getting the short description in english, even though OpenWeathermap seem to support Hungarian already

Thanks!

mgfcf, 2019-05-06:
Thanks for the reply!

- In the next commit I added a simple dictionary to make things easier.
- Now, there is also a more dynamic way of setting the locale, using the locale package to retreive a list of available locales, already as dictionary with the according language key. Now including an option to set the encoding of the datetime locale in the settings file.
- I've looked into the language part of the owm-api and it seems like, that the language paramter only affects the more detailed description, but not the short one. Due to the length I am currently just using the short one. I will look into this a little further and inform you about changes.

I hope that will help a little bit, otherwise, as always, just let me know!

mgfcf, 2019-05-10:
Hey @bbence84 ,
I've now added a simple translator, based on the local dictionary, so that the short_description of the weather can now be translated too. You simply need to add the translations for your language in the format:
`'en' : 'Rain',
'hu' : 'Eső'`

If you want, you can open a pull request after that or simple send me the dictionary file, so that I can add the hungarian translations to the repo.

And as always, feel free to open a new issue, if you have any problems or ideas.

Have a nice day!
Max

bbence84, 2019-05-12:
Awesome, I have ran the update and filled in the Hungarian dictory. Works perfect! The only thing is that my Unicode characters font was overwritten, but I can restore it. :) I will send the dictionary file soon. Thanks again!

bbence84, 2019-05-12:
I have another issue, not sure if it's an issue in the fork or the original or maybe even if it's an issue in my setup. I am using the Waveshare 3 color e ink display with a Pi Zero W. The problem is that the screen does not seem to refresh after a while. I have set it to have it refreshed every 10 minutes, yet the the calendar events stay the same. According to the log, the refreshes are going through successfully:
__________Starting new loop__________
Datetime: 2019-05-12 15:20:00.189323
Fetching weather information from open weather map
Fetching events from your calendar
Fetching posts from your rss-feeds

Starting to render
Converting image to data and sending it to the display
This may take a while...

Data sent successfully
Powering off the E-Paper until the next loop

1 of 1 rendered
=> Finished rendering

... and for the other times I see the same, so seems to work fine.
However again the screen itself does not refresh. Sometimes it refreshes after a couple of hours (I am not sure though, I have not checked it specifically). After a reboot, it always work.

Could it be an issue with my screen or screen driver is glitching? Or what else?

mgfcf, 2019-05-12:
So, if I understand you correctly, the software is running successfully in the background, but your screen does not seem to perfom the described updates?

I had several times issues with the connection. If the screen behaved weird, I just carefully unplug the cables of the display driver and plug them back in again, to make sure that they have full contact.

If you want to see some more information, you can set print_technical_data in the Debug section to `True` and have the time of update printed onto the screen. It isn't beautiful, but rather useful. The shown time should never exceed 10 minutes from now, otherwise the display does not seem to print the latest refreshes.

Another idea is, to have a look on the display driver-board. You can see some switches that should be set to `B` and `0`, at least in my case with the tricolor 7.5" display.

bbence84, 2019-05-12:
Thanks! It works kind of but sometimes does not refresh the screen. It seems to be random. I have enabled the print_technical_data and also the render_to_file. I just caught a case where the design_exported.png contains a recent calendar, whereas the screen itself did not refresh. Then it refreshed the next time. What could be the reason? There's nothing in the error log. Maybe the screen is faulty, but then I don't get why it works sometimes, sometimes not, even if I don't move the frame physically. Maybe the 10 minute frequency is too much?

mgfcf, 2019-05-13:
That may be the case. Try out lower frequency like 20 minutes.

A calibration, to prevent ghosting, takes ca. 15 minutes. That may be the cause for some missing refreshes, but you would still see the display calibrating and, if I remember correctly, you modified the calibration_hours to reduce the calibrations. To shorten the time a calibration needs, you can set the calibration cycles down to one cycle in the `Epd7in5bAdapter` (bwr) / `Epd7in5Adapter` (bw) in the calibrate-method. Just set
```
    def calibrate (self):
        for _ in range(2):
```
to
```
    def calibrate (self):
        for _ in range(1):
```
But be aware, that this reduces the effect of the calibration and ghosting is more likely!

aceisace, 2019-07-24:
## Update
As the branch of @mgfcf is independent of the the main branch, this issue will be closed. If you have any more questions/bug reports/ suggestions/ critics, you can open a seperate issue for that on the fork of @mgfcf . Thanks for understanding.

closed 2019-07-24 by aceisace