aceinnolab/Inkycal #26: Can't get started

ericwass, 2019-04-17:
Hi,
I've followed all the installation instructions but cannot get an image on my Waveshare 7.5in three color display. 

* **Version info;** 
_Version: 1.5
Installer version: 1.5 (Early February 2019)
configuration file: /home/pi/E-Paper-Calendar/Calendar/settings.py
If the time was set correctly, you installed this software on:
Tue Apr 16 16:11:34 PDT 2019_

* **Error output:** 
Traceback (most recent call last):
  File "/home/pi/E-Paper-Calendar/Calendar/E-Paper.py", line 33, in <module>
    from icon_positions_locations import *
  File "/home/pi/E-Paper-Calendar/Calendar/icon_positions_locations.py", line 19, in <module>
    weekday = im_open(opath+'weekday.png')
  File "/usr/local/lib/python3.5/dist-packages/PIL/Image.py", line 2609, in open
    fp = builtins.open(filename, "rb")
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Calendar/Calendar/other/weekday.png'
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                         ne 19, in <module>
    weekday = im_open(opath+'weekday.png')
  File "/usr/local/lib/python3.5/dist-packages/PIL/Image.py", line 2609, in open
    fp = builtins.open(filename, "rb")
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Calendar/Calendar/other/weekday.png'
Traceback (most recent call last):
  File "/home/pi/E-Paper-Calendar/Calendar/E-Paper.py", line 33, in <module>
    from icon_positions_locations import *
  File "/home/pi/E-Paper-Calendar/Calendar/icon_positions_locations.py", line 19, in <module>
    weekday = im_open(opath+'weekday.png')
  File "/usr/local/lib/python3.5/dist-packages/PIL/Image.py", line 2609, in open
    fp = builtins.open(filename, "rb")
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Calendar/Calendar/other/weekday.png'
Traceback (most recent call last):
  File "/home/pi/E-Paper-Calendar/Calendar/E-Paper.py", line 33, in <module>
    from icon_positions_locations import *
  File "/home/pi/E-Paper-Calendar/Calendar/icon_positions_locations.py", line 19, in <module>
    weekday = im_open(opath+'weekday.png')
  File "/usr/local/lib/python3.5/dist-packages/PIL/Image.py", line 2609, in open
    fp = builtins.open(filename, "rb")
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Calendar/Calendar/other/weekday.png'
Traceback (most recent call last):
  File "/home/pi/E-Paper-Calendar/Calendar/E-Paper.py", line 33, in <module>
    from icon_positions_locations import *
  File "/home/pi/E-Paper-Calendar/Calendar/icon_positions_locations.py", line 19, in <module>
    weekday = im_open(opath+'weekday.png')
  File "/usr/local/lib/python3.5/dist-packages/PIL/Image.py", line 2609, in open
    fp = builtins.open(filename, "rb")
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Calendar/Calendar/other/weekday.png'

* **Changes made:** 
No changes

* **Description**: 
_Unable to get it to work at all. This is the result from terminal:
pi@raspberrypi:~/E-Paper-Calendar/Calendar $ sudo python3.5 E-Paper.py
Traceback (most recent call last):
  File "E-Paper.py", line 33, in <module>
    from icon_positions_locations import *
  File "/home/pi/E-Paper-Calendar/Calendar/icon_positions_locations.py", line 19, in <module>
    weekday = im_open(opath+'weekday.png')
  File "/usr/local/lib/python3.5/dist-packages/PIL/Image.py", line 2609, in open
    fp = builtins.open(filename, "rb")
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Calendar/Calendar/other/weekday.png'_

* **Additional Notes:** 
_There may be other issues with the display and/or drivers. I'm unable to get Waveshare demo code to work. Terminal always returns "e-Paper busy"

I have checked and re-checked that I've installed all drivers/python/libraries, etc. I'm very new to this and bought it to learn, so please forgive something I might be missing. Appreciate your work on this._

mgfcf, 2019-04-17:
Hi @ericwass ,
it seems like there is something wrong with you path. I have no complete insight in the code, but it may work, if you change the path in `icon_positions_locations.py` which should be located in `/home/pi/E-Paper-Calendar/Calendar/`.

Change the 12th line
`path = '/home/pi/Calendar/Calendar/`
to
`path = '/home/pi/E-Paper-Calendar/Calendar/`.

aceisace, 2019-04-17:
@ericwass 
Hi, thanks for joining Github, your interest in this software and for notifying about the bug. I'm working on this at the moment so hopefullly it'll be fixed by today. I apologize for the inconvenience so far and will notify you as soon as it's fixed. Thanks for understanding.

ericwass, 2019-04-17:
Thank you. That definitely solved one of the problems. Right now I'm still
getting this:

*pi@raspberrypi*:*~/E-Paper-Calendar/Calendar $* sudo python E-Paper.py

SyntaxError: can not delete variable 'image' referenced in nested scope

But moreover, I'm just concerned that the Pi doesn't see the e-paper
display properly. I've never been able to get an image on there and none of
the Waveshare demo code seems to work. I always get "e-Paper busy"
responses when I run the python scripts.

I really wish I knew what I was doing. I know just enough to be dangerous
but I'm trying to learn.

aceisace, 2019-04-17:
@ericwass 
The bug is causing a bit of a domino-effect. All these errors are related to the incorrect path.
I can suggest these things for now:
* Have you already done the setup from the README? ([Click here to view the setup](https://github.com/aceisace/Inky-Calendar#setup))
* If you've done that already, check what the downloaded folder with the E-Paper software is called. It should be called 'Inky-Calendar', so if isn't, you can try changing that.
* Please also note that the software only supports python3.x (recommended is python3.5). To wun the software with python3, please use the following command:
`python3.5 /home/pi/Inky-Calendar/Calendar/E-Paper.py`

If these don't fix the errors, please use the uninstalling option in the installer to remove the software and run the installer again when it's ready.

mgfcf, 2019-04-17:
Hey @aceisace ,
in the given error-message it looks like the path is `E-Paper-Calendar` instead of `Inky-Calendar`, but the set path in the icon_positions_locations.py-file is actually `Calendar/Calendar/`, which cannot be correct.

The exception also mentions `python3.5` in its path.

ericwass, 2019-04-17:
Thank you for the helpful replies.

I did follow the README set up very carefully. I also followed the
Waveshare setup step by step.

I generally was using python3.5 but then I just tried old python to see if
it made a difference.

The directory was actually called E-Paper-Calendar, not Inky-Calendar. I
have changed it.

Here's what I got:

*pi@raspberrypi*:*~/Inky-Calendar/Calendar $* python3.5
/home/pi/Inky-Calendar/Calendar/E-Paper.py

Traceback (most recent call last):

  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 33, in <module>

    from icon_positions_locations import *

  File "/home/pi/Inky-Calendar/Calendar/icon_positions_locations.py", line
19, in <module>

    weekday = im_open(opath+'weekday.png')

  File "/home/pi/.local/lib/python3.5/site-packages/PIL/Image.py", line
2609, in open

    fp = builtins.open(filename, "rb")

FileNotFoundError: [Errno 2] No such file or directory:
'/home/pi/U=Inky-Calendar/Calendar/other/weekday.png'

ericwass, 2019-04-17:
Sorry, I realize there was a typo in line 19 that I fixed. That cleared up
that error. I then started getting another error:

*pi@raspberrypi*:*~/Inky-Calendar/Calendar $* python3.5
/home/pi/Inky-Calendar/Calendar/E-Paper.py

_________Starting new loop___________

Date: Wed 17 Apr 19 Time: 11:42

Traceback (most recent call last):

  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 445, in <module>

    main()

  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 85, in main

    if hour in calibration_hours:

NameError: name 'calibration_hours' is not defined

I was able to fix that by manual entering it into the settings.py file (the
Settings_Web_UI.html file was not creating it). That fixed that error. I
ran it again and got this:

*pi@raspberrypi*:*~/Inky-Calendar/Calendar $* python3.5
/home/pi/Inky-Calendar/Calendar/E-Paper.py

_________Starting new loop___________

Date: Wed 17 Apr 19 Time: 11:50

Connecting to Openweathermap API servers...

Traceback (most recent call last):

  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 445, in <module>

    main()

  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 151, in main

    observation = owm.weather_at_place(location)

  File
"/home/pi/.local/lib/python3.5/site-packages/pyowm/weatherapi25/owm25.py",
line 231, in weather_at_place

    _, json_data = self._wapi.cacheable_get_json(uri, params=params)

  File
"/home/pi/.local/lib/python3.5/site-packages/pyowm/commons/http_client.py",
line 87, in cacheable_get_json

    status_code, data = self.get_json(uri, params=params, headers=headers)

  File
"/home/pi/.local/lib/python3.5/site-packages/pyowm/commons/http_client.py",
line 32, in get_json

    HttpClient.check_status_code(resp.status_code, resp.text)

  File
"/home/pi/.local/lib/python3.5/site-packages/pyowm/commons/http_client.py",
line 155, in check_status_code

    raise api_response_error.NotFoundError('Unable to find the resource')

pyowm.exceptions.api_response_error.NotFoundError: The searched item was
not found.

Reason: Unable to find the resource

This is definitely the furthest I've gotten but now I'm stuck again. And,
as of yet, I've never seen that Waveshare screen so much as blink. Is that
normal?

aceisace, 2019-04-18:
@ericwass 
Hi again, sorry it took so long but the installer has been updated and tested now.

 I'd recommend you use a fresh install of Raspbain (Lite or Desktop) and use the installer again. If that is too difficult, you can try using the update function of the installer. If something doesn't work right, you can try uninstalling the software (using the installer) and re-installing it after that.

Please let me know if it works. I've done some thorough checks, but there is always a chance for bugs. Thanks in advance.

ericwass, 2019-04-18:
No need to apologize at all. I appreciate everything you’re doing. I’m
prepared to wipe and start over with a fresh install of Raspbian. But can
you clarify one thing for me: do I need to install the Waveshare files
first (or at all)? I went through a whole long procedure of installing the
Waveshare stuff after I installed Inky Calendar thinking it might help but
it didn’t.

aceisace, 2019-04-18:
@ericwass 
Hi, thanks for your kind words :). 

About the waveshare files, please do not install anything from waveshare as it will cause some collisions with this library as they are not compatible in the same environment. One of the reasons is that the waveshare files only support python2.x, while this library uses only python3.x. What's more is that there are differences in how the image data is processed so it will cause issues with displaying the image correctly. So, for this software, you don't need to install anything form waveshare at all.

I made the installer so that there would be no need to follow the somewhat difficult-to-follow instructions from waveshare. It will install the required libraries, driver files (which are different form waveshare) and additional details so you only need to insert your details. 

I've also fixed the web-ui for the settings.py file. You can use that to easily generate the settings.py file :)

ericwass, 2019-04-18:
Ok, did a full wipe and reinstall of Raspbian and followed your
installation instructions exactly. Seem to be having an error with Google
calendar now.  Any thoughts?

*pi@raspberrypi*:*~ $* python3.5 /home/pi/Inky-Calendar/Calendar/E-Paper.py

_________Starting new loop___________

Date: Thu 18 Apr 19 Time: 11:07

Connecting to Openweathermap API servers...

weather data:

Temperature: 75 °C

Humidity: 49%

weather-icon name: wi-cloudy-windy

Wind speed: 2km/h

Sunrise-time: 6:18

Sunset time: 7:25

Cloudiness: 75%

Weather description: Clouds

Fetching events from your calendar

Traceback (most recent call last):

  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 445, in <module>

    main()

  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 237, in main

    ical = Calendar(decode)

  File "/usr/local/lib/python3.5/dist-packages/ics/icalendar.py", line 58,
in __init__

    container = string_to_container(imports)

  File "/usr/local/lib/python3.5/dist-packages/ics/parse.py", line 175, in
string_to_container

    return lines_to_container(txt.splitlines())

  File "/usr/local/lib/python3.5/dist-packages/ics/parse.py", line 171, in
lines_to_container

    return parse(tokenize_line(unfold_lines(lines)))

  File "/usr/local/lib/python3.5/dist-packages/ics/parse.py", line 162, in
parse

    for line in tokenized_lines:

  File "/usr/local/lib/python3.5/dist-packages/ics/parse.py", line 157, in
tokenize_line

    yield ContentLine.parse(line)

  File "/usr/local/lib/python3.5/dist-packages/ics/parse.py", line 65, in
parse

    raise ParseError("No ':' in line '{}'".format(line))

ics.parse.ParseError: No ':' in line '<!DOCTYPE html>'

ericwass, 2019-04-18:
Sorry, scratch that. I provided a bad Google Calendar link. Right now it's
waiting on "Initializing E-Paper Display" I'll let you know if I get past
this point.

ericwass, 2019-04-18:
Yeah, so 20 minutes later it's still stuck on "Initializing E-Paper
Display." I don't know how to diagnose if there's something wrong with it,
or I've got it connected incorrectly or what. Yesterday the Waveshare
python demos kept throwing the error "E-Paper display is busy" but, as you
noted, I had both yours and Waveshare's drivers competing with each other.
You've already been extremely helpful and I know it's not your job to help
me with this issue but if you have any insight as to what might be the
problem or where I should start looking, I would be very grateful (can I
send you some beers or whisky?). I'm attaching a photo just in case you
notice something obviously amiss.

Thanks so much.

aceisace, 2019-04-18:
@ericwass 
Hi, I'm been using the software from today and so far there haven't been any errors. From that, I'd say the problem is not software-related, but rather hardware related. I'm just taking a wild guess, but I'd say it might be a problem with the connections. 

Just in case, could you please attach the following:
* An image of the E-Paper display from the front side (to see if the E-Paper is in tact)
* An image of the connector (extension) of the E-Paper display. This is to see if they're plugged in the right way)
* An image of the driver-board that is plugged in to the GPIO-headers of the Raspberry Pi.
* An image showing the gpio headers from the bottom and the side. (To check if the connections are working)
If you can provide these images, I'd be able to help you more specifically.

You don't need to thank me for helping you with the issue. There was indeed a problem with the installer and I'm responsible for fixing that. Thank you as well for pointing out the errors :)

I'm sorry but I don't drink. But if you want to, you can send a small donation via Paypal (link in the README) to express your thanks :)

ericwass, 2019-04-18:
Yeah, my hunch is that it's hardware related too. I was even wondering if I
was supplying the Pi enough power so I stepped it up but it didn't fix it.
I couldn't find your PayPal link in the README so I tried by using your
email address. But PayPal wants your full name and country. So let me know
that info or send me the link and I'm happy to throw you some cash.

Here are the pictures. The display and Pi are brand new. See anything???
[image: IMG_1644.jpg]
[image: IMG_1646.jpg]
[image: IMG_1647.jpg]
[image: IMG_1648.jpg]
[image: IMG_1649.jpg]
[image: IMG_1650.jpg]

aceisace, 2019-04-18:
@ericwass 
I think the power supply is most likely not the issue. It hardly makes a difference powering the E-Paper display since they consume almost no power (only when updating, and at most a few milliamps).

I'm sorry to tell you this, but the images you uploaded can't be seen in the comment. I can only see the image-filenames but not the actual images themselves. Could you try uploading them directly on github? Thanks in advance.

If you want to donate, you can do so with this link: [image]. I forgot to mention that the link was inside a button on the README page, just like this one.

ericwass, 2019-04-18:
Donation sent!

[image]
[image]
[image]
[image]
[image]
[image]

aceisace, 2019-04-18:
@ericwass 
Thank you very much for your donation :) You're awesome :100: :1st_place_medal:!

From the images, it seems as though all cables are connected correctly. Your GPIO headers are also firm and the driver hat configuration is also correct. That only leaves the E-Paper. It was a bit strange that you could manage to access the E-Paper (via SPI) but not receive a signal from the E-Paper telling the Raspberry it's ready to accept data. This would mean that only some of the connections are working fine.

The third image in your comment shows a crack at the front side of the E-Paper and unfortunately, at a very delicate spot (logic board inside the E-Paper). It would be very difficult to fix such a problem without special equipment so if you've just recently bought the E-Paper, I'd suggest you ask for a replacement.

I'm sorry I couldn't be of more help, but this is the best advice I can give you.

ericwass, 2019-04-18:
You're more than welcome. I appreciate your hard work and will no doubt be
asking you for more help in the future. To be honest, I didn't even notice
that crack on the display. Waveshare has been terrible about responding to
my technical support inquiries. Hopefully they will be better about
arranging a replacement. In any event, it will likely be a couple of weeks
before I have a new one and am ready to give Inky Calendar another try.
We'll talk then.
Best,
Eric

aceisace, 2019-04-18:
@ericwass 
Hi Eric, I'm glad to hear that the cause of this problem could be found. 
You're most welcome to ask for any more help regarding the Inky-Calendar. I hope you can get your E-Paper replaced easily form waveshare. They do indeed offer very slow technical support, but replacements are normally handled much faster than that.
Best Regards, Ace

Since the purpose of this issue has been fulfilled (fixing the installer and certain parts of the software), I'll be closing this issue now. Should you feel a need to re-open the issue, please don't hesitate to do so.

closed 2019-04-18 by aceisace

hapklaar, 2019-09-28:
@ericwass Curious, did you ever get a replacement and did it work afterwards? I'm having similar issues with always getting 'e-Paper busy' repsonses.

ericwass, 2019-09-28:
Yes, the new display fixed it. It was a little bit of a hassle getting
Waveshare to coordinate a replacement but, ultimately, they did and it
works fine.