aceinnolab/Inkycal #28: Error during connection using weather API

anglewf, 2019-04-24:
We get this error when trying to run the code. We have set the API key in the setting.py and not sure what the problem is. Thanks for your help

Calibration complete
Connecting to Openweathermap API servers...
1
Traceback (most recent call last):
  File "E-Paper.py", line 446, in <module>
    main()
  File "E-Paper.py", line 152, in main
    observation = owm.weather_at_place(location)
  File "/usr/local/lib/python3.5/dist-packages/pyowm/weatherapi25/owm25.py", line 231, in weather_at_place
    _, json_data = self._wapi.cacheable_get_json(uri, params=params)
  File "/usr/local/lib/python3.5/dist-packages/pyowm/commons/http_client.py", line 87, in cacheable_get_json
    status_code, data = self.get_json(uri, params=params, headers=headers)
  File "/usr/local/lib/python3.5/dist-packages/pyowm/commons/http_client.py", line 32, in get_json
    HttpClient.check_status_code(resp.status_code, resp.text)
  File "/usr/local/lib/python3.5/dist-packages/pyowm/commons/http_client.py", line 155, in check_status_code
    raise api_response_error.NotFoundError('Unable to find the resource')
pyowm.exceptions.api_response_error.NotFoundError: The searched item was not found.
Reason: Unable to find the resource

aceisace, 2019-04-24:
@anglewf 
Hi, thanks for interest in this software and for opening your first issue :).

About the error you're getting, it seems to be related to a 404 error (i.e. not found error). As this may be caused by a variety of reasons, please try the following first to narrow down the cause,
* If you've just created your api-key from openweathermap, please wait a few hours so that your key is active. In rare cases, it might even take a full day.

* Please check your settings.py file to ensure the syntax of the api_key line is correct. You can open your settings.py file, like so: `nano /home/pi/Inky-Calendar/Calendar/settings.py`. The api_key line should look like this: `api_key = "somesecretapikey"`. Your actual key should be wrapped by this sign `"` on either side. If you were using the settings.py generator (web-ui), just can insert just your key, without these marks (`"`).

* Try to test the connectivity of your Raspberry Pi by typing this in the terminal:
`ping 8.8.8.8`. This will try to 'ping' google to test if it's available. You can terminate the command by typing the [Ctrl] key and [C] key simultaneously.
If you don't receive something similar to `64 bytes from 8.8.8.8: icmp_seq=3 ttl=120 time=26.5 ms`, please post the result in your comment.

Do let me know of the problem was fixed. Thanks in advance.

aceisace, 2019-04-26:
@anglewf 
Hi, has your problem been solved now? 
If yes, please don't forget to close this issue, otherwise, please provide some additional feedback. Thanks in advance.

aceisace, 2019-04-29:
Closing issue due to lack of feedback. If you feel a need to re-open the issue, please feel free to do so.

closed 2019-04-29 by aceisace

aceisace, 2020-01-30:
@anglewf 
If you're still interested in a possible solution, please check the specified location in the settings file. 
[Click here on how to select the correct location](https://github.com/aceisace/Inky-Calendar/wiki/Weather-module-(inkycal_weather)#choosing-the-correct-location)