aceinnolab/Inkycal #30: Old calendar events

ericwass, 2019-05-03:
# Submitting Issues

Finally got a working Waveshare e-ink 3-color display and got Inky Calendar up and running. However, while it is displaying the correct date the list of events on the bottom are very old (some weeks, some years). Also the sunrise/sunset times are very strange and, I assume, connected to a different timezone. Any thoughts?

aceisace, 2019-05-03:
@ericwass 
Hi again ericwass :)
I'm glad to hear you've managed to get the software up and running. 

I apologise for the bug with the outdated events. This bug was reported in issue #20 and a bugfix also being tested right now. It's basically caused by filtering events from let's say May, without restricting the year, leading to events that would normally be in past/future years. As you probably know, a new release (v1.6 with lots of improvements) will be made public next week. You can either wait until then for the bugfix or just replace this line (line 256 in E-Paper.py):

`if events.begin.date().month == today.month:`
with:
`if events.begin.date().month == today.month and events.begin.date().year == today.year:`

The sunrise/sunset timings are fetched from openweathermap api using the given location. Could you please double-check if the location entry (in `settings.py`) contains your city name and country code, like:
`Chicago, US`. If it is not the correct location, you should change it accordingly.

If the problem with the timings still persists or if the location was correct, the only thing that comes to mind is the system time. Have you set the correct timezone according to the [instructions](https://github.com/aceisace/Inky-Calendar#getting-the-raspberry-pi-zero-w-ready) from the README?

ericwass, 2019-05-03:
Sorry, I didn't see that bug report (I actually still don't see it, even with the link you sent). But I made the adjustments to the E-Paper.py line 256. However that doesn't seem to have fixed the problem. Not sure why not. 

The sunrise/sunset bug was, in fact, due to improper system time. I actually had set it but for some reason I guess it reset itself. That is fixed.
[image]

aceisace, 2019-05-04:
@ericwass 
Hi, sorry for the late reply. I'm glad to hear the sunrise/sunset time issue could be fixed. Most likely, the reason the time reset itself was due a system update.

About the bug in the events fetcher, thanks for your feedback. I'm going to take a deeper look of the cause for the problem in the next coming days. You don't need to apologise since the bug could not be fixed until now. 

Was there any difference before and after changing line 256? For example, are certain events (recurring ones) being displayed incorrectly or are all events being displayed incorrectly? Thanks in advance.

ericwass, 2019-05-04:
I don't believe there was a difference before and after I changed line 256.
Today the events are different than yesterday but also very old. Is there a
simple reliable way to stop the Inky-Calendar process? If so, that would
make it easy to troubleshoot. I could stop it, change the line back to the
original and then restart it.
Thanks

aceisace, 2019-05-05:
@ericwass 
Hi, thanks for your feedback. It'll be useful fior finding the error. 

You can stop the auto-start feature of the Inky-Calendar software with this command:
`sudo service supervisor stop E-Paper`
To find out the satus, you can use the following command:
`sudo service supervisor status E-Paper`
And re-enable the auto-start feature with this command:
`sudo service supervisor start E-Paper`

If you make any changes to the main (E-Paper.py) file, the changes won't come into effect until either a restart (`sudo reboot`) or at least restarting supervisor (`sudo service supervisor restart E-Paper`).

ericwass, 2019-05-05:
I've tried starting and stopping the service, rebooting the RPi, etc.
Nothing changes. The weird thing is: most of the events that it's
displaying did not happen on this date in any year. There's a couple from
2014, one from 2010 but only one took place in May and it was May 4th, not
today, May 5th. The one good thing is: one of the events listed (on the
bottom) is accurate and did happen today.

aceisace, 2019-05-06:
@ericwass 
Thanks for the info. It's a bit strange that the events are off not only by the year, but also months. This will be useful for debugging. As soon as there is a fix, I'll let you know.

ericwass, 2019-05-08:
I don't know if this is helpful but the erroneous events that it shows are
the same ones every day. And they were all recurring events at some point
in the past. So my guess is that somehow your parser isn't recognizing that
the recurrence period is past. I could be totally wrong because I don't
know how to code but it's my best guess.

=Eric

ericwass, 2019-05-08:
Hmmm... commenting those lines didn't seem to work (I restarted the service and the RPi)
[image]
All of these are old, recurring events. My actual events aren't there. Also, now there's a new (and very, very old) event at the bottom. (Pig is the cat, by the way)

aceisace, 2019-05-08:
@ericwass 
Thanks for testing the proposed changes. 

If those did not work, could you please test what happens when the entire recurring-events section is missing? I've attached the modified E-Paper.py file so you can just copy it to your system:
[E-Paper.zip](https://github.com/aceisace/Inky-Calendar/files/3158691/E-Paper.zip). 
Thanks for testing.

ericwass, 2019-05-08:
That worked perfectly! Wow. Now I can really use it.

aceisace, 2019-05-09:
@ericwass 
Glad to hear that it works now :). I'll take note of the changes from the attached E-Paper.py file and will implement them in the next release sometime this week. If you notice any more bugs/errors, please feel free to report them via mail or opening an issue.

Thank you very much for your help and for testing out the proposed changes :100:

aceisace, 2019-05-12:
@ericwass 
Hi Eric, just wanted to let you know that the new release is ready for the official launch. Along with the bugfixes, it contains the Agenda-View, support for many more languages and many back-end improvements. Before I make the release official, may I ask you to test out the new software? You can update the software with the installer. 

Please use the settings-web-ui to generated the settings file since the format of the file has changed slightly. Thanks in advance. 

If you notice any problems, issues or errors, please report them in #20.

Since this issue has been resolved and the new release already contains the bugfix, I'll be closing this issue now. If you feel a need to re-open this issue, please feel free to do so.

closed 2019-05-12 by aceisace