aceinnolab/Inkycal #34: Support, Discussion, Suggestions, Critics for release v1.6

aceisace, 2019-05-18:
Here you have an option to chat about improvement ideas regarding release v1.6 (Mid May 2019) of the E-Paper Calendar software.

As usual, nothing and nobody is perfect, but we can strive to get closer towards that. Please post your comments in a friendly way and remember to be nice. Thanks for your understanding.

All improvement ideas are welcome.

zAm1987, 2019-05-20:
I tested the new version today and it's simply great. Even the problematic (second) ical from my wife works now.

There are still some really minor things but these are really nitpicks:
1. Some characters have (H and U so far) have an strange "small side". A Pictures follows...
[
[image]
](url)
2. The weather description ist mit localized. But that's maybe an OWM problem...

Anway, I really like the new version and the agenda view! :)

aceisace, 2019-05-21:
@zAm1987 
Thanks for your feedback about the new release. I'm glad to hear that the software now works with both of your iCalendars and that you like the new agenda view :)

Regarding the slightly less-readable characters, it should get slightly better after some fine-tuning of the image pre-processing. Once I have some satisfying results, I'll let you know so you can test it out.

As for the weather description, owm has 2 weather-description types, a shorter one and a detailed one. Currently, the shorter version is used (due to some lack of space in the top-section), which does not support translations. 
The idea is to first create the required space by re-arranging the weather-section and then displaying the detailed weather description, which can also be displayed in the local language.

You can track the status of these suggestions/ improvements at the **[project page](https://github.com/aceisace/Inky-Calendar/projects/2)**

aceisace, 2019-05-29:
@zAm1987 
I've made some adjustments to the image pre-processing operations so that the text will be displayed slightly better now. The (detailed) weather description is now localised. To make space for the detailed weather description, the sunrise/sunset timings have been 'merged', showing only the next sun-event (e.g. if the sunset is coming next, the sunset time will be displayed and vice versa). 
Due to some issues with the weather-icon images during the image pre-processing, weather icons are now font-based. As a result, some icons might appear slightly different.

You can now test out the updated version of the software by using this installer (from the dev branch):
`bash -c "$(curl -sL https://raw.githubusercontent.com/aceisace/Inky-Calendar/master/Installer-with-debug.sh)"`

Please let me know if the readability of text is better now and whether the weather description is shown in the specified language. Once you can confirm it works, I'll push the changes to the main branch, Thanks in advance.

lasercow5, 2019-06-01:
HI 1.6 looks awesome, thanks for all the hard work! As a suggestion could you add a last updated time to the calendar?

aceisace, 2019-06-02:
@lasercow5 
Hi, thanks for your interest in this software as well. I'm glad to hear you like the new update :)

About the suggestion, a few users have suggested this feature, so I think it's about time to implement this as well. Do you have an idea on how and at which place of the display this info (last updated) should be? Thanks in advance.

lasercow5, 2019-06-02:
It can be somthing as simple as `update = time.strftime("%I") + ':' + time.strftime("%M") + ' ' + time.strftime("%p")` and the bottom right corner seems like a good spot. Or possibly to the right of the month just under the weather section.

aceisace, 2019-06-02:
@lasercow5 
Thanks. I also agree that the place at the bottom right corner seems suitable fot this task. I'll add this on the to-do list of this project and will work on it soon. Once I have tested everything, I'll let you know.

zAm1987, 2019-06-03:
@aceisace 

I tried to run the installer with the update function in order to test your latest changes. Interestingly this option is not accepted by the installer and the output looks like this
`pi@raspberrypi:~ $ bash -c "$(curl -sL https://raw.githubusercontent.com/aceisace/Inky-Calendar/Stable/Installer-with-debug.sh)"

Please select an option from below:
Enter 1  to update the E-Paper software
Enter 2  to install the E-Paper software
Enter 3  to uninstall the E-Paper software
Confirm your selection with [ENTER]
Waiting for input...  invalid number, aborting now
pi@raspberrypi:~ $ 1
-bash: 1: command not found`

In addition, since two days the calendar has a strange behaviour concerning the loaded events. It somehow stops in the middle of the defined timeframe and does not show the last events. But interestingly the stopping point is not the same each day....
I attached a picture to show the behaviour:

[image]

aceisace, 2019-06-03:
@zAm1987 
Thanks for your feedback. I've not made any changes to the Installer in the last few days so it's a bit strange as to why the installer stopped working temporarily, but I'm glad to hear it works now.

I've also not made any changes to the event fetching section in the last few days either. For the agenda view, events that end within the next 22 days are filtered. This value (22 days) is fixed and cannot be changed via the settings file. Then, depending on how much space is available, the calendar will try to display as many events as possible, sorted by their starting date.

I'm not sure as to why this behaviour is occuring, so I'll take a look at it in the next few days. In the meanwhile, if you notice any difference after updating, please let me know. Thanks in advance.

zAm1987, 2019-06-04:
@aceisace
The behaviour of not fetching all Events still occured after the latest update at least for two calibration cycles. This morning everything worked fine.
As I read of an widespread Google server outage, I think it was somehow related to this...

Regarding the latest changes, the characters Like the "H" are rendered much better now and I think the text looks a bit sharper in general, too.

Also the weather description localization Seemann to work fine.

As always, thanks for your effort and keep up the good work! ;)

aceisace, 2019-06-04:
@zAm1987 
I'm glad to hear that the problem with the events was solved automatically. It was most likely related to google's server outage since I could not reproduce the mentioned problem today.

Thanks for your feedback about the readability of text on the display. It's nice to hear that the text is more readable now and that the weather description is being displayed in the set language correctly. 

I'm glad to hear you like the new update. If you have any more improvements or suggestions, please feel free to share them here. Thanks :)

zAm1987, 2019-06-10:
The problem with the events occured again today.
Somehow only the events until friday are fetched and the days after friday are shown empty, despite events existing there.

Also, some events in the timeframe until friday are simply ignored and do not show up on the display.

A reboot did not fix this behaviour and until now I cannot see a pattern concerning the event-type or event-attributes which cause this behaviour...
My only observation is that events with (more) fiesta other than the creator are affected more often.

Do you have a clue?

aceisace, 2019-06-11:
@zAm1987 
It's hard to pinpoint the cause of this error since I could still not reproduce this error on my end. Currently, there are two issues with the parsing of events:
* Recurring events are not currently supported to their difficulty of parsing
* Events which do not end within 21 days from today are not displayed. This means that some multi-day events may be missing on the Calendar
Other than that, all events should have been parsed correctly.

If a reboot could not fix the error, then most likely a modification to the event-parsing snippet is needed.  I'll try to re-write that section in the next coming days and ask you to test out the new version.

To help with debugging, could you please test your iCalendar in the calendar view and see if the problem persists there as well? Thanks in advance.

aceisace, 2019-06-11:
@lasercow5 
Hi again, I've been working on the 'last-update' feature in the past few days. After testing several locations, I found the bottom line more suitable for this purpose. Below is a preview of the generated image:
[image]

Does this look satisfactory or do you think something could be improved? Thanks in advance.

zAm1987, 2019-06-11:
@aceisace 
I switched to calendar view and the circles / rectangles are drawn for the days with events that are not shown in the agenda view. The events section in the calendar view however shows events from the last days and not from the next days...

Interestingly, one (weekly) recurring event is displayed in the agenda view despite not being supported

aceisace, 2019-06-11:
@zAm1987 
Thanks for your feedback. If the event-markers are shown correctly on the Calendar, the event parsing code must at least be filtering the events correctly.

To test the event-parsing snippet in more detail, I created some dummy events on my iCalendar and compared the events on my Inky-Calendar. Below are the results:

1) On Google Calendar:
[image]
2) On the Inky-Calendar:
[image]

As you can see, all events are parsed and sorted correctly, without any differences. When choosing 'Events' for the bottom section, the result is the same. It's a bit unclear as to why the issue is happening on your end.

Could you try running the modified version of the E-Paper.py file from below? It will print out two lists, one which contains all filtered events and one which shows the generated list for the agenda view. Thanks in advance.
[E-Paper.zip](https://github.com/aceisace/Inky-Calendar/files/3278940/E-Paper.zip)

zAm1987, 2019-06-12:
@aceisace
I will try the modified version tomorrow.

Maybe it ist somehow related to the number of events, because right now it displays one more event, which was not shown before. It seems like it shows only ~14 events, which usually is a three to four day timeframe...

aceisace, 2019-06-12:
@zAm1987 
As the space on the E-Paper is limited, the Agenda view will display as many lines as possible to fill the empty space, either with events or dates, whichever comes first in chronological order. 

If the bottom section is not set to empty, there are fewer lines available (15) for the Agenda view. When the bottom-section is not set, there are 21 lines available.
Could you try using the Agenda-view as the middle section and use nothing for the bottom section? That way, the Agenda-view will utilise the entire display for events/dates.

zAm1987, 2019-06-12:
@aceisace 
I ran your modified version an the printed out list looks like this:

```
Date: Wed 12 Jun 19 Time: 20:13

Connecting to Openweathermap API servers...
weather data:
Temperature: 18 °C
Humidity: 49 %
Wind speed: 4 km/h
Sunrise-time: 05:14
Sunset time: 21:34
Cloudiness: 20 %
Weather description: Ein paar Wolken

sunset coming next
Internet connection test passed

Fetching events from your calendar

Generated list of all filtered events. Event-timings are given in UTC.
List of all filtered events:
 [<all-day Event 'Wäldchestag' begin:2019-06-08 end:2019-06-11>, <all-day Event 'Wäldchestag' begin:2019-06-09 end:2019-06-10>, <all-day Event 'Urlaub Eileen' begin:2019-06-10 end:2019-06-21>, <all-day Event 'Wäldchestag' 2019-06-10>, <all-day Event 'Wäldchestag' 2019-06-11>, <all-day Event 'Urlaub Eileen' begin:2019-06-11 end:2019-06-20>, <all-day Event 'Urlaub Eileen' begin:2019-06-12 end:2019-06-20>, <Event 'Chor Soulfood' begin:2019-06-12T18:50:00+02:00 end:2019-06-12T21:30:00+02:00>, <all-day Event 'Urlaub Eileen' begin:2019-06-13 end:2019-06-20>, <Event 'Augenarzt Frankfurt | Dr. Julia Seiler - Augendruck' begin:2019-06-13T10:15:00+00:00 end:2019-06-13T10:30:00+00:00>, <Event 'Zahnarzt 🦷Kontrolle und Reinigung' b egin:2019-06-13T14:30:00+00:00 end:2019-06-13T15:30:00+00:00>, <Event 'Hausverwaltung' begin:2019-06-13T16:00:00+00:00 end:2019-06-13T18:00:00+00:00>, <Event 'FKV mit Patrik' begin:2019-06-13T16:15:00+00:00 end:2019-06-13T20:00:00+00:00>, <Event 'Fahrt von Bad Vilbel nach Frankfurt(M)Hauptwache' begin:2019-06-13T17:34:00+00:00 end:2019-06-13T18:00:00+00:00>, <all-day Event 'Französischer Markt' begin:2019-06-14 end:2019-06-16>, <all-day Event 'Lamboyfest' begin:2019-06-14 end:2019-06-16>, <all-day Event 'Urlaub Eileen' begin:2019-06-14 end:2019-06-20>, <Event 'Frühstück mit Rudolphs' begin:2019-06-14T09:00:00+00:00 end:2019-06-14T12:00:00+00:00>, <Event 'Night of Science 2019' begin:2019-06-14T15:00:00+00:00 end:2019-06-15T04:00:00+00:00>, <all-day Event 'Schweizer Straßenfest' begin:2019-06-15 end:2019-06-16>, <all-day Event 'Französischer Markt' 2019-06-15>, <all-day Event 'Lamboyfest' 2019-06-15>, <all-day Event 'Urlaub Eileen' begin:2019-06-15 end:2019-06-20>, <Event 'Leichte Tomatensuppe mit Reis' begin:2019-06-15T14:30:00+00:00 end:2019-06-15T16:30:00+00:00>, <all-day Event 'Straßenfest' 2019-06-16>, <all-day Event 'Dottenfelder Hoffest' 2019-06-16>, <all-day Event 'Französischer Markt' 2019-06-16>, <all-day Event 'Lamboyfest' 2019-06-16>, <all-day Event 'Schweizer Straßenfest' 2019-06-16>, <all-day Event 'Urlaub Eileen' begin:2019-06-16 end:2019-06-20>, <all-day Event 'Kaiserstuhl erkunden' begin:2019-06-17 end:2019-06-23>, <all-day Event 'Urlaub Eileen' begin:2019-06-17 end:2019-06-20>, <all-day Event 'Kaiserstuhl erkunden' begin:2019-06-18 end:2019-06-22>, <all-day Event 'Urlaub Eileen' begin:2019-06-18 end:2019-06-20>, <all-day Event 'Kaiserstuhl erkunden' begin:2019-06-19 end:2019-06-22>, <all-day Event 'Urlaub Eileen' begin:2019-06-19 end:2019-06-20>, <all-day Event 'Kaiserstuhl erkunden' begin:2019-06-20 end:2019-06-22>, <all-day Event 'Urlaub Eileen' 2019-06-20>, <all-day Event 'Kaiserstuhl erkunden' begin:2019-06-21 end:2019-06-22>, <all-day Event 'Urlaub Eileen' 2019-06-21>, <all-day Event 'Kaiserstuhl erkunden' 2019-06-22>, <Event '? Sommerfeier Siersleben' begin:2019-06-22T14:00:00+00:00 end:2019-06-22T18:00:00+00:00>, <all-day Event 'Kaiserstuhl erkunden' 2019-06-23>, <Event 'Nagelkosmetik Hände & Füße' begin:2019-06-24T16:30:00+00:00 end:2019-06-24T18:00:00+00:00>, <Event '[Einladung] Frankfurter Get together - Event am 25.06.2019' begin:2019-06-25T14:00:00+00:00 end:2019-06-25T19:00:00+00:00>, <Event '[Einladung] Frankfurter Get together - Event am 25.06.2019' begin:2019-06-25T14:00:00+00:00 end:2019-06-25T19:00:00+00:00>, <Event 'Eigentümerversammlung' begin:2019-06-25T17:00:00+00:00 end:2019-06-25T19:30:00+00:00>, <Event 'Eigentümerversammlung' begin:2019-06-25T17:00:00+00:00 end:2019-06-25T19:30:00+00:00>, <Event '(?) CDU Kommunikationskonzept' begin:2019-06-26T17:00:00+00:00 end:2019-06-26T19:00:00+00:00>, <all-day Event 'Hanauer Apfelweinfest' begin:2019-06-28 end:2019-06-30>, <Event 'Kosmetik Gesicht' begin:2019-06-28T13:30:00+00:00 end:2019-06-28T16:00:00+00:00>, <all-day Event 'Heddernheimer Straßenfest' 2019-06-29>, <all-day Event 'Sparkling Vibes Festival' 2019-06-29>, <all-day Event '(?) Konrad Geburtstag' 2019-06-29>, <all-day Event 'Hanauer Apfelweinfest' 2019-06-29>, <Event '30ter Konrad' begin:2019-06-29T16:00:00+00:00 end:2019-06-29T20:00:00+00:00>, <all-day Event 'Ironman' 2019-06-30>, <all-day Event 'Hanauer Apfelweinfest' 2019-06-30>, <Event 'Kosmetik Sugaring' begin:2019-07-01T15:00:00+00:00 end:2019-07-01T16:30:00+00:00>]
Generated the following list for the Agenda View:
 [{'value': 'Mi 12 Jun 19', 'type': 'date'}, {'value': 'Urlaub Eileen', 'type': 'full_day_event'}, {'value': '18:50 Chor Soulfood', 'type': 'timed_event'}, {'value': 'Do 13 Jun 19', 'type': 'date'}, {'value': 'Urlaub Eileen', 'type': 'full_day_event'}, {'value': '12:15 Augenarzt Frankfurt | Dr. Julia Seiler - Augendruck', 'type': 'timed_event'}, {'value': '16:30 Zahnarzt \U0001f9b7Kontrolle und Reinigung', 'type': 'timed_event'}, {'value': '18:00 Hausverwaltung', 'type': 'timed_event'}, {'value': '18:15 FKV mit Patrik', 'type': 'timed_event'}, {'value': '19:34 Fahrt von Bad Vilbel nach Frankfurt(M)Hauptwache', 'type': 'timed_event'}, {'value': 'Fr 14 Jun 19', 'type': 'date' [... cut ...]
[image]

To show the difference I also attached an older image, which goes more into the future, but has less events in general. Thats why I thought that this behaviour has to do with the amount of events.

[image]

Last but not least, here is the last snippet of my settings.py:
```
week_starts_on = "Monday"
events_max_range = "48"
calibration_hours = [0,12,18]
display_colours = "bw"
language = "de"
units = "metric"
hours = "24"
top_section = "Weather"
middle_section = "Agenda"
bottom_section = ""
```

aceisace, 2019-06-13:
@zAm1987 
Hi again. Thanks for your feedback. Thanks for the log with the unfiltered and filtered list. I now have a clue on what is causing this behaviour. You were right that events towards the bottom section have been missing without a reason.
The problem was caused by 2 things:
* ics.py has a bug where full day events are off by 2 days (in the future). The software only fetched events that end within the next 14 / 21 days, leading to some missing events towards the end of the timeframe.
* When creating the agenda view, 2 lists, one containing only dates and the other containing only events are 'merged' and sorted according to date. Seems this part needs some improvements.

I'll work on the solution for this issue in the next few days. Once I have tested it out, I'll let you know so you can test the fixes. Once again, thanks for the report and the log file. You're awesome :)

P.S.: I'll delete the log file from your comment as it contains some personal information.

lasercow5, 2019-06-16:
The last update time look awesome! i've been having this issue with the calendar setting the start of the week to be Sunday `Traceback (most recent call last):
  File "E-Paper.py", line 519, in <module>
    main()
  File "E-Paper.py", line 225, in main
    image.paste(weekday, weekday_pos['pos'+str(now.isoweekday())], weekday)
KeyError: 'pos7'` 
any idea why this is happening?

aceisace, 2019-06-17:
@lasercow5 
I'm glad to hear you like the last-update feature. About the issue with the weekday-start on Sunday, thanks for letting me know. I'll fix that in the next few days. Once it's ready, I'll let you know.

aceisace, 2019-06-18:
@lasercow5 @zAm1987 
As the main file is getting a bit too big to maintain a clear overview, I am going to do some refactoring on the code. This means that the main file will be chunked off into smaller parts, with each part being responsible for a specific task.
Doing so will make further development / debugging a lot faster, but it might take a few weeks to refactor the full code. The mentioned bugs will be debugged during the refactoring. I apologise for the delay with the debugging.

aceisace, 2019-07-23:
## Quick Update
I apologise to everyone for the delay with the new release. Due to a few personal circumstances, the development of this software was slowed down significantly. But now it is back to normal.

A large proportion of the new release is already complete. I'm working on the remaining parts while testing the completed parts. The new release is sheduled for next week and will make any further development a lot easier.

lasercow5, 2019-07-23:
No need for apologies! Thank you for all the hard work. Looking forward to the new release.

aceisace, 2019-07-24:
@lasercow5 
Thanks for your comment :). I'm happy to hear you're looking forward to the new release.

aceisace, 2019-08-13:
##Quick Update
Due to a few personal reasons, I'll be out of country until mid October. I apologise for the delay in the development and the new release. Until I'm back, I'll still offer support for the current release in case there are some minor problems.

dkudlek, 2019-10-03:
@aceisace I did a bit of refactoring and restructuring of your code. It is not as extensive as the version of mgfcf. Just the minimum necessary to suit the needs.

The idea is to support multiple "panels" with "widgets". Sometime in the future you could have some push buttons and switch the panels.  Each panel covers the whole screen and consists of widgets. A widget can have a specifc backend to receive data.

In case you or someone else finds this interesting, you can check it out on my profile.

aceisace, 2019-10-09:
@dkudlek 
Hi David, thanks for the comment and the information. Thank you very much for the time and effort spent on the refactorisation of the current version :)
As I am also refactoring the software, I'll use parts of your fork to handle some of the features I wanted to implement in the next release.

ipaid2win, 2019-10-17:
Hi there,

First off a big thanks for developing the code, works well. I tried to write a own little script at first but failed, my coding skills are not that great I guess.

I noticed that recurring events are not showing up on the Agenda view, only the initial event is in the list. I currently use 2 different calendars and one of them mainly uses recurring events. 
I looked at the code and tried to look for a way to have it push all the events to the screen, but no luck there. (again I'm not really good with python)

I saw that someone else and you mentioned this problem in June, so I guess its still WIP? 
I'll probably just copy/split those recurring events into individual events as they won't change.

Again, big thanks!
Alex

aceisace, 2019-10-18:
@ipaid2win 
Hi Alex, thanks for your interest in this software. I'm glad to hear you like the Inky-Calendar software and that the software is working well :). No need to worry about your coding skills, everyone is a beginner at some point.

The parsing of recurring events is a lot more complicated than normal (fixed-time) events. The Python library used for parsing the iCalendar (ics.py) has some difficulties with the parsing of events. In short, there are 2 ways to parse recurring events with ics.py:
* Using the timespan function to fetch all events in a certain timespan. I did try this but then parsing would take too much time (several minutes) since every single event would have to be scanned.
* Using the beginning (or the ending) time of an event. This approach causes less crashing but it doesn't work well on recurring events (since their beginnging time may be well in the past).

As both approaches have their own problem, a new and better parsing algorithm is required. Someone suggested one way, but it only worked for some users while for others, it would lead to frequent crashing. I've therefore not implemented this feature yet, but I'm working on this for the new release. Once I have a working parser, I'll let you know so you can test it out.

ipaid2win, 2019-10-18:
@aceisace 
Thanks for the reply.

I tried the first way yesterday and it (like you wrote) either took way too long or didnt work at all. It also only looks at new events, which filters out all recurring events beginning in the past. At that point I gave up :).

For now I created a second synced calendar and exported the main calendar with the recurring events. I then used Outlook to import the .ics from the Google Calendar and exported it as .csv. This basically turned the recurring events into individual events. I then imported the csv back into Google Calendar in the second calendar. This "workaround" would be better if Google Calendar supported csv exports right away.
In my case this will probably work just fine as the recurring events have been finalised and are not going to change at all (hopefully) as its my course schedule for university. 

Another thing I noticed was that the documentation refers to a /Calendar folder within /Inky-Calendar. The setup script doesn't create one and having the settings.py file in the main directory (where E-Paper.py also sits) works without a problem. I did get an error log at some point which basically said that  /Calendar/E-Paper.py can't be found, but it still worked. I wasn't really able to reproduce the creation of the error log, but also didn't spend too much time on it as there was no weird behaviour. I guess some part of the code still looks for that folder, maybe only if you manually create it (which I did at first after reading the documentation) and only put the settings.py there but not the E-Paper.py.

Thanks,

Alex

aceisace, 2019-10-18:
@ipaid2win 
Hi Alex, thanks for sharing the workaround on how to use turn recurring events into 'normal' ones. I'll add a note about this on the README. For the new release, I'll see if I can either convert recurring events with rrule into single events or use another library to parse them. 
You are also correct that the documentation is a bit outdated, mainly because the current release was  developed too fast at the beginning. I'll hopefully take care of that in the coming days.

ipaid2win, 2019-10-19:
@aceisace

Obviously exporting the ICS, turning it into an CSV with another program and then importing it again is pretty dirty. 

I think there are some python libraries that should make the conversion ICS -> CSV -> ICS doable on the raspi itself. Until ICS.py is not ready to parse repeating events, this might be a suitable approach. No idea about the performance hit though. Sadly my schedule is completely full (or at least the Inky-Calendar tells me so haha), or else I could give such a conversion program a try.

Best regards,
Alex

aceisace, 2019-10-20:
@ipaid2win 
The conversion from ics to csv and back sounds reasonable if at least the software can display the events correctly (including recurring events). I've had a look at some python-based libraries for this conversion but I could not find any that explicitly supported recurring events. I'll keep searching though.
No need to worry about your shedule. I'm glad to hear that the software is fulfilling the purpose of helping you stay organised :). Best Regards.

aceisace, 2019-10-24:
@surak @mgfcf Hi Alex, Hi Max,
I did some repositioning of the weather section to create some space for the weather forecast. The following image shows the new weather section in planning:
[image]

What do you think about the re-arrangement and what do you think can be improved? Of course other readers can give a comment as well. Thanks!

mgfcf, 2019-10-25:
I really like the look of it! Especially, that there is now information displayed.
The only thing that seems a little crouched are the sunset and -rise times on the right and the humidity, etc. on the left. It intuitively seems like they are part of the hourly forecast, because of the vertical lines on their sides. It may get a little more clear if you could remove them or maybe make them stronger, or somehow differentiate them from the actual vertical lines for the forecast.
But that is just an idea, maybe others have no problems with it. :D

surak, 2019-10-28:
Hi!

I love it.

We could leave it more ample:

Next hour | later today | tomorrow | day after tomorrow|

aceisace, 2019-10-28:
@surak @mgfcf 
Thanks for your precious feedback. I've slightly re-arranged the weather section to give a better overview:
[image]
Since owm provides weather forecasts for every 3 hours, the forecast 'sections' also show the weather in that interval. 

Does anything else need some improvement in the weather section?

surak, 2019-10-28:
This is really great! Is it committed here already?

aceisace, 2019-10-30:
@surak 
The new weather section is not yet committed in this branch. It will be available in the next release.

aceisace, 2019-11-05:
**Good news everyone;** 
The new weather module for the next release (v1.7) is nearly complete. The following image was generated by the new weather module:
[image]
In comparison, the image designed with the graphics software, which was used as the template:
[image]

aceisace, 2019-11-05:
@surak 
Glad you like it :). I did not understand the second part, would you mind elaborating on that a little?

surak, 2019-11-06:
> Glad you like it :). I did not understand the second part, would you mind elaborating on that a little?

You have next "hours" - each column is 3 hours after the previous. My question would be to add days - like tomorrow, day after tomorrow and the next (planning the weekend for example)

zAm1987, 2019-11-06:
> **Good news everyone;**
> The new weather module for the next release (v1.7) is nearly complete. The following image was generated by the new weather module:
> [image]
> In comparison, the image designed with the graphics software, which was used as the template:
> [image]

Looks great, This is a good use of the limited space. Really looking forward to the next release :)

aceisace, 2019-11-06:
@surak 
You do have a point that in some cases, it does make more sense to show the weather forecast of the next day(s), but for now, I'll leave it as is (to decrease complexity). In short, you can't change the time within the settings file, but you can easily change it in the weather module itself, for example:
```
    """Prepare timings for forecasts"""
    in_3h = now.replace(hours = 3)
    in_6h = now.replace(hours =  6)
    in_9h = now.replace(hours = 9)
    in_12h = now.replace(hours =  12)
```
to:
```
    """Prepare timings for forecasts"""
    in_3h = now.replace(days = 1)
    in_6h = now.replace(days =  2)
    in_9h = now.replace(days = 3)
    in_12h = now.replace(days =  4)
```

aceisace, 2019-11-06:
@zAm1987 
Hi, glad you like the new weather section :). 
If there are any suggestions you think that can still be made, do let me know :).

ipaid2win, 2019-11-06:
I really like the design of the new weather module, I think it also uses the space way more effienctly. A true forecast (for the day) is way more useful than only the weather at the time of update! Good job!

aceisace, 2019-11-08:
@ipaid2win 
Hi Alex, thanks for your comment. I'm glad to hear you like the new weather section design :)

aceisace, 2019-11-08:
@surak @mgfcf @zAm1987 @ipaid2win 
Finished the refactoring of the RSS module for the new release :D. 
The main improvements are bullet-points seperating each feed, titles and description being shown (previously, titles were missing), a better algorithm for selecting feeds (only feeds that can be displayed fully are shown) and better readability.

Below is the image generated by the RSS module. Is there anything that you think can be improved?
[image]

aceisace, 2019-11-08:
Also, this is the new Calendar image in planning for the next release:
 
[image]

The main difference is that the events from the current day (and next) will be shown below the Calendar. The bottom section will still be available (for rss feeds for example).

baljo, 2019-11-25:
@aceisace 
One small suggestion is to change the colour of the temperature to red when below 0 (at least with Celsius, perhaps also with Fahrenheit?). The minus-sign (-) is quite short and not that visible from a distance while the red colour tells it all (I've verified this on my own display). Of course for those with a black and white screen, this does not help.
Another suggestion that might be harder to fulfill and perhaps not even wanted by all, is to display current temperature and windspeed with 1 decimal. I do understand that space is limited so this might perhaps not be feasible in the new version. The forecasted values should not have decimals as it's just a forecast anyway.

aceisace, 2019-11-25:
@baljo 
Hi, thanks for the suggestions. Let me summarise the suggestions:
* Change text colour to red if the temperature drops below 0°C. -> I'll implement this feature in the new release
* Add support for decimal places for temperature and windspeed -> No harm with implementing this as an option. 
If you take a look at the dev branch (which contains all the files for the new release), you'll realise that there are a lot more configuration options in each module (i.e. weather). Adding several options is not a problem anymore :D
* Add support for wind direction. I wouldn't mind implementing this, but it's nearly impossible to show this anywhere on the weather section. 
Since this is pretty easy to implement, I can show you how you can get this show on your build.

baljo, 2019-11-26:
@aceisace 
Great that you are able to implement several things, some of them optional!
What it comes to the wind direction, I already coded this quick and dirty and got it to work, so right now I don't need any support, thx anyway.

                        winddeg = weather.get_wind()['deg']
                        if winddeg > 337 or  winddeg <  23: windtxt = "N"
                        if winddeg >  22 and winddeg <  68: windtxt = "NE"
                        if winddeg >  67 and winddeg < 113: windtxt = "E"
                        if winddeg > 112 and winddeg < 158: windtxt = "SE"
                        if winddeg > 157 and winddeg < 203: windtxt = "S"
                        if winddeg > 202 and winddeg < 248: windtxt = "SW"
                        if winddeg > 247 and winddeg < 293: windtxt = "W"
                        if winddeg > 292 and winddeg < 338: windtxt = "NW"

                        write_text(40, 35, windtxt, (175, 0))

aceisace, 2019-11-27:
@baljo 
Thanks for providing the code. I've implemented this feature on my build but I'm still wondering if the wind direction should be shown in the form of an icon (e.g. an arrow), which is rotated by x-degrees, where x is the direction of the wind in degrees instead of the letters.

Right now, it looks a bit too cramped and less readable:
[image]

baljo, 2019-11-28:
@aceisace 
A wind icon instead of letters would be perfect and also needs less space.

aceisace, 2019-11-28:
@baljo 
Thanks for the info. I figured this out a few days ago (i.e. no wind direction being shown for the current weather). I also found out that the wind direction is provided in forecast objects. I'm simply using the wind direction from the next couple of hours to prevent N/A signs.

aceisace, 2019-12-04:
## Looking for Beta testers
The new release (v1.7) is now ready for launch. The fine-tuning is also finished. Before I push the changes on the main branch, I'd like to ask a few people (beta-testers) to test the new release. 

If you are interested, please let me know via a comment so I can send you the link of the new OS file. Thanks in advance.

surak, 2019-12-04:
Me, of course

zAm1987, 2019-12-05:
If it's not time critical (i have some busy days this weekend) I can test it, too :)

[9 more replies]

closed 2019-12-18 by aceisace