aceinnolab/Inkycal #39: Calender crashes on sunday if week_starts_on = "Sunday"

dkudlek, 2019-06-25:
[image]

When `week_starts_on = "Sunday"` and it is sunday the program crashes. 

Issue can be repeated on every sunday.

aceisace, 2019-06-26:
@dkudlek 
Hi, thanks for letting me know about this error. I am aware of this error from here: [weekstart-sunday problem](https://github.com/aceisace/Inky-Calendar/issues/34#issuecomment-502475848).
The problem is that since the software was growing at a pretty fast rate, it was getting difficult to debug/further develop any features without causing errors.
I am currently refactoring the software to solve the problem with the maintainance (breaking the main software into smaller scripts) but it will take a few weeks until it's complete (including the bugfixes). When it is ready, I'll let you know in this issue.

aceisace, 2019-12-18:
I'll be closing this issue now, @dkudlek since the new release has been published.
Due to the new release. this issue is now irrelevant. 

If you feel a need to re-open this issue, please feel free to do so.

closed 2019-12-18 by aceisace