aceinnolab/Inkycal #40: New install doesn't work

surak, 2019-08-11:
It does this: 

[image]

aceisace, 2019-08-13:
@surak  
Hi, I think the problem might be that the settings file is not configured yet. Could you please take a look at this section on how to generate the settings file?  [Creating a settings file](https://github.com/aceisace/Inky-Calendar#adding-details-to-the-programm)

surak, 2019-08-17:
As I have the Pi inside of a glued box, connecting it to a screen is not an option. I'll scp the html and see what it does, then. I would rather have a functional settings file as it was before, though.

surak, 2019-08-17:
Ok, now I understand. That is the new "default" view. I'm not sure it's a good default, though.

closed 2019-08-17 by surak