aceinnolab/Inkycal #41: Multiple issues with calendars

surak, 2019-08-17:
I have an "abfallkalender" - some app creates a calendar with the days of garbage here in Germany, which you know how it works.

As this thing creates a calendar, I shared it from iCloud, and it creates a webcal link. As this is just the garbage, here is the example URL:

webcal://p33-caldav.icloud.com/published/2/MTMxNDQ2NDAxMzE0NDY0MIs4JThLcd2nh4XXItgMUl_XVRv5v22MZ9br9jNqMlgQR4sRlGVaofsVp55wQ10ggLVBxDxL7yZCz0dDVMxQTHY

aceisace, 2019-08-18:
@surak 
Hi Alex, I'm glad you like the new version :)
That's interesting. I did exppect webcal iCalendar URLs to work, but it seems as though that is not the case here. In regards to the http protocol error, which version of ics and arrow are you using?

Thanks for sharing the info on how to get a webcal working with the software. It seems as though google is still the best option.

surak, 2019-08-18:
No clue. I fresh-installed the thing, and it gave me the error. Should I check?

aceisace, 2019-09-03:
@surak 
Sorry for the late reply. 
I'll check this up myself and will provide the fix in the new release.

surak, 2019-09-03:
Hi, thanks! Or just let people know that it is a limitation of the ics library. Perhaps I should test it more and report to them!

aceisace, 2019-09-05:
@surak 
I'll add a note about the limitation of the ics library when I'm back. If you could test the ics library more, that would be helpful.

aceisace, 2019-10-09:
@surak 
Hi Alex, I've added a note on the readme of the repo about the limitation of the ics library in regards of webcal based calendars.

As this issue is now resolved, I'll be closing this issue now. If you feel a need to re-open the issue, please feel free to do so.

closed 2019-10-09 by aceisace

surak, 2020-10-10:
How about this?
https://github.com/C4ptainCrunch/ics.py/issues/157