aceinnolab/Inkycal #42: Hourly weather/next day weather

surak, 2019-08-17:
You know how I enjoy this  project. And to make it even more useful, my suggestion is that the weather should show the conditions for the rest of day instead of the current conditions (after all, we can just look outside for the current conditions).

The rationale is: "should I take an umbrella? Do I ride the bike or the car?"

surak, 2019-08-17:
ps: I like the number 42 of this issue

aceisace, 2019-08-18:
@surak 
I'm glad you like the software :). Thanks for the suggestion about the weather forecast as well.

I'll try to implement this feature in the new release, although that will take some time. As the weather forecast does take up some space, I was thinking of adding an option for the bottom section (currently used for events or rss feeds) to display the weather forecast. Does this sound satisfactory or do you have any better suggestion? Thanks in advance.

mgfcf, 2019-08-18:
@aceisace 
That is a great idea! Even though I do not use your calendar, a detailed weather section at the bottom sounds great to show a lot more details. Would it be possible to create some kind of temperature-graph as a way to visualize this?

surak, 2019-08-18:
No, that's the easy way :-)
I thought about better repositioning everything. We are not pixel-perfect yet, and there's a lot of space.

I thought about something before the calendar

Perhaps the code to position the days could be calculated in a proportion of the space available for the calendar instead of hardcoded, what do you think?

aceisace, 2019-09-05:
@mgfcf 
Hi Max, 
thanks for your opinion about the weather section feature. A temperature-time-graph would definitely look awesome, but requires a lot of space. I'll have to test if that's possible with the limited space.

aceisace, 2019-09-05:
@surak 
Do you have some kind of rough sketch to show a better layout of the weather information?
The new release uses relative positions instead of hardcoded positions. Everything is split into sections and space is allocated automatically.

surak, 2019-09-05:
I'm looking at the a picture of it here at work. I suppose all the displays are the same resolution, yes?

So, my suggestions: 

Weather area:
- Remove weather condition (icon is enough)
- Remove dawn time
- Remove wind speed

With this, you have space for weather conditions for the next hours, say, every 2 hours, with a chance of rain for the day. 

For the main area:
- The month is huge and there's a lot of space around. If you don't know what month it is, we have bigger problems than that :-P
- The calendar itself has a lot of empty vertical space. It can be used better.
- There's plenty of space between the calendar and the appointments.
- There are six appointments. Maybe having a couple less would clear up even more space.

aceisace, 2019-10-09:
@surak Hi Alex, I've taken note of the suggestions you mentioned. I'll try re-arranging the top-section to allow showing a weather forecast. Hopefully this can be implemented in the next release. Once there is some progress, I'll let you know.

I'll be closing this issue now. If you feel a need to re-open this issue, please feel free to do so.

closed 2019-10-09 by aceisace

aceisace, 2019-10-24:
Find out more about this feature request here:
https://github.com/aceisace/Inky-Calendar/issues/34#issuecomment-546131341