aceinnolab/Inkycal #43: Missing icon?

surak, 2019-08-19:
I thought it should show something at the upper-left corner. In other weather conditions there's always something there.

[image]

aceisace, 2019-09-05:
@surak 
Sorry, there is currently a bug with a few weather conditions. The problem is that the icon code for some weather conditions is not set correctly. This will also be fixed in the new release.

aceisace, 2019-10-09:
@surak I've taken note of this bug in the Project-section of this repo. I'll try fixing this bug soon.

As the bug has been noted, I'll be closing this issue. If you feel a need to re-open this issue, please feel free to do so.

closed 2019-10-09 by aceisace