aceinnolab/Inkycal #44: Wavefront 9.7inch e-Paper HAT as well?

n00b42, 2019-09-04:
Is it possible to use a Wavefront 9.7inch e-Paper HAT as well?

https://www.waveshare.com/wiki/9.7inch_e-Paper_HAT

aceisace, 2019-09-05:
@n00b42 
Hi, thanks for your interest in the Inky-Calendar software :)
To answer your question, it is possible to use the software on the 9.7" E-Paper display from waveshare. This specific repo only supports the 7.5" variant, but I made one specifically for the 9.7"one here: 
[9.7 inch Inky-Calendar software](https://github.com/aceisace/9.7-E-Paper-Calendar-software)

Unfortunately, I don't actually have the 9.7"E-Paper so it's a bit difficult to further develop the software so it's not as good as this one but at least it works.

aceisace, 2019-10-09:
@n00b42 
I'll be closing this issue now. If you feel a need to re-open this issue, please feel free to do so.

closed 2019-10-09 by aceisace