aceinnolab/Inkycal #45: Python 3.7 default

ahobitt, 2019-09-13:
The calendar didn't auto start at boot. I installed supervisor as described in the wiki, but the new default version for python seems to be python3.7.

Suggestion:
Change Line 257 in Installer-with-debug.sh from
`command = sudo /usr/bin/python3.5 /home/pi/Inky-Calendar/Calendar/E-Paper.py ` to `command = sudo /usr/bin/python3.7 /home/pi/Inky-Calendar/Calendar/E-Paper.py `

Everything else works like a charm, even for my Nextcloud Calender. Thank you!

KunalGautam, 2019-09-19:
I think python3 can be better option since it will support python3.x

`command = sudo /usr/bin/python3 /home/pi/Inky-Calendar/Calendar/E-Paper.py`

ahobitt, 2019-09-19:
That is true.

aceisace, 2019-10-09:
@ahobitt @KunalGautam 
Well noted with thanks. I've changed the mentioned line so it defaults to python3 now. Hopefully this should resolve the issue with the auto-start.

I'll be closing this issue now. If you feel a need to re-open this issue, please feel free to do so.

closed 2019-10-09 by aceisace