aceinnolab/Inkycal #46: Current day and day with appointment are almost the same in B/W displays

surak, 2019-10-28:
It's basically impossible to tell the difference between a day with an appointment and the current day on the black and white display. The current day's border is slightly bigger (as it goes around the appointment), but that's it.

Suggestion: make it dashed on b/w displays.

aceisace, 2019-10-28:
@surak 
You have a valid point. I've taken a note of this request and will swap the icon in question soon

aceisace, 2019-11-18:
@surak 
Hi, I've been  experimenting with a few icons, which can be dynamically adjusted in size (for the next release). Unfortunately, drawing a dotted/dashed border is pretty difficult in PIL without the use of bitmaps, which I'd like to avoid.

Hence, I chose this icon to mark days with events:
[image]
Do you think this is good enough or does it still need some improvement? Thanks in advance.

surak, 2019-11-21:
I like it. Don' know how good it actually looks on the screen. To be honest, the current system is not bad - you can see from the other side of the room. Especially with this new way you made the current day - very different from the appointment.

How would it look like, this red circle on the current day, and the round squares on appointments?

baljo, 2019-11-22:
I second Surak’s proposal. The dots aren’t as visible as squares. Or was that perhaps the intention?

aceisace, 2019-11-24:
@surak @baljo 
Thanks for the feedback. Since the intention of the new icon is a better differentiation between the current day and an event-day, any icon can be used which fulfills this purpose. 

I've just about finished the core of the new release so I'm doing some fine-tuning and debugging. I'll use the time in the next few days to test @surak 's proposal. When it's done, I'll post images of the E-Paper showing both variants (dots and old event icon).

If you have any more suggestions, please feel free to share them. Thanks.

baljo, 2019-11-25:
@aceisace 
Added a suggestion to the separate thread for suggestions

aceisace, 2019-12-15:
I've added both event-icon options in the next release, which you have probably tested. Think this issue can be closed now or is there still something that needs to be done with the event icon @surak  ?

surak, 2019-12-15:
Absolutely. It looks way better now!

closed 2019-12-15 by surak