aceinnolab/Inkycal #47: VERSION file with the version number

surak, 2019-10-28:
Since you have version numbers now, it could be that either a file named VERSION containing the current version, or at least github tags, where one could see which one is the current one (I keep my fork in line with yours and I have no clue which version is it hehehe)

Feel free to close this if you disagree, of course!

aceisace, 2019-10-30:
@surak 
There is a [CHANGELOG file](https://github.com/aceisace/Inky-Calendar/blob/Stable/Changelog.md) I created to give an overview of all the changes of each release. 
Additionally, this sticker here on the README page also shows the current release. which is v1.6:
[image]

Is there still a need to create the version file?

aceisace, 2019-11-18:
I'm closing this issue as the purpose of this issue had already been fulfilled. If you feel  a need to re-open this issue, please feel free to do so.

closed 2019-11-18 by aceisace