aceinnolab/Inkycal #48: New crash fetching events

ericwass, 2019-11-06:
**Describe the bug**
No longer able to launch InkyCalendar. Process crashes fetching events from calendars

**To Reproduce**
Steps to reproduce the behavior:
python3.5 /home/pi/Inky-Calendar/Calendar/E-Paper.py

**Expected behavior**
Inky-Calendar should launch and load events.

**Screenshots**
Fetching events from your calendar

Traceback (most recent call last):
  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 527, in <module>
    main()
  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 345, in main
    decode = str(urlopen(icalendars).read().decode())
  File "/usr/lib/python3.5/urllib/request.py", line 163, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.5/urllib/request.py", line 472, in open
    response = meth(req, response)
  File "/usr/lib/python3.5/urllib/request.py", line 582, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.5/urllib/request.py", line 510, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.5/urllib/request.py", line 444, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.5/urllib/request.py", line 590, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 400: Bad Request

**Desktop (please complete the following information):**
Raspbian GNU/Linux 9 (stretch)

**Additional context**
For some reason Inky-Calendar reset to default "settings.py". I went through the Web-UI and re-created the same settings I've always had. But it's now throwing errors.

aceisace, 2019-11-07:
@ericwass 
Hi, thanks for the report. To help you out with, the error code (error 400) seems to be related to an invalid iCalendar URL. Please check [if the URL is valid](https://icalendar.org/validator.html).

Alternatively, you can paste a copy of your settings.py file without any sensitive data (replacing it with dummy data). In that case, I can check the systax as well.

ericwass, 2019-11-07:
Well, the iCalendar validator keeps throwing a "Maximum file size exceeded"
error and I don't know why. I'm generating these URLs straight from the
Google Calendar website using "Secret address in iCal format". I've also
regenerated them a couple of times, just to double check. If I paste my
settings.py file and replace data with dummy data how will you test it?

aceisace, 2019-11-07:
@ericwass 
If the iCal validator is throwing the max file size exceeded, it means that the iCalendar contains too many events (probably a lot of events from the past as well). If that is the case, then the error caused by the Inky-Calendar software would be due to 2 reasons:
* The fetching of data took too long as there are too many events
* The internet connection from your RPIto your router is slow, leading to timeouts.
In both cases, the fix is relatively simple; freeing up some space from your iCalendar by deleting outdated events. 

You don't need to post the settings file now. Since there was a tiny bug in the settings web-UI, I was worried if the generated settings file had a syntax error. But since the iCal URL is fine and working, this possibility is out of question.

ericwass, 2019-11-07:
Thanks for troubleshooting this with me. I guess that makes sense but, like I said, it all happened out of the blue. I haven't added a huge backlog of events or anything and I've tried using both calendars and then each calendar separately (one has way more events than the other) and I get these errors every time. 

I'm also wondering about your theory that the process is timing out because of slow internet and / or fetching events. The Pi is literally 4 inches from the router and the router is giving me consistently 240Mbps across the room on my iPhone. I'm embedding a link to a screen recording I did of the process so you can see how quickly it throws the error.

<iframe src="https://player.vimeo.com/video/371718101" width="400" height="300" frameborder="0" webkitallowfullscreen mozallowfullscreen allowfullscreen></iframe><p><a href="https://vimeo.com/371718101/7a7c1ddf56">Launch_Inky-Calendar</a> from <a href="https://vimeo.com/user3163103">Eric Wasserman</a> on <a href="https://vimeo.com">Vimeo</a>.</p>

aceisace, 2019-11-07:
@ericwass 
Hi Eric, if the possibility with the slow internet is ruled out, then the only thing that comes to mind is the large number of events. To confirm this, could you try downloading your own Calendars (on your computer or laptop) and check it's filesize?
I'll dig deeper into this problem on the weekend to ensure it's working correctly.

ericwass, 2019-11-07:
Just exported them as .ics files from Google and the first one is 1.2mb and
the second is 956k. That doesn't seem very big to me but is it?

aceisace, 2019-11-08:
@ericwass 
That's strange. With this size, fetching the iCalendar shouldn't take that long. I'll dig deeper into this issue in the next few days.

aceisace, 2019-11-18:
@ericwass 
I've been keeping an eye on my Inky-Calendar in regards to errors while handling events, but so far, there were no errors at all. 
Try deleting all setting files in the Inky-Calendar/Calendar folder. Then, regenerate a new settings file from the web-ui and copy that file to  Inky-Calendar/Calendar. After that, please reboot the RPI to apply the changes and it should start using the new settings file.

ericwass, 2019-11-19:
Thanks a lot for looking into this. I did what you suggested but am still
getting errors. I'm so confused as to what's happening.

Traceback (most recent call last):

  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 527, in <module>

    main()

  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 345, in main

    decode = str(urlopen(icalendars).read().decode())

  File "/usr/lib/python3.5/urllib/request.py", line 163, in urlopen

    return opener.open(url, data, timeout)

  File "/usr/lib/python3.5/urllib/request.py", line 472, in open

    response = meth(req, response)

  File "/usr/lib/python3.5/urllib/request.py", line 582, in http_response

    'http', request, response, code, msg, hdrs)

  File "/usr/lib/python3.5/urllib/request.py", line 510, in error

    return self._call_chain(*args)

  File "/usr/lib/python3.5/urllib/request.py", line 444, in _call_chain

    result = func(*args)

  File "/usr/lib/python3.5/urllib/request.py", line 590, in
http_error_default

    raise HTTPError(req.full_url, code, msg, hdrs, fp)

urllib.error.HTTPError: HTTP Error 400: Bad Request

aceisace, 2019-11-19:
@ericwass 
From the error log, this is what I can interpret:
The problem is caused by the URL of the iCalendar, which is causing a [Error 400](https://airbrake.io/blog/http-errors/400-bad-request). This in turn is caused by the server, but it is unclear where the exact issue is. Regardless, the server does not respond and this causes timeouts.

If you don't mind and your iCalendar does not have any sensitive information, can you post the current settings.py file to my email address (at the bottom of the README?) and not in this issue?

aceisace, 2019-11-28:
@ericwass 
Hi Eric. Do you think we can close this issue since the problem was fixed? Thanks in advance.

ericwass, 2019-11-28:
Actually it's still not working for the calendars I need it to work on. You
offered to have me send you settings file but, even though I trust you,
there's definitely a lot of personal/sensitive stuff on those calendars.
I'm not sure what to do.

aceisace, 2019-11-29:
@ericwass 
You do have a point. Then it's best to leave this issue open. 
I will soon finish the new release (in a few days) so maybe then, you can test if the iCalendar can be parsed correctly.

ericwass, 2019-12-12:
So, not long after our last interaction Inky Calendar stopped displaying anything at all. I would see the Waveshare display refresh now and then but no image would ultimately resolve. I tried killing and then relaunching Inky Calendar, rebooting the Raspberry Pi, etc. Nothing worked. So tonight I uninstalled Inky Calendar and then did a fresh install. But now I'm getting an error on launch. Can you decipher?

Traceback (most recent call last):
  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 29, in <module>
    from ics import Calendar
  File "/usr/local/lib/python3.5/dist-packages/ics/__init__.py", line 3, in <module>
    from .alarm import AudioAlarm, DisplayAlarm
  File "/usr/local/lib/python3.5/dist-packages/ics/alarm/__init__.py", line 1, in <module>
    from ics.alarm.audio import AudioAlarm
  File "/usr/local/lib/python3.5/dist-packages/ics/alarm/audio.py", line 31
    self._sound: Optional[ContentLine] = None

aceisace, 2019-12-12:
@ericwass 
Hi Erik, I'm sorry to hear you had to go through all this. Due to too many flaws and bugs in the current (public) release, I did a full re-write of the software which is nearly complete. Unfortunately, the bugs in the current release were not taken care of. I apologise for that. 

On the other hand, the next release (v1.7 beta) is available here:
[v1.7 (BETA) release testing](https://github.com/aceisace/Inky-Calendar/issues/51)
It has lots of new features, stability improvements,  is faster and more reliable than the current release.

To answer your question with, it seems that the problem is related to an incompatability between the ics library versions. The installer installed the latest version of ics, the library used for parsing events, which is still too buggy. In the BETA release, this was fixed.

ericwass, 2019-12-12:
Thanks, Ace. I took a quick look at that beta link. Do I understand right
that it's a full install of Raspbian Stretch as well as the 1.7
beta version of Inky Cal? This would be a total overwrite of my existing
RPi setup, correct?

aceisace, 2019-12-12:
Yes, you are right, @ericwass . At the moment, only the full image is the 'best' way of setting up the new software. I am working on the installer but there are some bugs that need my attention before that.

If you think you can do some trial-and-error, you can actually try the new version without the OS. If you still have some time now, I'll help you set it up.

For starters, please uninstall any previous version of the Inky-Calendar software or waveshare E-Paper software. Next, type the following command in the terminal:
`git clone -b dev https://github.com/aceisace/Inky-Calendar`
This will download the software from the next release. It does take some time to download the 120 Megabytes, so please keep that in mind.
Next, install the requirements for the software. 
`pip3 install -r /Inky-Calendar/requirements.txt`
Again, this might take some time.

aceisace, 2019-12-12:
Next, create 2 symlinks (kind of like shortcuts on Windows, but not exactly) of the settings and configuration file:
`ln -s /home/pi/Inky-Calendar/settings/settings.py /home/pi/Inky-Calendar/modules/`
`ln -s /home/pi/Inky-Calendar/settings/configuration.py /home/pi/Inky-Calendar/modules/`

aceisace, 2019-12-12:
Next, add your settings in the settings file:
`nano /home/pi/Inky-Calendar/settings/settings.py`
The settings file already contains some info about filling in the details.

aceisace, 2019-12-12:
Lastly, run the Inky-Calendar software with:
`python3 /home/pi/Inky-Calendar/modules/inkycal.py`

The software should now be up and running :)

ericwass, 2019-12-12:
Ok. Thanks for all this. I will give it a go. The only step I'm concerned
about is removing the Waveshare files. Is there a simple way to do that?

aceisace, 2019-12-12:
How did you install the waveshare files? Did you follow any instructions from a website? If you could provide the link, I can 're-track' the changes and find commands to 'un-do' them.

ericwass, 2019-12-12:
So kind of you. I don't want to make you do a bunch of extra work. But, if
you're interested, I believe this is where I found the files/instructions:
https://www.waveshare.com/wiki/Pioneer600#Libraries_Installation_for_RPi

aceisace, 2019-12-12:
You're welcome :)
Thanks for providing the link. The only collision I could see was this command:
`sudo apt-get install python-imaging`
It installs PIL (Python Imaging library), but the Inky-Calendar uses the Pillow library (a fork of PIL). Unfortunately, both cannot co-exist at once. You can uninstall that package with:
`sudo apt-get remove python-imaging && sudo apt-get clean && sudo apt-get autoremove -y`
That should do the trick.

ericwass, 2019-12-14:
Got to step 2 and it's throwing an error:

*pi@raspberrypi*:*~ $* pip3 install -r /Inky-Calendar/requirements.txt

Could not open requirements file: [Errno 2] No such file or directory:
'/Inky-Calendar/requirements.txt'

aceisace, 2019-12-14:
@ericwass Sorry, my bad, it was a typo and is supposed to be:
`pip3 install -r /home/pi/Inky-Calendar/requirements.txt`

ericwass, 2019-12-14:
Seems to be working now. I'll let you know when I get through it all. Thank
you

aceisace, 2019-12-14:
Sure :). You're welcome

ericwass, 2019-12-14:
Working except for weather. API key is correct. This error mean anything to
you?

Initialising weather... Done

Cleanup of previous images...Done

Current Date: 13 Dec 2019

Current Time: 16:48

-----------Main programm started now----------

Calibration.. not required. Continuing...

Weather module: Connectivity check passed, Generating
image...__________OWM-ERROR!__________

Reason:  The searched item was not found.

Reason: Error: the specified time is not included in the weather coverage
range

Calendar module: Generating image...Done

Initialising E-Paper Display...Done

Optimising Image for E-Paper displays...Done

Creating image buffer and sending it to E-Paper display...Rotated image by
270 degrees...Done

aceisace, 2019-12-14:
Glad to hear you managed to set it up, @ericwass . Have you done any modifications to the weather module? Maybe something related to the time of forecasts?

If not, try to run the weather-module by itself with:
`python3 /home/pi/Inky-Calendar/modules/inkycal_weather.py`

ericwass, 2019-12-14:
Haven't touched the weather module or any other. How do you change the time
of the forecasts? Here's what happened when I ran the weather module by
itself;

*pi@raspberrypi*:*~ $* python3
/home/pi/Inky-Calendar/modules/inkycal_weather.py

Initialising weather... Done

Weather module: Connectivity check passed, Generating
image...__________OWM-ERROR!__________

Reason:  The searched item was not found.

Reason: Error: the specified time is not included in the weather coverage
range

aceisace, 2019-12-16:
Hi, @ericwass . Sorry for the late reply. It's a bit strange that you are getting this error. Can you try re-starting the Rpi and re-test the weather module?

ericwass, 2019-12-16:
Hi Ace,
I restarted and retried. Here's what I got:

*pi@raspberrypi*:*~ $* python3
/home/pi/Inky-Calendar/modules/inkycal_weather.py

Initialising weather... Done

Weather module: Connectivity check passed, Generating
image...__________OWM-ERROR!__________

Reason:  The searched item was not found.

Reason: Error: the specified time is not included in the weather coverage
range

aceisace, 2019-12-17:
@ericwass 
Thanks for testing the weather module again. It seems as though this problem requires further investigation since it is only happening on your end at the moment.

Please unzip the attached folder, copy the file to the Raspeberry Pi inside the modules folder within the Inky-Calendar folder and run it with [F5]. Then, once it crashed, please paste the full error via a comment. Thanks in advance.
[weather_test.zip](https://github.com/aceisace/Inky-Calendar/files/3970802/weather_test.zip)

ericwass, 2019-12-17:
Thanks, as always, for seeing this through. Here's what happened:

*pi@raspberrypi*:*~ $* python3
/home/pi/Inky-Calendar/modules/weather_test.py

Initialising weather... Done

Weather module: Connectivity check passed, Generating image...now: 16 Dec
19 18:38, forecast1: 16 Dec 19 21:38, forecast2: 17 Dec 19 00:38,
forecast3: 17 Dec 19 03:38,forecast4: 17 Dec 19 06:38

Traceback (most recent call last):

  File "/home/pi/Inky-Calendar/modules/weather_test.py", line 194, in
<module>

    forecast_fc1 = forecast.get_weather_at(fc1.datetime)

  File
"/home/pi/.local/lib/python3.5/site-packages/pyowm/weatherapi25/forecaster.py",
line 444, in get_weather_at

    timeformatutils.to_UNIXtime(timeobject))

  File
"/home/pi/.local/lib/python3.5/site-packages/pyowm/utils/weatherutils.py",
line 112, in find_closest_weather

    'not included in the weather coverage range')

pyowm.exceptions.api_response_error.NotFoundError: The searched item was
not found.

Reason: Error: the specified time is not included in the weather coverage
range

*pi@raspberrypi*:*~ $*

aceisace, 2019-12-20:
Hi @ericwass ,
I sent you an email with some additional instructions. Have you had a chance to take a look at it?

aceisace, 2020-01-18:
@ericwass Hi Eric, I'm sorry for the late reply. I previously had no clue what was causing this error. I think this might be related to the timezone you are using. 

May I ask you which timezone is causing this problem? Thanks in advance.

ericwass, 2020-01-21:
HI Ace,
Sorry for the late reply. I was out of town. I'm in California, US. So
Pacific Time zone.
Thanks,
Eric

aceisace, 2020-01-22:
@ericwass 
No problem. Thanks for informing me about the timezone causing these problems. I'll come up with a fix soon.

aceisace, 2020-02-16:
@ericwass 
Have you received the news about the new software update? Please let me know if this specific issue persists on the latest release. If it doesn't, please don't forget to close this issue. Thanks for understanding.

ericwass, 2020-02-19:
Hi,
I finally got a moment to update Inky-Calendar and test it out. I went
ahead and did a full update of Rasbian as well as Inky. I renamed the old
Inky-Calendar folder before I started. Installation took a little while and
eventually generated a couple of errors (pasted below). I did a test run
and noticed that while it was saying "calibrating ePaper" the display
wasn't actually flashing as it used to. I left for a while and came back to
see some errors had been thrown (pasted below) and the display is stll
stuck on December 2019 as it has been for 2 months.

INSTALLATION ERROR LOG:

Collecting python-dateutil (from ics==0.4->-r requirements.txt (line 3))
  Using cached
https://files.pythonhosted.org/packages/d4/70/d60450c3dd48ef87586924207ae8907090de0b306af2bce5d134d78615cb/python_dateutil-2.8.1-py2.py3-none-any.whl
Collecting six>1.5 (from ics==0.4->-r requirements.txt (line 3))
Exception:
Traceback (most recent call last):
  File
"/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py",
line 594, in urlopen
    chunked=chunked)
  File
"/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py",
line 391, in _make_request
    six.raise_from(e, None)
  File "<string>", line 2, in raise_from
  File
"/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py",
line 387, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/lib/python3.5/http/client.py", line 1198, in getresponse
    response.begin()
  File "/usr/lib/python3.5/http/client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.5/http/client.py", line 266, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without
response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 215, in
main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 353,
in run
    wb.build(autobuilding=True)
  File "/usr/lib/python3/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 380, in
prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 554, in
_prepare_file
    require_hashes
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 278,
in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 465, in
find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 423, in
find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/usr/lib/python3/dist-packages/pip/index.py", line 568, in
_get_pages
    page = self._get_page(location)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 683, in _get_page
[... 88 lines cut ...]

Current Time: 13:11

-----------Main programm started now----------

Calibration.. not required. Continuing...

[Errno 2] No such file or directory:
'/home/pi/Inky-Calendar/images/inkycal_weather.png'

Calendar module: Generating image...Done

RSS module: Connectivity check passed. Generating image...Done

Initialising E-Paper...Done

Sending image data and refreshing display...Done

Sending E-Paper to deep sleep...Done

49 Minutes left until next refresh

Cleanup of previous images...Done

client_loop: send disconnect: Broken pipe

aceisace, 2020-02-19:
@ericwass Deleted old email message and replacing it with a correctly formatted one:
Hi,
I finally got a moment to update Inky-Calendar and test it out. I went
ahead and did a full update of Rasbian as well as Inky. I renamed the old
Inky-Calendar folder before I started. Installation took a little while and
eventually generated a couple of errors (pasted below). I did a test run
and noticed that while it was saying "calibrating ePaper" the display
wasn't actually flashing as it used to. I left for a while and came back to
see some errors had been thrown (pasted below) and the display is stll
stuck on December 2019 as it has been for 2 months.

INSTALLATION ERROR LOG:
```
Collecting python-dateutil (from ics==0.4->-r requirements.txt (line 3))
  Using cached
https://files.pythonhosted.org/packages/d4/70/d60450c3dd48ef87586924207ae8907090de0b306af2bce5d134d78615cb/python_dateutil-2.8.1-py2.py3-none-any.whl
Collecting six>1.5 (from ics==0.4->-r requirements.txt (line 3))
Exception:
Traceback (most recent call last):
  File
"/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py",
line 594, in urlopen
    chunked=chunked)
  File
"/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py",
line 391, in _make_request
    six.raise_from(e, None)
  File "<string>", line 2, in raise_from
  File
"/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py",
line 387, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/lib/python3.5/http/client.py", line 1198, in getresponse
    response.begin()
  File "/usr/lib/python3.5/http/client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.5/http/client.py", line 266, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without
response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 215, in
main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 353,
in run
    wb.build(autobuilding=True)
  File "/usr/lib/python3/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 380, in
prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 554, in
_prepare_file
    require_hashes
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 278,
in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 465, in
find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/usr/lib/python3/dist-packages/pip/index.py", line 423, in
find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/usr/lib/python3/dist-packages/pip/index.py", line 568, in
_get_pages
    page = self._get_page(location)
[... 54 lines cut ...]
Cycle 3 of 3 complete
[Errno 2] No such file or directory:
'/home/pi/Inky-Calendar/images/inkycal_weather.png'
Calendar module: Generating image...Done
RSS module: Connectivity check passed. Generating image...Done
Initialising E-Paper...Done
Sending image data and refreshing display...Done
Sending E-Paper to deep sleep...Done
34 Minutes left until next refresh
Cleanup of previous images...Done
Current Date: 18 Feb 2020
Current Time: 13:11
-----------Main programm started now----------
Calibration.. not required. Continuing...
[Errno 2] No such file or directory:
'/home/pi/Inky-Calendar/images/inkycal_weather.png'
Calendar module: Generating image...Done
RSS module: Connectivity check passed. Generating image...Done
Initialising E-Paper...Done
Sending image data and refreshing display...Done
Sending E-Paper to deep sleep...Done
49 Minutes left until next refresh
Cleanup of previous images...Done
client_loop: send disconnect: Broken pipe
```

aceisace, 2020-02-19:
@ericwass From the traceback, it seems there are actually two kind of errors you are getting:
1) Your network connection seems to be pretty slow as it is causing several timeout responses. I suggest you move your rpi closer to your router. Then, run the steps from [this page](https://github.com/aceisace/Raspberry-Pi-Setup/wiki/Testing-the-networking-speed).
2) Something is wrong with the weather module. Can you check your:
* API-key (use your own if you aren't)
* location
* timezone

Once done, please let me know of the outcome. Thanks in advance.

ericwass, 2020-02-19:
Thanks for your reply. I will input my own values into the configuration
file. I left it at default just to make sure things were working correctly.
As far as the network goes, the rpi is literally 3 inches away from the
router and I'm getting 226mbps downstream on my iPad which is across the
room. So I'm afraid it's something else.

ericwass, 2020-02-19:
Now I'm getting a calendar error. I've copied and pasted it directly from
Google into the Settings-File Generator (I "X"ed out the private info but
can send to you directly):

*pi@raspberrypi*:*~ $* python3 /home/pi/Inky-Calendar/modules/inkycal.py

Traceback (most recent call last):

  File "/home/pi/Inky-Calendar/modules/inkycal.py", line 12, in <module>

    from configuration import *

  File "/home/pi/Inky-Calendar/modules/configuration.py", line 14, in
<module>

    from settings import *

  File "/home/pi/Inky-Calendar/modules/settings.py", line 1

    ical_urls = [
https://calendar.google.com/calendar/ical/XXXXXXXXX%40gmail.com/private-XXXXXXXXXXXXXXXXXXXXXX/basic.ics
]

                      ^

aceisace, 2020-02-19:
@ericwass Just formatting your comment for easier readability:
Now I'm getting a calendar error. I've copied and pasted it directly from
Google into the Settings-File Generator (I "X"ed out the private info but
can send to you directly):

```python
pi@raspberrypi*:*~ $* python3 /home/pi/Inky-Calendar/modules/inkycal.py
Traceback (most recent call last):
  File "/home/pi/Inky-Calendar/modules/inkycal.py", line 12, in <module>
    from configuration import *
  File "/home/pi/Inky-Calendar/modules/configuration.py", line 14, in
<module>
    from settings import *
  File "/home/pi/Inky-Calendar/modules/settings.py", line 1
    ical_urls = [
https://calendar.google.com/calendar/ical/XXXXXXXXX%40gmail.com/private-XXXXXXXXXXXXXXXXXXXXXX/basic.ics
]
```

aceisace, 2020-02-19:
@ericwass 
I think you mentioned this before about the wifi router. But considering the error messages, it all points to timeouts. Did you try resetting your router? What is your bandwidth? Did you run the speedtest-client?

Also, run the installer to first uninstall the software, then install it again. Choose the web-ui option to add your details. Once done, please re-run the software to check if this issue persists.

ericwass, 2020-02-19:
Hi,
First of all, I'm sorry that I keep pasting poorly formatted log files.
Since we're doing this by email (and I can't find the thread on GitHub) I
don't know how to format it properly. Anyway I uninstalled and then
reinstalled the software and recreated the settings file using the web UI
using three different Google calendar links. The issue persists. Also, I
installed speedtest cli on the RPi and am seeing that the speeds are much
slower on the RPi than on my iPad/iPhone/Laptop running speedtest in the
same room: Ping of 56.943ms and DL of 17.58 Mbit on the RPi as opposed to
Ping of 14ms and DL of 225Mbps on the iPad/iPhone/Laptop in the same room.
Very strange. Not sure what's happening.

*pi@raspberrypi*:*~ $* python3 /home/pi/Inky-Calendar/modules/inkycal.py

Traceback (most recent call last):

  File "/home/pi/Inky-Calendar/modules/inkycal.py", line 12, in <module>

    from configuration import *

  File "/home/pi/Inky-Calendar/modules/configuration.py", line 14, in
<module>

    from settings import *

  File "/home/pi/Inky-Calendar/modules/settings.py", line 1

    ical_urls = [
https://calendar.google.com/calendar/ical/tdpc8ktofh0q3j86dh7nn6lec4%40group.calendar.google.com/private-4140d1bdbe1e8e4a0dbab14ed7c7515b/basic.ics
]

                      ^

SyntaxError: invalid syntax

ericwass, 2020-02-19:
By the way, I just noticed I didn't hide private info so I've reset the
private calendar link on Google so it won't work now anyway.

On Wed, Feb 19, 2020 at 3:54 PM Eric Wasserman <ericwass@gmail.com> wrote:

> Hi,
> First of all, I'm sorry that I keep pasting poorly formatted log files.
> Since we're doing this by email (and I can't find the thread on GitHub) I
> don't know how to format it properly. Anyway I [...]

aceisace, 2020-02-20:
@ericwass 
OK, it seems I have found the issue. It seems that your settings file is not formatted correctly. In the new release, on the web-ui, you have to input your URLs like this:
'URL1','URL2', but you forgot these (' ') signs. As a result, python thinks that your icalendar are not calendars, but rather functions or variables. To fix this, open the settings file:
`nano /home/pi/Inky-Calendar/settings/settings.py` and change this:
`ical_urls= [https://calendar.google.com/calendar/ical/tdpc8ktofh0q3j86dh7nn6lec4%40group.calendar.google.com/private-4140d1bdbe1e8e4a0dbab14ed7c7515b/basic.ics]`
to :
`ical_urls = ["https://calendar.google.com/calendar/ical/tdpc8ktofh0q3j86dh7nn6lec4%40group.calendar.google.com/private-4140d1bdbe1e8e4a0dbab14ed7c7515b/basic.ics"]`
Then try running the software with:
`python3 /home/pi/Inky-Calendar/modules/inkycal.py`

ericwass, 2020-02-20:
Thanks, as always, for continuing to help me with this. I'm sorry it's taking so long to get it right. I blame you for creating such a great software tool that I love.

I fixed the calendar feed (as well as the RSS feed) and that seemed to clear up some errors. However, I'm getting different ones now. On top of that, even though it says the display is refreshing it is definitely not. I actually haven't seen the display change at all in two months. I'm not sure what's going on. I haven't changed anything on the RPi other than update your software and the Raspian OS (as part of the Inky Cal update process). Image attached...

```
pi@raspberrypi:~ $ python3 /home/pi/Inky-Calendar/modules/inkycal.py
Cleanup of previous images...Done
Initialising weather... Done
Weather module: Connectivity check passed, Generating image...Done
Calendar module: Generating image...Failed!
Error in Calendar module!
Reason:  HTTP Error 404: Not Found
RSS module: Connectivity check passed. Generating image...Done
Current Date: 20 Feb 2020 
Current Time: 10:01
-----------Main programm started now----------
Calibration.. not required. Continuing...
Weather module: Connectivity check passed, Generating image...Done
Calendar module: Generating image...Failed!
Error in Calendar module!
Reason:  HTTP Error 404: Not Found
[Errno 2] No such file or directory: '/home/pi/Inky-Calendar/images/inkycal_calendar_col.png'
RSS module: Connectivity check passed. Generating image...Done
Initialising E-Paper...Done
Sending image data and refreshing display...Done
Sending E-Paper to deep sleep...Done
59 Minutes left until next refresh
Cleanup of previous images...Done
```

[image]

aceisace, 2020-02-21:
@ericwass No need to apologise :).
Two things I noticed now:
The display not being refreshed at all usually means either 
* wiring (connection) issues (At least it looks alright)
* incorrect driver file. Can you post the entry: `model = "something"` from the settings file?
* Check if SPI is enabled ; run: `sudo sed -i s/#dtparam=spi=on/dtparam=spi=on/ /boot/config.txt` and then reboot: `sudo reboot`

The second thing is that the iCalendar URL does not seem to work (dead link?) as it returns error 404 (not found). Please check your iCalendar URL. Copy it directly (from Google Calendar), validate it by entering it here: [iCal Validator](https://icalendar.org/validator.html). If the validator doesn't show any error messages, you can then paste it in the settings file:
`ical_urls = ["URL1", "URL2"]`

ericwass, 2020-02-22:
Alright, now we're getting so close. I found the issue with the e-Paper display. For some reason I can't understand it seems like the settings-UI was generating the incorrect display: v2 colour instead of v1 colour. Anyway, I fixed it manually using nano. 

Still having issues with the calendars. It seems to be one specific calendar that causes the issue. I can't validate in on iCal Validator because it complains that the file is too big. It's 1.2 mb if I download it from the Google settings page. But I know it's good, so I wonder if Inky-Cal doesn't like the size either. It's too bad because it's my main calendar.

```
pi@raspberrypi:/home $ python3 /home/pi/Inky-Calendar/modules/inkycal.py
Cleanup of previous images...Done
Initialising weather... Done
Weather module: Connectivity check passed, Generating image...Done
Calendar module: Generating image...Failed!
Error in Calendar module!
Reason:  End must be after begin
RSS module: Connectivity check passed. Generating image...Done
Current Date: 21 Feb 2020 
Current Time: 12:34
-----------Main programm started now----------
Calibration.. required. Performing calibration now.
----------Started calibration of E-Paper display----------
Calibrating... black... colour... white...
Cycle 1 of 3 complete
Calibrating... black... colour... white...
Cycle 2 of 3 complete
Calibrating... black... colour... white...
Cycle 3 of 3 complete
Weather module: Connectivity check passed, Generating image...Done
Calendar module: Generating image...Failed!
Error in Calendar module!
Reason:  End must be after begin
[Errno 2] No such file or directory: '/home/pi/Inky-Calendar/images/inkycal_calendar_col.png'
RSS module: Connectivity check passed. Generating image...Done
Initialising E-Paper...Done
Sending image data and refreshing display...Done
Sending E-Paper to deep sleep...Done
26 Minutes left until next refresh
Cleanup of previous images...Done
```

[image]

aceisace, 2020-02-23:
@ericwass The size of the calendar is large, but it shouldn't stop the program from executing though. With the current error message: `End must be after begin`, it means that something isn't working right from the software side, but it might also be the calendar itself. 

For now, I'd suggest turning off support for recurring events:
`nano /home/pi/Inky-Calendar/modules/inkycal_icalendar.py`, then change:
`use_recurring_events = True` to
`use_recurring_events = False`.
If the

ericwass, 2020-02-24:
I changed the value of recurring events but still getting the same issue:

```
pi@raspberrypi:~ $ python3 /home/pi/Inky-Calendar/modules/inkycal.py
Cleanup of previous images...Done
Initialising weather... Done
Weather module: Connectivity check passed, Generating image...Done
Agenda module: Generating image...Failed!
Error in Agenda module!
Reason:  End must be after begin
RSS module: Connectivity check passed. Generating image...Done
Current Date: 24 Feb 2020 
Current Time: 11:24
-----------Main programm started now----------
Calibration.. not required. Continuing...
Weather module: Connectivity check passed, Generating image...Done
Agenda module: Generating image...Failed!
Error in Agenda module!
Reason:  End must be after begin
[Errno 2] No such file or directory: '/home/pi/Inky-Calendar/images/inkycal_agenda_col.png'
RSS module: Connectivity check passed. Generating image...Done
Initialising E-Paper...Done
Sending image data and refreshing display...Done
Sending E-Paper to deep sleep...Done
36 Minutes left until next refresh
Cleanup of previous images...Done
```

I don't want this to occupy all your time. I'm sort of ready to give up. I suppose I could send you a private email with the actual calendar link if you think that would help.

Let me know. Thanks.

aceisace, 2020-02-25:
@ericwass Sorry I can't pinpoint the root of this problem. It would be better to continue this on Discord or Email. If you don't have any sensitive information in your iCalendar, I could help debug it myself if you want to share the link (discord, email). Thanks in advance.
P.S: I'll be closing this issue now.

closed 2020-02-25 by aceisace