aceinnolab/Inkycal #49: Crash on first use

baljo, 2019-11-13:
I'm new to GitHub and Inky-Calendar, so apologizing if this is not the way to inform about issues.

I have 1 or 2 issues when using the program the first time, the first issue might perhaps be because of or related to the second. 

1. Nothing is shown on the 3-color e-ink display. 
It's connected to a Rpi zero WH through the e-Paper HAT.  When running the Inky-Calendar calibration.py from the terminal, the e-ink display is first blinking black, then white (shouldn't it blink red?) and then finally white. This process is run twice and takes almost 10 minutes which I've understood is to be expected. The label on the back of the display has the text SKU13380 which is the 3-color display.

2. When running the E-Paper.py from a terminal, the weather is shown in the terminal window, but then a crash occurs (see below). Nothing is happening on the e-ink display, it's completely blank.
.....
pi@raspberrypi:~ $ pi@raspberrypi:~ $ python3 /home/pi/Inky-Calendar/Calendar/E-Paper.py
Temperature: 2 °C
Humidity: 100 %
Wind speed: 3 km/h
Sunrise-time: 08:45
Sunset time: 15:49
Cloudiness: 90 %
Weather description: moderate rain

sunset coming next
Traceback (most recent call last):
  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 527, in <module>
    main()
  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 222, in main
    prev_weekstart = now.replace(days = - now.weekday())
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/arrow.py", line 602, in replace
    raise AttributeError('unknown attribute: "{}"'.format(key))

aceisace, 2019-11-14:
@baljo 
Hi, thanks for joining github and for the bug report. I am a little busy working on the new release, but I'll try fixing this issue as soon as possible. I apologise for the inconvenience.

baljo, 2019-11-15:
@aceisace 
Thx for responding! In the meantime, is there an earlier version that I could try?

aceisace, 2019-11-18:
@baljo 
I'm very sorry I don't have any backups of previous versions while they were stable. Given a week or two, the new release will be finished. When that happens, I'll let you know.

baljo, 2019-11-19:
@aceisace 
No problem. I was actually able to get it to work with an older version of arrow as I understood the error might be related to compatibility issues or similar. What I did was to install version 0.13.0 and copying all files to python3.7 (~/.local/lib/python3.7/site-packages/arrow $). 
Voila!

aceisace, 2019-11-19:
@baljo 
Thanks for pointing that out :). 
In the new release, I'll set ics.py to v0.4, which uses arrow 0.13 /0.14. That will fix the issue with the incompatability.

baljo, 2019-11-19:
@aceisace 
No problem. Just a question, by checking the OpenweatherMap API I understand that the API is returning wind speed in meters per second (m/s) while the display is showing the wind speed as being in kilometers per hour. Is there a conversion somewhere or is the label incorrect? 
Personally I'd prefer m/s as that is what we are used to in the Nordic countries but I can of course always change it in my own code.

aceisace, 2019-11-19:
@baljo 
I just checked the API and indeed you are right. The API does indeed return m/s and pyowm does not convert it either. In that case, the windspeed is shown in m/s. You can change the label of the windspeed from km/h to m/s.
I've taken care of this in the new release. Thanks for finding that :)

baljo, 2019-11-20:
@aceisace 
You're welcome!

aceisace, 2019-12-07:
@baljo 
Nearly forgot to answer your first question about the calibration from the first comment. Sorry about that.

To answer your question, 'calibration' is the process of flushing the image with a single colour several times to prevent ghosting. Ghosting is a E-Paper specific thing. It's like writing with a pencil on paper and erasing it again. Eventually, you'll see marks of the previous images on the new one. 

To prevent ghosting, the display is flushed with all the supported colours (black & white & (red) ), usually three times. Doing so ensures that the vibrant colours of each pixel remain that way.

In you case, if the display is not calibrating the red colour, then maybe you forgot to select the right display type in the settings file. Doing so will also calibrate the red colour.

P.S.: In the BETA release, the speed of calibration has been increased, so 3 cycles on the coloured E-Papers now take about 7 mins :)

aceisace, 2019-12-18:
I'll be closing this issue now @baljo, since the new release is now public. With the new release, this issue is irrelevant.

If you feel a need to re-open this issue, please feel free to do so.

closed 2019-12-18 by aceisace