aceinnolab/Inkycal #50: Problems with starting the "E-Paper.py" program

Druggzz, 2019-11-19:
Hello!
I have problems with my Inky Calender. I have try many things,but it does not want to work. I have test my URl with the iCalendar validation and it's fine. So I don't no where the problem is. If I want to start the calender the terminal says that:

pi@raspberrypi:~/Inky-Calendar/Calendar $ python3 /home/pi/Inky-Calendar/Calendar/E-Paper.py
Initialising weather
_________Starting new loop___________

Date: Tue 19 Nov 19 Time: 22:31

Internet connection test passed

Fetching events from your calendar

Traceback (most recent call last):
  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 527, in <module>
    main()
  File "/home/pi/Inky-Calendar/Calendar/E-Paper.py", line 391, in main
    date = now.replace(days=+i)
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/arrow.py", line 602, in replace
    raise AttributeError('unknown attribute: "{}"'.format(key))
AttributeError: unknown attribute: "days"

I hope you can help me with this information.
Thank you for so nices project.
And sorry for my bad english skills.

Druggzz

baljo, 2019-11-20:
@Druggzz 
I experienced the same issue which seems to be due to incompatibilities between different versions. I was though able to get it to work by manually copying files from the 'arrow'module. Please see issue #49 for some more information.

Druggzz, 2019-11-20:
Hi @baljo 
thank you for your fast answer.
How can I install arrow 0.13.0? 
I found this: pip install arrow==0.13.0
But I don't know how i can copying all files to python3.7 (~/.local/lib/python3.7/site-packages/arrow $).
Can you please tell me,how you did that?

baljo, 2019-11-20:
Hi @Druggzz 
Please note that I'm completely new to Linux and probably have done things incorrectly or at least in a cumbersome way. 
What I started with was (referenced from [https://www.gitmemory.com/issue/TailorDev/Watson/301/520773782]):
$ pip uninstall arrow
$ pip install --user arrow==0.13.0
...but this did not as such help and I needed to copy files manually. It might be that I should have left out the --user part to get it installed on system level.
I propose you try with uninstalling arrow (pip uninstall arrow) and install it your way (pip install arrow==0.13.0). 
If it works then you won't need any manual copying, if not, pls inform and I'll try to remember from where I copied the files.

Druggzz, 2019-11-20:
Hallo @baljo 
I must to tell you that it not work. I have unistall arrow and reinstall arrow 0.13.0 like you write me. I hope you remember which data you have copy or so. That would be very nice. 
And thank you for your help. I'm also new in Linux and program RaspberryPi

Thank you
Druggzz

baljo, 2019-11-21:
Hi @Druggzz 
Ok, I believe your theory is right. 
In a nutshell I copied the files from  **//home/pi/.local/lib/python2.7/site-packages/arrow**  to  **//usr/local/lib/python3.7/dist-packages/arrow/**  and to  **//home/pi/.local/lib/python3.7/site-packages/arrow/**   (pi here is the user name I logged into the terminal with)
Before copying, I made a backup of the existing files, just in case.
Why twice? Well, once to get the software to work when running it manually, and once to get it to work when running it 'automatically' (i.e. after rebooting).

If you can, I strongly recommend to connect the RPi to a monitor and keyboard and copy the files with the graphical user interface. In my case it was overall a bit easier to do it from a terminal as I otherwise would've needed to move lot of stuff on my mess....crowded desk.

If you do it through a terminal window, this is how I have reconstructed how I did it, might not be 100% but hopefully close enough:
1. _cd //usr/local/lib/python3.7/dist-packages/arrow/_
2. _ls //home/pi/.local/lib/python2.7/site-packages/arrow_  --> shows you all files in the source directory
3. _sudo cp //home/pi/.local/lib/python2.7/site-packages/arrow/arrow.py arrow.py_
...do this for every file (I did not copy the subdirectories), I know there's some way to copy all files with one command, but the MS-DOS way of doing it did not work out for me, so I copied each file separately

Repeat once more, this time by replacing the command in 1. with _cd //home/pi/.local/lib/python3.7/site-packages/arrow/_

Hopefully this helps you to get it working. After some time the author will release a new version where among other things the compatibility issues have been fixed.

Druggzz, 2019-11-21:
Hi @baljo 
thank you for you good manual. I copy every file you wrote me and it works fine.
I done this with the terminal like you because it to much expenditure to connect the raspberrypi zero with monitor and a keyboard.
But it works fine.
Thank you so much!

baljo, 2019-11-21:
@Druggzz Glad that it worked and that I could help!

closed 2019-11-22 by Druggzz