aceinnolab/Inkycal #51: BETA testing for release v1.7

aceisace, 2019-12-07:
# BETA testing
Welcome to the BETA-testing of release v1.7. Thank you for willing to test the new software.
As the Installer of the new release is still work in progress, an alternate way to install the software had to be found. For now, the Inky-Calendar software is offered as a pre-configures OS image, which you can simply flash on a blank MicroSD card.

Below is the link on where you can download the image file. Below are the steps on how to set up the OS and the Inky-Calendar software:
1) Download and unzip the OS file (with [7z](https://7-zip.org/)) from here: [BETA-RELEASE](https://github.com/aceisace/Inky-Calendar/releases/download/v1.7(BETA)/inkycal-v1-7-1.7z)

2) Format an empty MicroSD card (with [SD Card Formatter](https://www.sdcard.org/downloads/formatter/)) with at least 8GB space and flash the OS on the card (with [Balena Etcher](https://www.balena.io/etcher/))

3) Open **Editor** (on Windows) and paste the following to setup WiFi:
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
 ssid="WiFi-Name"
 psk="WiFi-Password"
}
```
Edit the 2-letter country-code, ssid and psk with your own. Make sure you don't add spaces after the `=` signs. 
Save the file as `wpa_supplicant.conf` in the boot directory.
If you have access to an Ethernet connection, you don't need to setup WiFi.

4) Add the SD card in the RPi and connect via VNC Viewer (preferred) or SSH. 
The hostname is `inkycal.local` ,the username is `pi` and the password is `raspberry` (default values). You can also use the IP address of the Rpi instead of the hostname.

5) VNC Viewer: Go to Inky-Calendar, open the settings folder and paste your details in settings.py 
SSH: `cd /home/pi/Inky-Calendar/settings && nano settings.py`

6) VNC Viewer: Open the modules folder inside the Inky-Calendar folder, open `inkycal.py` and run with the main programm file by pressing the `[F5]` key.
SSH: `cd /home/pi/Inky-Calendar/modules && python3 inkycal.py`

Once the main programm has started, you'll see the E-Paper display being refreshed.

surak, 2019-12-07:
Is it a whole system image? Then I can't. The raspberry is inside of that picture frame I've shown you before, all glued. If I can clone it from a branch on git, it's fine.

aceisace, 2019-12-07:
@surak 
Yes, this is a whole system image. I'm working on the new installer, but until it's ready, it'll take at least a few days.

ipaid2win, 2019-12-09:
After accidentally flashing my old Raspian image I managed to install your image (took me some time to notice that SSH and VNC were disabled and Inky was not there lol)  

Few things I noticed:

- The font is smaller than before, I prefer the larger font of 1.6. (or whatever version I was running before).
- The letters "a" and "o" look identical now, both look like a small "a". Obviously from the context of the word it's still easy to understand what is meant.
- Agenda points with really long names (40 characters+) get cut off
- That being said, the layout of the Agenda view now looks "cleaner". 
- When not using the RSS feed in the bottom section, the Agenda view no longer extends all the way down (wastes 20% of the screen space) 
- The new weather part is great, looks good and shows more information. 

The 1.7 beta seams completely usable, no crashing or weirdness so far.  Great work, thanks!

Alex

zAm1987, 2019-12-09:
Installing and configuring went fine. The main script seems to work without problems, too. I think it is even a bit faster now :)

As I conpared the settings file to my old one, I noticed the number of fetched events ist not configurable anymore. I am not really missing this feature, but I am curious what is the reason behind this ^^

My observations of the result the main script produced:
- As @ipaid2win already mentioned, the font is smaller than before and in addition the old (and fixed) problem of the pixelated characters with one "thin" side is now back. I have only the black_and_white display, so maybe this is not the case on the colour version.
- Weather section is a bit different (compared to agenda), as it has also smaller and pixelated font. but the characters are not distorted. Seems like there is some anti-aliasing missing or something like this. Besides that, I really like the new arrangement and infos of the weather section :)
- Localization does not work in the weather section (displaying "now") despite the set language is not english.
- Recurring events are not shown. I think I am the only one who got these working / displayed in the last version, but now I am in the same boat as the rest and they are not shown anymore.
- Agenda section layout looks very clean and "straight". This makes it easy to spot what's on schedule next very easily. Only the small font contradicts this a bit when looking from far away.
- As @ipaid2win said already, too, the agenda section does not expand all the way down when not specifing another module for the bottom section.

aceisace, 2019-12-09:
@ipaid2win @zAm1987 
Thank you very much for testing the new release and for your honest commments :)

Concluding what you both have said, the biggest issue is the font, or rather, the disortion of the font without anti-aliasing. Although anti-aliasing was never supported, the font in the last release was thicker than the current one. I'll fix this problem by switching the current fonts with thicker ones. For the black and white E-Paper displays, I'll activate the dithering option to improve readability.

In regards to the size of the font, you can change the fontsize (in some modules) to make the text look bigger or smaller. I'll implement this feature in all modules soon. 

You (@ipaid2win ) mentioned that the event names/titles are truncated of they are too long (40+ char). As the layout of the Agenda module has changed, the space available for event names decreased. What would you recommend doing in this case to improve this?

You both also mentioned that the Agenda module does not extend to the bottom if the bottom section is empty. I'll work on this so it automatically extends to the bottom.

I'm sorry that new release does not support recurring events @zAm1987 . If at some point, you were able to use recurring events, then you might be able to use them again. As gitub saves every commit, I'll search for the code that did support it once. The main reason I chose not to use that code was because of frequent crashing. If you want to help with that, I'll re-write the code so it shouldn't crash.

You are right that the weather section does not support the locale's 'now' translation. Unfortunately, translating just the word 'now' doesn't work with the time-managing module (arrow). Arrow will almost always use at least 2 words (e.g. 'just now') to describe now in nearly all languages. As two words can't fit in that small space, the weather module can't take the set language in consideration. 
But don't worry, if you open the weather module (inkycal_calendar), you can change the string 'now' to your desired text.

Apart from that, I'm glad you both liked the new weather and agenda section :)

zAm1987, 2019-12-10:
@aceisace thanks for the explanation :)
I will try the manual change for "now".

After a day of usage I have made to other observations:
- Despite being tuesday, the agenda still shows events from monday (and the das monday in general). Maybe something crashed and therefore has not been updated. I will try a restart and see If it happens again
- In agenda view localisation does not work, too ("all day")

zAm1987, 2019-12-10:
Update:
Restarting the pi did not help, the display still showed monday as first day in the agenda despite being tuesday today. I started the script manually again and now tuesday is the correct first day, but has the events from wednesday in it. In addition to that the last event from today/tuesday is shown on top of the first "day-line" and therefore having no assigned day in the agenda table.

Maybe a timezone problem or something similar?

zAm1987, 2019-12-11:
Update 2:
Similar problem as yesterday, the display was not updated automatically. It seems that the "cicle" to start the script again does not work properly.

Anyway, as the display still shows the strange layout drin yesterday, I made a photo of it. The explanation follows below:

[image]

- The first event "Zahnarzt" (the font can't display emojis, maybe a font with more characters is suited better?) is in the real calendars of me and my wife the last event of tuesday at 17:00.
- The events "Urlaub" until "Kosmetik Sugaring" are displayed as tuesday, but are for real correctly scheduled for wednesday.
- After "Kosmetik Sugaring" a recurring event should be there, which is not displayed.
- The event "CDU JHV Kernstadt"  is scheduled for wednesday and is correctly displayed on wednesday.
- The event "Nagelkosmetik Hände & Füße" is again a strange thing as it is displayed for wednesday, but happens at thursday.
- The other events are in a false order, too.

Hope this helps ;)

aceisace, 2019-12-11:
Thank you for the report on the bug in the agenda module @zAm1987 . Seems like I made a mistake with the sorting of events. I'll look into this after fixing the other bugs.

In regards to the localisation option for the agenda view (All-day), I'll see if I can display it in the set language. Normally, just like the 'now' from the weather section, this is simply a string which can be changed. 

May I kindly ask you to test the calendar module and see if the problem with the wrong event-timings persists? Thanks in advance.

zAm1987, 2019-12-12:
@aceisace 
I switched to calendar view and the events from today are displayed correctly on the bottom, only thing is they are off bei one hour. In addition events from today that that have ended hours ago are still shown, but I am not sure if this is just the way it actually should work.

Concerning the events from other days, I am not sure because there are so many, that I cannot tell from the symbols alone if every event is parsed correctly. On the other hand, the few "free" days are "free", so it seems to work.

aceisace, 2019-12-12:
@zAm1987 @ipaid2win 
I've fixed some of the bugs (non-readable fonts, incorrect event-times) and implemented the feature of auto-expanding the agenda module if the space below is free.
Recurring events and custom fontsizes for each module are coming soon. 

You can find instruction on how to update on the first comment.

aceisace, 2019-12-12:
@zAm1987 You mentioned that in the Calendar module, all events were off by one hour. I think that might be related to non-set system timezone. You can set the timezone with:
`sudo dpkg-reconfigure tzdata`. This should fix the problem.

In regards to the events of the Calendar section, they are made to show events from today and tomorrow, regardless of the time. I'll change that soon so it only doesn't show past events.

aceisace, 2019-12-12:
I've got some good news, @surak . It is now possible to install the software without the OS. In case you are interested on the instructions, you can find them here: [updating software from previous release](https://github.com/aceisace/Inky-Calendar/issues/48#issuecomment-565212329)

ipaid2win, 2019-12-12:
Alright, not sure if it was directly mentioned before, but I found some bugs:
- My "week_starts_value" is set to "Monday", but it shows a single event from Monday on Sunday.
- On Monday itself, events for Tuesday are shown. Two are shown correctly (except for being on the wrong day), but one is shown twice. 
- The rest of the week is also shifted.
--> I guess the calendar starts on Sunday and shifts all events one day forward.

Ok, it just updated at 00:00 and everything is correct again. Events are starting on Monday and the duplicate is gone too. Weird, let's see if it happens again.

I noticed that the spacing between events is sometimes different. On Tuesday I have 3 events, the spacing of the first two events is smaller than that between event 2 and 3. On Monday, I only have two events, their spacing is the same as the last two events on Tuesday. I'll create some test events to double-check.

Other than that, updating seams to work (forgot to have the process running after closing putty at first). Is it still possible to put the time of the last update on the screen? The previous settings.py had an option for that.

Thanks,
Alex

aceisace, 2019-12-12:
Thank you for the report @ipaid2win . You managed to pinpoint the issue with the shifted events. I also recently found out about that. I've fixed it in the new, well update so that should no longer be a problem.

In regards to the spacing, may I ask you to clarify this a bit further? Is is the fontsize or the line height or width?

I'm glad to hear the updating instructions work as expected :). Thank you for testing that.

As for the time of update, I still need to implement that. Should the time remain at the same place as the last version (bottom right) or should it go somewhere else?

ipaid2win, 2019-12-12:
@aceisace 

With spacing I meant the distance between each line of text or rather between the events. I just measured with some calipers, the spacing between the times in front of the events is always the same.  Ai also noticed that the time stamps in front of the events are also sometimes not in one horizontal line with the text behind them. I think its the same issue, only happens on the second to last event on a day.

[image]

zAm1987, 2019-12-13:
@aceisace 
Updating went fine. My observations:

- Sorting of events works now correctly.
- The font is definitely more readable. But with some characters it is now the opposite of the old problem, because they are too bold for this font size and the resolution of the display. Still better than before though and hopefully fixed with the upcoming font size customization ;)
- Agenda now uses empty space correctly.
- As you guessed, my timezone was somehow not set anymore. Now the times are correct.
- I located the string to change "now" in the weather section, but I am not sure whicht string to change for "all day" in the agenda section. Could you clarify, where I have to change the string?

aceisace, 2019-12-13:
Thank you for the additional feedback @zAm1987 . It's good to hear the sorting of the events works fine now :). The problem with some letters being too bold will be fixed soon. It's related to the pixel colour-remapping (function: reduce_colours) for the black-and-white E-Paper displays and is caused by creating more 'darker' pixels than desired. 

It's also good to hear the expansion of the agenda module works as expected and that the times are shown correctly after setting the timezone.

About the 'all-day' placeholder on the agenda module, I'm sorry it isn't easy to change it right now. 
As a temporary solution, you can open the agenda module ('Inky-Calendar/modules/inkycal_agenda.py') and edit line 82 and 87, so these parts:
`'time':'All day'` look like this: `'time':'Heute'`. 
I'll add this feature on my to-do list and work on this as soon as I get the recurring events done.

aceisace, 2019-12-13:
Thanks for pointing that out, @ipaid2win . I did a short check on what might be the cause of the slightly shifted lines. It's related to the fact that the function used for writing text ensures that all words are centered to the middle of the line by default. As a result, only lines containing letters that nearly touch the line (e.g. 'y') are slightly higher. 

I'll see what can be done to make them as straight as possible.

zAm1987, 2019-12-14:
@aceisace 
Thanks for the explanation about the "all day" string. I will try this in the next days :)

My last problem right now is, that the script or display does not automatically update. I dont know if this is the normal behaviour (as it is a beta version) or if it should actually work. Right now my display still shows the same information as Yesterday.

surak, 2019-12-14:
Hi

I have it running.

I really like the new calendar. It's way better.

[image]

What I don't like:

- There's something at the right of the current temperature. No clue what.
- Font rendering is... off. I don't know.
- I don't have events for today and tomorrow... So bring the next ones - better than this emptyness under, don't you think?
- The bitmaps seem to be reduced and the new rendering is strange - look at the clouds' bottom lines, or the wind speed. I think we can get pixel perfect on those :-)

Thanks a lot. The effort you've done so far is amazing!

surak, 2019-12-14:
Line 159 of inkycal_calendar.py has a hardcoded max number of events. but you have a variable for that at the beginning of the file.

ps: Accented characters work!

aceisace, 2019-12-14:
You are right @zAm1987 that the script does not automatically start at boot. This is mainly because this is a BETA release and the stability is being tested. 
On the other hand, you can use a simple crontab job to get the job done. As a side-effect, you won't be able to see any error messages or logs.

**Activating startup at boot**
`crontab -e`
You might be asked which editor to choose. Just choose 'nano' for now. Scroll at the botom of the file and paste this command on the new line:
`@reboot cd /home/pi/Inky-Calendar/modules/ && python3 inkycal.py &`
Press [Ctrl] and [X] together. Then press [Y] followed by [Enter] to confirm the changes. 
Lastly, reboot the Raspberry Pi with:
`sudo reboot`
This will make the software run while booting.

**Deactivating startup at boot**
`crontab -e`
Scroll down to the line at the end of the file, beginning with:
`@reboot cd /home/pi/Inky-Calendar/modules/ && python3 inkycal.py &`
and uncomment it so it looks like this:
`# @reboot cd /home/pi/Inky-Calendar/modules/ && python3 inkycal.py &`
Press [Ctrl] and [X] together. Then press [Y] followed by [Enter] to confirm the changes. 
That's all. After the next reboot, the software will not start at boot.

aceisace, 2019-12-16:
@ipaid2win @surak @zAm1987  Thank You everyone for all your feedback.
It took quite some time to implement the feature of parsing recurring events, but it's finally available in the BETA release now. I'd like to ask you all to perform one last update to check if something still needs some improvement in the new software or if it is ready for public. 

The main improvements are:
* Better rendering of text and icons on black-and-white E-Paper displays.
* Added support for recurring events!
 **Please open the module 'inkycal_icalendar.py' inside the modules folder and set 'use_recurring_events = True' to activate parsing of recurring events.**
* Improved layout and parsing of events in the event section of the calendar.
* Better language support (it's easier to change the default english strings with custom text)
* You can now choose a desired fontsize in nearly all modules and the layout will adjust itself automatically.
* Minor code cleanup and fixed typos

zAm1987, 2019-12-16:
@aceisace 
I tried using the experimental function for recurring events and it seems that it has some problems with the events of my calendar. Can I somehow see which events are the problematic ones in order to pinpoint the issue?

`-----------Main programm started now----------
Calibration.. not required. Continuing...
Weather module: Connectivity check passed, Generating image...Done
Agenda module: Generating image...Something went wrong while parsing recurring events
Something went wrong while parsing recurring events
Something went wrong while parsing recurring events
Something went wrong while parsing recurring events
Something went wrong while parsing recurring events
Failed!
Error in Agenda module!
Reason:  Begin must be before end
Initialising E-Paper Display...Done`

After setting the recurring events back to "False" the rest worked fine. The font looks actually cleaner now on my display and changing the strings in weather and agenda via the variable worked fine, too. Good work! :)

ipaid2win, 2019-12-16:
Just updated to the newest beta.

I get no errors when using the experimental feature for recurring events, everything shows up. This is great! Now I don't have to keep a copy of my calendar with only individual events.

The text is now way more readable from normal distances. Changing the font size of the agenda to larger than 14 cuts of part of the date text in front of the events. At 15 its only on Monday and Wednesday (Mon 16 De and Wed 18 De). I got rid of the month part, as I usually know which month it is. Or only have a single text banner on top with the month? Not sure what the style variable does, in date_str its hardcoded so I changed that and it now only shows Wed 18 without month and I turned up the font size.

Regarding the issue with long events getting cut off, (obviously with larger fonts even more text will be cut off) it would be really nice if events could use 2 lines (or as many as it needs) if they exceed a certain number of letters. Cutting it off after the last word that still fits (by looking for spaces between words) could look fairly clean. But just like the different spacing between events, I wouldn't give this too much priority :). 

I don't know why, but my cron job never really works. It only prints a blank screen on the eink display. Also, once I close my putty session, the calendar no longer refreshes. Haven't done much on Linux this year, so I'm a little rusty and have no idea whats wrong haha

Still, the new update is great!

aceisace, 2019-12-17:
Thanks for testing the update @zAm1987  :)
I'm sorry to hear that the recurring events are not parsed correctly. I did some testing with Google and Yahoo iCalendars and it seemed to be working fine. Are you using an iCalendar from a different provider?

On the other hand, the recurring events were at least filtered correctly, so that is working fine. To see which events caused the problems, you can try to use the attached module. Simply extract it, copy it to the RPi (inside the module folder of the Inky-Calendar folder) and run it with [F5]. It should print the first recurring event. Next, copy the line starting with 'RRULE' and paste it in a comment.

[recurring_test.zip](https://github.com/aceisace/Inky-Calendar/files/3970732/recurring_test.zip)

aceisace, 2019-12-17:
I'm glad to hear the update works as expected @ipaid2win :)
The one that took the most time was the support for recurring events. It that works now, the rest is a piece of cake. 

It's also good to hear the text is more readable now. I've added a note about contructing 'lines' in the agenda module instead of splitting up lines and pasting text in sections. The note about line wrapping is also there.

You asked about the `style` variable in the agenda module. This variable is used for debugging purposes to print exact times instead of just the date (when setting `print_events` to True).

In regards to the autostart of the software, I'm working on it at the moment. There was a problem with this release and supervisor, which prevented a successful start due to some incompatabilities, but I've found the issue and have fixed it just now.

You can either wait for a couple of days until the new installer is ready, which will take take of that, or you can simply do the following:
```bash
sudo apt-get update && sudo apt-get install supervisor -y
sudo mousepad /etc/supervisor/conf.d/inkycal.conf
```
In the new file, paste the following content:
```
[program:Inky-Calendar]
command = /usr/bin/python3 /home/pi/Inky-Calendar/modules/inkycal.py
stdout_logfile = /home/pi/Inky-Calendar/logs/logfile.log
stdout_logfile_maxbytes = 5MB
stderr_logfile = /home/pi/Inky-Calendar/logs/errors.log
stderr_logfile_maxbytes = 5MB
user = pi
startsecs = 30
```
Then save the file and exit.

Next, create a logs folder in the Inky-Calendar folder with:
`mkdir /home/pi/Inky-Calendar/logs`

Reload supervisor and start the program:
`sudo service supervisor reload && sudo service supervisor start Inky-Calendar`
The software should then run at every boot.

Before rebooting, don't forget to deactivate crontab.

aceisace, 2019-12-17:
Thank you everyone for your precious time and feedback. It took a bit longer than expected, but since there are no more major problems or issues with the v1.7 BETA release, I'm now going to make the release public.

The Installer and web-ui are now back and online. The auto-start at boot is also working as expected. 
I'm going to leave this issue open for now as there still are some unfinished parts, which will hopefullly be finished soon.

A special thanks to @zAm1987 @ipaid2win and @surak for volunterring to test the release. You are all simply awesome 💯

ipaid2win, 2019-12-18:
@aceisace 

I noticed that when using the bottom for the agenda as well, the lower part won't fully refresh when it switches over to the next day. The old data stays and the new data is put over it. This results in this mess:
[image]

Manually restarting the program or rebooting the pi usually fixes it. During a single day it's not happening, but once it shifts to the next day the glitch occurres. It actually already happened yesterday, but I accounted it to accidentally having multiple instances of the program running. After rebooting and double checking that there's only a single instance earlier today, it still persists. 

I'll reboot it now to fix it again, maybe it also happens when I force the calendar to shift around by creating some new events.

Alex

aceisace, 2019-12-18:
Thanks for the report, @ipaid2win . Indeed I did forget to 'erase' the bottom section when using the expanded agenda view. I'll fix this asap.

aceisace, 2019-12-18:
OK ,it's done @ipaid2win .
To fix that, open inkycal_agenda.py in modules (inside Inky-Calendar folder) and change this part:
```
  try:
    clear_image('middle_section')
```
to:
```
  try:
    clear_image('middle_section')
    if not bottom_section:
      clear_image('bottom_section')
```

zAm1987, 2019-12-18:
@aceisace

Yes me and my wife are using google calendar only. 
I ran your supplied recurring_test.py in order to find the problematic events in my calendar(s) and I am quite surprised what the script printed out... The problematic ones seem to be all yearly events (birthdays, "couple-day", etc). The real interesting thing however is, that there are several birthdays included from contacts I deleted years ago and that are actually not displayed in my google calendar as well.

Here is the de-personalized result of the script:
`BEGIN:VEVENT
RRULE:FREQ=YEARLY;WKST=MO;UNTIL=20170121
CREATED:20111016T153353Z
LAST-MODIFIED:20161030T201359Z
SEQUENCE:0
STATUS:CONFIRMED
DTSTAMP:20191218T141700Z
DTSTART;VALUE=DATE:19880122
DTEND:19880124T000000Z
SUMMARY:Contact I deleted several years ago
DESCRIPTION:Birthday
TRANSP:TRANSPARENT
UID:xxxxxx@google.com
END:VEVENT
Something went wrong while parsing recurring events
BEGIN:VEVENT
RRULE:FREQ=YEARLY;WKST=MO;UNTIL=20170310
CREATED:20130421T125605Z
LAST-MODIFIED:20161030T201425Z
SEQUENCE:0
STATUS:CONFIRMED
DTSTAMP:20191218T141700Z
DTSTART;VALUE=DATE:19880311
DTEND:19880313T000000Z
SUMMARY:Contact I deleted several years ago number 2
DESCRIPTION:Birthday
TRANSP:OPAQUE
UID:xxxxxxk@google.com
END:VEVENT
Something went wrong while parsing recurring events
BEGIN:VEVENT
RRULE:FREQ=YEARLY;UNTIL=20180827
CREATED:20180630T110224Z
LAST-MODIFIED:20180630T110423Z
SEQUENCE:0
STATUS:CONFIRMED
DTSTAMP:20191218T141700Z
DTSTART;VALUE=DATE:20180828
DTEND:20180830T000000Z
SUMMARY:Couple-Day
TRANSP:TRANSPARENT
UID:xxxxxx@google.com
END:VEVENT
Something went wrong while parsing recurring events
BEGIN:VEVENT
RRULE:FREQ=YEARLY;WKST=MO;UNTIL=20180917
CLASS:PRIVATE
CREATED:20170921T114013Z
LAST-MODIFIED:20180630T110902Z
SEQUENCE:0
STATUS:CONFIRMED
DTSTAMP:20191218T141700Z
DTSTART;VALUE=DATE:20170918
DTEND:20170920T000000Z
SUMMARY:Birthday Another Contact
TRANSP:TRANSPARENT
UID:xxxxxx@google.com
END:VEVENT
Something went wrong while parsing recurring events
BEGIN:VEVENT
RRULE:FREQ=YEARLY;UNTIL=20181022
CREATED:20171023T203624Z
LAST-MODIFIED:20180630T111413Z
SEQUENCE:0
STATUS:CONFIRMED
DTSTAMP:20191218T141700Z
DTSTART;VALUE=DATE:20171023
DTEND:20171025T000000Z
SUMMARY:Birthday Another Contact 2
TRANSP:TRANSPARENT
UID:xxxxxx@google.com
END:VEVENT
Something went wrong while parsing recurring events
Traceback (most recent call last):
  File "recurring_test.py", line 49, in <module>
    events.begin = events.begin.to(timezone)
  File "/home/pi/.local/lib/python3.7/site-packages/ics/event.py", line 131, in begin
    raise ValueError('Begin must be before end')
ValueError: Begin must be before end`

Sorry for the formatting, I am somehow too stupid to get this right here ^^

Anyway, I hope this helps :)

aceisace, 2019-12-18:
@zAm1987 
Thanks for providing all the data from the problematic events. You don't need to worry about the formatting as I usually try to keep the formatting consistent in all issues :)

What I now found out is that the problem was not with the parser, but rather with the events themselves. These events end before they begin, which causes issues with calculating the duration for example. This is an interesting find however, as I can now try to 'fix' problematic events with this specific problem (or at least bypass them).

I'll keep you updated in the next few days about the progress.

baljo, 2019-12-21:
@aceisace First of all, sorry I was not able to beta test as I had intended to do, life suddenly intervened.

Now when back on track, I've installed the new version and have similar problem as @zAm1987, i.e. I get message:
   **Error in Calendar module!
   Reason: Begin must be before end"**

The result is that the calendar section is completely blank, no dates, no events, just a white area. The weather and RSS sections are shown.
I don't have any recurring events, in this calendar actually only 1 event in January and 1 in February. I tried to run the recurring_test.py but get similar error:
**"Traceback (most recent call last):
  File "recurring_test.py", line 82, in <module>
    events.begin = events.begin.to(timezone)
  File "/home/pi/.local/lib/python3.7/site-packages/ics/event.py", line 131, in begin
    raise ValueError('Begin must be before end')
ValueError: Begin must be before end"**

baljo, 2019-12-21:
@aceisace Another smallish issue, easy to fix:
In inkycal_weather.py  you have the following code (starting on row 30 in my script):

"""Set the optional parameters"""
decimal_places_temperature = None if round_temperature == True else 1
**_decimal_places_windspeed_** = None if round_windspeed == True else 1

...while starting on row 223 below code. This is causing an error if using decimal places

"      else:
        meters_sec = round(windspeed_now, ndigits = **windspeed_decimal_places**)
        miles_per_hour = round(windspeed_now * 2,23694,
                               ndigits = **windspeed_decimal_places**)

baljo, 2019-12-21:
@aceisace 
Sorry for another post, please note that what you've done is awesome!

Related to the calendar error, as I only had a couple of events in this particular calendar, I created another calendar as a quick fix and moved the events there, this worked alright as a temporary solution.
Now I see that the time for events is shown in 12-hour format even if I've chosen 24-hour format, e.g. an event at 14:30 is shown as 02:30. This would be good to get fixed.

gabe-sudo, 2019-12-21:
Thanks for all your hard work, Ace
I've been getting a strange error since the 1.7 beta, weather display works if i dont set my timezone(but doesnt display the correct time obviously). however if i set my timezone, OWM throws out the error: "the specified time is not included in the weather coverage range" on running, and the weather info is not displayed.

Any ideas?

baljo, 2019-12-21:
Seems that the **update_interval** in settings.py does not work for me, regardless of what I put there, the refresh is happening every full hour only. I've tried with different values, e.g. 5, 10, 15 minutes but with the same result.
Just to make sure that the variable is being read from settings.py, I put "xx" there which resulted in an expected crash.
When running the inkycal.py manually from the command line, I always get a similar message as below, i.e. counting minutes to next full hour. This is thus regardless of what the update_interval has been set to.

"... Sending E-Paper to deep sleep mode...Done
31 Minutes left until next refresh..."
(above was shown at 21:29)

zAm1987, 2019-12-22:
@aceisace 
I found another strange behaviour with events, which are going on for several days and have the beginning in the past (view from the moment of refreshing the display):

[image]

In the attached photo the event(s) "Urlaub" is shown 12 times before / on top the actual day (So 22. Dez) because it has the beginning 6 days ago and is in 2 calendars (mine and the one from my wife). It seems that the script displays each day the event lasted before the actual day, as long as the event has not ended. (Note: I disabled recurring events in the script, because of the already explained error)

Also worth mentioning: I used the installer a few times in order to install, uninstall, ando so on. There were some issues in the log with dpkg... but everything seems to work fine and I really like the new function at the end to choose the way to configure the settings. That's a really comfortable install-script! :)

aceisace, 2019-12-22:
### Update
I'm sorry to everyone for the short notice, but I'll be outside the country until mid January. Unfortunately, I won't be able to work on the Inky-Calendar software until then. I apologise for the inconvenience.

Please feel free to report any bugs/errors. Once I'm back, I'll start working on them :)

zAm1987, 2019-12-28:
Next observation:

The calibration process does not work properly in my (black_and_white) display. While calibrating only the left half of the display does the full black and full white cycle. The right half only displays some weird pixel pattern.

The following update process however works fine and all areas of the display are properly updated.

I am not sure If this is a new behaviour or happened with older versions, too. But at least I have not actively seen this behaviour with the last versions...

ipaid2win, 2020-01-03:
@zAm1987 

I can confirm that it happened in earlier versions of 1.7 (or the beta). 

I currently have major issues with an event that spans from mid of December to next week. On the current day it shows multiple entries for that event. I haven't counted the number of repeated entries, but it looks like for every day that passed it added one additional event. I haven't noticed this specific behaviour previously, might be the experimental recurring events feature.

Alex

aceisace, 2020-01-13:
OK everyone, I'm back now and can finally work on debugging the issues reported in this thread. Once something has been fixed, I'll post an update in this thread.

aceisace, 2020-01-18:
@gabe-sudo Hi, sorry for the late reply. I've not come across this error myself. Could you tell me which timezone is causing the problems? That way, I can try to come up with a bugfix

gabe-sudo, 2020-01-18:
Hi Ace, no problem and thanks again for your hard work on this very cool project. I'm in Victoria, BC, Canada, so Pacific time zone. The OWM API key I'm using is for "Langford, CA" and timezone set for Vancouver.

aceisace, 2020-01-18:
Thanks for the info, @gabe-sudo . I'll look into this problem with the specified timezone.

aceisace, 2020-01-18:
## Update - New (minor) release
I've been working on the bugs and feature requests suggested by the community in the past few days. The main points are:
* added support for 4.2", 5.83", 7.5" (v2) E-Paper display
* Added driver files for above mentioned E-Paper displays
* Slight changes in naming of generated images
* Slight changes in importing module names (now using dynamic imports)
* Changed driver files for all E-Papers with the latest ones from waveshare (v4)
* Slightly changed the way modules are executed
* Removed option for selecting colour from settings file
* Fixed a problem where the calibration function would only update half the display on the 7.5" black-white E-Paper
* Implemented a possible bugfix for 'begin must be before end' error.

As the changes are bigger than expected, the new features cannot be directly integrated in the current release. Therefore, a new (minor) release will be published soon.

I'll be closing this thread and open up another one for the new release to prevent compatabilities problems. Thanks for your understanding.

closed 2020-01-18 by aceisace