aceinnolab/Inkycal #64: Calendar Sync from Devices

Ribitsch, 2019-12-26:
Hi
I would like to know, if iCalendar is also synced from Devices like an iPhone? 
Would be awesome to know (:

aceisace, 2019-12-26:
Hi, thanks for your interest in the Inky-Calendar software and for your comment.

Yes, sure. It is possible to sync the Calendar from your iOS device (or really any device nowadays).

In the case of iOS, if you go to
Settings>Passwords and Accounts, you‘ll find a list of accouts associated with your device. If you click on any account, make sure the ‚Calendar‘ sync is turned on. That way, if you add an event in the calendar of your iOS device, it‘ll be saved automatically on the iCalendar.

Next, open a browser (on a PC/Laptop) and login with the account from above. Go to the Calendar page (if you are using Google, it‘ll be Google Calendar) and get the sharing URL of your iCalendar. Lastly, copy the link and paste it in the Inky-Calendar software (via the web-ui). 

If anything is still unclear, please let me know :)

Ribitsch, 2019-12-29:
Thanks for the answer. Before I buy the display, I have another question:
Is it possible to also use a Raspberry Pi 3B or are some Settings specifically programmed for the Pi Zero?
Best wishes

aceisace, 2019-12-29:
Yes, it‘s possible to use any Raspberry Pi as they all support SPI, which is required to drive the display. 

But the reason the Pi Zero W is preferred is due to the lower power consumption (5V; 0.2A = 1W) which is best suited for the low-power E-Paper display.

By the way, there are now 2 variants of the 7.5“ E-Paper display. The ‚old‘ one has a resolution of 640x384px, while the ‚new‘ one supports ~800x400px, so I‘d recommend the newer one, although it may be slightly more expensive. The Inky-Calendar will soon support the new model as well.

Best Regards

Ribitsch, 2019-12-29:
Hi 
Thank you. Then I will go for the higher resolution display with 3 colors 👍🏻

aceisace, 2019-12-30:
Sure. I‘ll be back in Mid-January. Then, I‘ll add the support for the new 7.5“ E-Paper display.

surak, 2020-01-01:
@Ribitsch @aceisace there are some problems with connecting directly with ical, and you can find some closed issue here.

What I did was to publish an ical from icloud, and import it on google calendar, and use the url from google. That one was readable by the system, and it updates from icloud just fine.

A bit convoluted, but I'm using for about a year now and it works.

Ribitsch, 2020-01-01:
Hmm, that is actually bad news, as I don’t like Google having all my personal calendar entries :/
Thanks for the information

surak, 2020-01-01:
@Ribitsch hi, here you can see it better: https://github.com/aceisace/Inky-Calendar/issues/41

surak, 2020-01-01:
I used google because it was the lazy, easiest option. No one stops you from doing any other - including having an owncloud or another calendar server on the raspberry pi itself and fetching from icloud. It is also worth to having a try, since the great @aceisace updated the whole thing since then.

aceisace, 2020-01-13:
Hi @Ribitsch ,
I'm back now and will test the support of iOS-generated iCalendars in the next coming days. Once I make some progress, I'll post an update in this thread.

surak, 2020-01-13:
Hi, I just did it again with google calendar. The trick is that 
1) at the moment you add a iCloud calendar to google, you must type that it’s public, and this can’t be changed. 
2) the url google shows is wrong. If you want to go this way, I can write you how I do it with 1.7

Ribitsch, 2020-01-14:
> Hi @Ribitsch ,
> I'm back now and will test the support of iOS-generated iCalendars in the next coming days. Once I make some progress, I'll post an update in this thread.

Thank you! I also ordered the higher resolution display, so I am looking forward to installing a new release.

Ribitsch, 2020-01-14:
> Hi, I just did it again with google calendar. The trick is that
> 
> 1. at the moment you add a iCloud calendar to google, you must type that it’s public, and this can’t be changed.
> 2. the url google shows is wrong. If you want to go this way, I can write you how I do it with 1.7

I will come back to you, once the new display will be supported (:

aceisace, 2020-01-20:
Hi @Ribitsch ,
I did some further investigation and came to the same conclusion as @surak . This means that an Calendar created by iCloud directly cannot be used with the Inky-Calendar software as it is not in .ics format.

At the moment, the best solution to sync events from your iOS device is to rely on a different iCalendar provider for the syncronisation. There are also iCalendar providers apart from Google (e.g. Yahoo etc.) which can handle this task.

In short, this is what needs to be done:
* Add email account of iCalendar providor (Google, Yahoo etc.) in the 'Accounts' section in 'Settings'
* Activate syncronisation of 'Calendar' for that account
* Go to the Calendar app, select 'Calendars' and tick the ones you want to show (in this case the ones from Google/Yahoo)
* You can now add events in your Calendar App in Yahoo/Google Calendar.
* On a Computer/Laptop, login with your Google/Yahoo account, go to 'Calendar' and copy the link for sharing your iCalendar (secret links are best) and paste it in the Inky-Calendar software using the web-ui.

surak, 2020-01-20:
The biggest problem so far is that google only fetches from ical every other day. 

Another thing is that now it gives you the wrong url. For an example, for the calendar with id

aiu7gdbcu6b7lk1u7s5plhribmf3svog@import.calendar.google.com

It creates an url like this:

https://calendar.google.com/calendar/embed?src=aiu7gdbcu6b7lk1u7s5plhribmf3svog%40import.calendar.google.com&ctz=Europe%2FMadrid

But the correct url for Inky is:

https://calendar.google.com/calendar/ical/aiu7gdbcu6b7lk1u7s5plhribmf3svog%40import.calendar.google.com/public/basic.ics

The url above works for me (I altered the ID of course, just for you to have an idea).

Ribitsch, 2020-01-20:
> The biggest problem so far is that google only fetches from ical every other day.
> 
> Another thing is that now it gives you the wrong url. For an example, for the calendar with id
> 
> [aiu7gdbcu6b7lk1u7s5plhribmf3svog@import.calendar.google.com](mailto:aiu7gdbcu6b7lk1u7s5plhribmf3svog@import.calendar.google.com)
> 
> It creates an url like this:
> 
> https://calendar.google.com/calendar/embed?src=aiu7gdbcu6b7lk1u7s5plhribmf3svog%40import.calendar.google.com&ctz=Europe%2FMadrid
> 
> But the correct url for Inky is:
> 
> https://calendar.google.com/calendar/ical/aiu7gdbcu6b7lk1u7s5plhribmf3svog%40import.calendar.google.com/public/basic.ics
> 
> The url above works for me (I altered the ID of course, just for you to have an idea).

Thanks for the instructions!
I will try it out 👍🏻

closed 2020-01-25 by Ribitsch