aceinnolab/Inkycal #65: Waveshare 7.5 inch V2 Display with new resolution

kabal00, 2020-01-09:
Hi everybody,

Great work with the software!

is it possible to use the Inky-Calendar with the new Waveshare 7.5 inch with a higher resolution (800 x 480)? I tried to change the libraries without any success. 

Thanks in advance.

aceisace, 2020-01-09:
Hi @kabal00 and thanks for your comment. Support for the new 7.5" is in planning and will be added soon once the current known bugs are fixed.

Once further progress is made, I'll post an update in this thread.

aceisace, 2020-01-13:
OK, I'm back @kabal00 now. I've started working on the support for the new E-Paper displays. Unfortunately, it might take a bit longer than expected (~1-2 weeks) as I have to update nearly everything in the driver files. Reason being is that waveshare uses a new approach to render images on the E-Paper displays.

As the modification required are bigger than expected, a new release will be pusblished to support this feature. Before pushing the changes on the master branch, I might ask you to test the new version of the Inky-Calendar software.

kabal00, 2020-01-13:
Hey aceisace,

sounds Great and I will verify the new software. 

Thanks for your fast response and I am looking forward to your new software release.

aceisace, 2020-01-17:
Hi @kabal00 ,
I'm nearly finished with the integration of several more displays. For now, the following E-Paper displays are supported:
* 4.2" E-Paper black-white
* 4.2" E-Paper black-white-red/yellow
* 5.83" E-Paper black-white
* 5.83" E-Paper black-white-red/yellow
* 7.5" E-Paper black-white (640x384px)
* 7.5" E-Paper black-white-red/yellow (640x384px)
* 7.5" E-Paper black-white (800x400px)
* 7.5" E-Paper black-white-red/yellow (800x400px)

I've attached an image below on how the software looks like on the 7.5" black-white-yellow (800x400px):
[image]
Of course it still needs some fine-tuning (some letters are shown half in yellow and half in black, but other than that, it works!

That being the case, please let me know in a comment when you have some time to test the new version. That would be a great help to check for bugs and errors :)

Ribitsch, 2020-01-17:
> Hi @kabal00 ,
> I'm nearly finished with the integration of several more displays. For now, the following E-Paper displays are supported:
>
> * 4.2" E-Paper black-white
> * 4.2" E-Paper [...]

That sound awesome!
Would like test it out, but I am still waiting for my E-Ink Display to arrive.

aceisace, 2020-01-18:
No problem, @Ribitsch . Just let me know when it has arrived. Thanks in advance.

aceisace, 2020-02-16:
@Ribitsch I'm closing this issue as this feature has now been implemented. If you feel something isn't working right, please open a new issue. Thanks for understanding.

closed 2020-02-16 by aceisace