aceinnolab/Inkycal #67: Bug reports for release v1.7.1

aceisace, 2020-01-18:
# Bug reports
Here you have an option to chat about the Inky-Calendar software (v1.7.1) with the author (me) as well as other users from the community.

This issue is mainly intented to help identify bugs/errors/problems in the software. Once something has been reported, I'll try to take care of it as soon as possible.

P.S: Please remember to be nice :)

aceisace, 2020-01-18:
Starting off with the currently known bugs:
* Rendering will be off as the new drivers use a different approach to render images
* The error 'Begin must be before end' might persist. If this is the case, please let me know.
* dpkg errors might be expierenced when using the installer. As a temporary fix, please use `pip3 install Pillow`
* Multiple entries for multi-day events with beginning in the past. This requires some further investigation.
* For some timezones, the weather module will not generate an image. This requires some further investigation. (Problematic timezone: Vancouver)
* The image module will not work. This will take some more time.

Ribitsch, 2020-01-23:
Unfortunately, the calibration process always gets stuck (I have the 7.5inch 3-color V2 model).
It happened every time (second tries) at the first automated calibration process - something must be going wrong there.
As can be seen in the images below, it gets stuck at the calibration process (I aborted the program after approximately 6hours after it got stuck, when I was at home again)
Hopefully the display didn't get damaged .

[image]
[image]
[image]

zAm1987, 2020-01-24:
I installed the new version using the update function and besides the known dpkg error during the installation the process went fine. After editing the settings file I had the already reported issue with a blank screen, as I filled the strings of the old settings file. Anyway, now it works.

Running the main script with the correct settings.py however results in two interesting things:
- The script still puts out some "Something went wrong while parsing recurring events" errors concerning recurring events, but on the display itself it seems that every recurring event is shown. So I think the fix actually works :)
- The image itself is rotated by 180 degree and therefore upside-down (7.5inch bw) in my picture frame. Don't know if this is meant by "rendering will be off" or only the slightly distorted characters

Anyway, good work as usual, especially with the recurring events! :)

aceisace, 2020-01-25:
Sorry for the late reply @Ribitsch . You're getting a strange error which is related to the driver files. In short, before refreshing the display, the RPi waits for a response, but it didn't receive it from the E-Paper display. As I'm using the same display but it didn't happen to me, my best guess would be an issue with the wiring. Just double-check (re-insert) the ribbon cables and the connection of the driver board to the Raspberry Pi and the issue should not happen again.

aceisace, 2020-01-25:
Hi @zAm1987 , thanks for testing the latest version :). I'm glad to hear the update now works as expected. In regards to the recurring-event problem, as long as it doesn't crash, it's fine. If you notice anything missing from the events, just let me know.

In regards to the 'incorrect' rotation of the image, I'll add an option to flip the image right or left. I think the new 7.5" needs a left-flip while the previous 7.5" uses the right-flip.

Ribitsch, 2020-01-25:
> Sorry for the late reply @Ribitsch . You're getting a strange error which is related to the driver files. In short, before refreshing the display, the RPi waits for a response, but it didn't receive [...]

Issue is already solved! Thanks, it was a stupid mistake on my side ;-)

Ribitsch, 2020-01-25:
But what I just discovered is, that only when I start the script, the image is created and then sent to the display, but the hourly updates are just showing the same picture, because no new image is created (I checked the time stamps of the images). For more details see image.

[image]

aceisace, 2020-01-25:
It seems that something is incorrect in your settings file @Ribitsch . Please move your old settings file from Inky-Calendar/settings and create a new one using the web-ui. That should fix everything.

See here for more details about this: [screen blank after update](https://github.com/aceisace/Inky-Calendar/issues/69#issuecomment-577404581)

Ribitsch, 2020-01-25:
> It seems that something is incorrect in your settings file @Ribitsch . Please move your old settings file from Inky-Calendar/settings and create a new one using the web-ui. That should fix [...]

That’s not the solution, I completely installed the inky calendar and it’s still not working.
The settings file also has to be correct, because the first time when I start the script it works just fine.
So I don’t know, why it the image generation is not called for the hourly updates.

zAm1987, 2020-01-26:
I can confirm, that something with the hourly update does not work as expected. I just took a look at my display and saw that it is still the same picture as the first one after the Installation.

As the first cycle went fine, i dont think it is related to the settings file...

aceisace, 2020-01-27:
@zAm1987 @Ribitsch Sorry for the incovenience. As I was busy working on some new features, I couldn't test the new release for a long period. I'm working on this right now. You can track the progress [here](https://github.com/aceisace/Inky-Calendar/issues/72)

aceisace, 2020-01-27:
@zAm1987 @Ribitsch Thanks for the bug-report. I've just now managed to find and test a bugfix which should take care of this problem. You can use your current settings file as-is. Just update using the installer, paste your old settings file in the correct location and it should be done.

Please let me know if it works. Thanks in advance.

zAm1987, 2020-01-28:
@aceisace I used the update-function of the installer, but after updating running the main script via python3 inkycal.py does not work. The output is the following:

`File "inkycal.py", line 12, in <module>
    from configuration import *
  File "/home/pi/Inky-Calendar/modules/configuration.py", line 10, in <module>
    from PIL import Image, ImageDraw, ImageFont, ImageColor
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/Image.py", line 41, in <module>
    from ._binary import i8, i32le
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/_binary.py", line 16, in <module>
    from ._util import py3
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 724, in exec_module
  File "<frozen importlib._bootstrap_external>", line 857, in get_code
  File "<frozen importlib._bootstrap_external>", line 525, in _compile_bytecode
ValueError: bad marshal data (unknown type code)`

Any idea what's happening there?

aceisace, 2020-01-28:
@zAm1987 . It seems something is wrong with the imaging library used by the software. My hunch would be that something went wrong while trying to install/set-up Pillow.
Could you try: `sudo pip3 uninstall Pillow` and then `sudo pip3 install Pillow==6.2.0` ? After that, please re-try running the software.

zAm1987, 2020-01-29:
@aceisace 
Trying to uninstall or install pillow returns the same errors as the (I tried it again) process during the installer with everything concering python2.7... I think I'll start from again from scratch with a fresh raspbian buster image.

aceisace, 2020-01-29:
Sorry to hear it can't be resolved by uninstalling pillow @zAm1987 . You can start off with a fresh raspbian image or you could try uninstalling and re-installing using the installer. The installer also has an option to uninstall in case something doesn't work at all. Maybe you can try that before using the new image.

zAm1987, 2020-02-01:
@aceisace 
I startet from scratch with a fresh image of raspian and ran the installer. After that I used the webui to generate a setttings file (with the middle section set to agenda and the bottom section set to nothing). This however results in an error when running the main script, as the script somehow cannot handle the empty bottom section parameter.

As a possible fix I downloaded the files you supplied in the other thread "auto-expansion of bottom section" and tried to run the script again. Now the "empy section" error is gone, but another one concerning the font-size came up.

So with or without the additional supplied files, I am not able to run the main script (without bottom rss) right now :-/

aceisace, 2020-02-01:
@zAm1987,
thank you very much for the error report. I‘ll fix in on Monday when I‘m back. For now, could you please paste the Traceback in another comment so I know the exact cause for the fontsize error? Thanks in advance.

zAm1987, 2020-02-02:
@aceisace 
`Cleanup of previous images...Done Initialising weather... Done Weather module: Connectivity check passed, Generating image...Done Traceback (most recent call last): File "/home/pi/Inky-Calendar/modules/inkycal.py", line 29, in <module> middle_section_module = importlib.import_module(middle_section) File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module return _bootstrap._gcd_import(name[level:], package, level) File "<frozen importlib._bootstrap>", line 1006, in _gcd_import File "<frozen importlib._bootstrap>", line 983, in _find_and_load File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked File "<frozen importlib._bootstrap>", line 677, in _load_unlocked File "<frozen importlib._bootstrap_external>", line 728, in exec_module File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed File "/home/pi/Inky-Calendar/modules/inkycal_agenda.py", line 21, in <module> font = ImageFont.truetype(NotoSans+'Medium.ttf', agenda_font_size) NameError: name 'agenda_font_size' is not defined`

aceisace, 2020-02-02:
@zAm1987 
OK, I get what went wrong. I haven't pushed the changes from my local (development) folder to the github repo just yet (testing in progress). This includes the correct configuration to leave sections empty without crashing. The other file you are currently using is not compatible due to the latest PR.

If you wait until Tuesday, I'll push my local changes here which include:
* weather timezone bugfix
* correct rendering of colours
* no crashes on empty section
* Option for flipping the image upside-down

Apologies for the delay.

aceisace, 2020-02-16:
@zAm1987 @Ribitsch @Atrejoe @surak 
OK everyone, the new release is now online (v1.7.2). It contains minor improvements and bugfixes. To test out the new update, please run the installer and choose the update option. I'll be closing this issue to prevent mixups between versions. If you feel something isn't working as expected, please open a separate issue. From now on, there will be no more 'long' discussion issues. All discussions should take place on the Inky-Calendar chatroom at Discord. Please visit the readme on how to enter. Thank you very much for all the bug reports and suggestions so far!

closed 2020-02-16 by aceisace