aceinnolab/Inkycal #69: Screen keeps blank with v1.7.1

ahobitt, 2020-01-19:
Hello,
thank you very much for the update. 
Unfortunately my screen remains white.

It works with the old 7.5 display, Waveshare 7.5 Inch E-Paper Display Hat(B) Module 640x384 Resolution

At startup the screen flickers only briefly, but the calendar does not appear.

aceisace, 2020-01-19:
Hi @ahobitt ,
thanks for the report. I tried recreating this on my own build but it is working just fine. Before I investigate further, could you please paste the content of the line starting with `model = ` from the settings file, which is located in `/home/pi/Inky-Calendar/settings/settings.py?`

Also, could you please paste the file `errors.log` located in `/home/pi/Inky.Calendar/logs/errors.log`? This will help identify the cause. Thanks in advance.

aceisace, 2020-01-22:
@ahobitt 
I just heard from another user that the agenda module isn't correctly generating an image. If you're using only the agenda module, could you please check if the problem persists with the Calendar module? Thanks in advance.

aceisace, 2020-01-22:
Update: I've just now fixed the agenda module. If you update the software, the agenda module should be working just fine.

ahobitt, 2020-01-22:
@aceisace 
Probably it is not due to the settings.py 
Since it doesn't create anything with the following either:
`model = "epd_7_in_5_colour" # Choose the E-Paper model (see below)
language = "en"                # "en" # "de" # "fr" # "jp" etc.
units = "metric"               # "metric" # "imperial"
hours = "24"                   # "24" # "12"
top_section = "Weather"        # "Weather"
middle_section = "Calendar"    # "Agenda" #"Calendar"
bottom_section = "RSS"         # "RSS"
`
The error log is empty. 
Surprisingly the generated image is white when I look at it directly on the Pi. This makes me suspect that the creation of the image does not work as expected. I see you did a lot of work on the code there, too.

aceisace, 2020-01-22:
Thanks for the info @ahobitt . I figured out the problem. In your settings file, the modules are not set correctly (`weather` should be `inkycal_weather`). I'm guessing you tried to copy-paste your old settings file, hence the problem. 

As a simple fix, backup your settings file anywhere (not inside the Inky-Calendar folder), re-run the installer (choosing update) and use the web-ui to insert your personal info from your backup up settings file. You can read more about the settings file [here](https://github.com/aceisace/Inky-Calendar/wiki/Settings-file).

Or you can just replace these three lines in your settings file:

```python
top_section = "Weather"        # "Weather"
middle_section = "Calendar"    # "Agenda" #"Calendar"
bottom_section = "RSS"         # "RSS"
```

with these: 

```python
top_section = "inkycal_weather"        # "Weather"
middle_section = "inkycal_calendar"    # "Agenda" #"Calendar"
bottom_section = "inkycal_rss"         # "RSS"
```

Hope this helps :)

ahobitt, 2020-01-25:
Thank you @aceisace . That was to easy :)
Would you please update the default settings file to this, I just looked at the default file, and there it has the same lines.

closed 2020-01-25 by ahobitt

aceisace, 2020-01-25:
@ahobitt  Done! Sorry for the inconvenience.