aceinnolab/Inkycal #70: Feature requests for future updates

Ribitsch, 2020-01-21:
Hi,

I just tried the software with the V2 7.5inch E-Ink display.

As you already mentioned, the text is sometimes not correctly rendered. But within first tests, the functionality is given! 

My main concern is the customization: I would just like to use the display for calendar entries, I really don't need the top and bottom area for news or weather. So currently these areas are empty for me.

Is there any possibility to use the calendar on the whole display (complete top to bottom)?
(or to use the agenda view on the complete display?)

Another thing that would be nice is to set a text size, because the calendar entries have very tiny letters.

Other than that, I am looking forward to using it. 

[image]

aceisace, 2020-01-22:
Welcome on board of the Inky-Calendar software @Ribitsch .Looks like you made it through the installation :)

As you correctly identified, the display is 'split' into 3 sections ( [see here](https://github.com/aceisace/Inky-Calendar/wiki/Layout#section-width)) as some module require more space than others. As a result, each section can have one allocated module (task). In your case, the Calendar module has already filled up it's allocated space and won't extend further than that.

I can implement the feature of expanding the calendar module if the bottom section is empty. In this case, the top-section would remain empty. If this solution is satisfactory to you, I'll go ahead with the implementation. The other option would be to automatically expand the module in the middle section to the top and bottom (if both other sections are empty). This would lead to no empty space. Which option do you think is better?

In regards to changing the fontsize, this option is available in nearly all modules (except for the calendar module). I'll add the dynamic fontsize support in the Calendar module soon.

I've also spotted a problem with the agenda module. I'll fix this asap.

I'm glad to hear you're looking forward to using it :)

aceisace, 2020-01-22:
Update: Agenda module fixed now. It was a typo. Sorry for the inconvenience.

aceisace, 2020-01-22:
@Ribitsch 
To see the customisation options of each module, you can check out the Wiki of this repo. For the Calendar module, these are the customisation options: [calendar module documentation](https://github.com/aceisace/Inky-Calendar/wiki/Calendar-module-(inkycal_calendar))

Ribitsch, 2020-01-22:
> Welcome on board of the Inky-Calendar software @Ribitsch .Looks like you made it through the installation :)
> 
> As you correctly identified, the display is 'split' into 3 sections ( [see here](https://github.com/aceisace/Inky-Calendar/wiki/Layout#section-width)) as some module require more space than others. As a result, each section can have one allocated module (task). In your case, the Calendar module has already filled up it's allocated space and won't extend further than that.
> 
> I can implement the feature of expanding the calendar module if the bottom section is empty. In this case, the top-section would remain empty. If this solution is satisfactory to you, I'll go ahead with the implementation. The other option would be to automatically expand the module in the middle section to the top and bottom (if both other sections are empty). This would lead to no empty space. Which option do you think is better?
> 
> I've also spotted a problem with the agenda module. I'll fix this asap.
> 
> I'm glad to hear you're looking forward to using it :)

Hello,

The second option, where no space is left, would be perfect (expanding the middle view to top and bottom)

aceisace, 2020-01-22:
Thanks for your opinion @Ribitsch . It might take a bit longer to implement this feature (~2 weeks I guess) depending on the changes required, but at least the empty space will be managed more efficiently.

Ribitsch, 2020-01-24:
> Thanks for your opinion @Ribitsch . It might take a bit longer to implement this feature (~2 weeks I guess) depending on the changes required, but at least the empty space will be managed more efficiently.

Thanks a lot 👍🏻

aceisace, 2020-01-28:
OK @Ribitsch . I've implemented the auto-expansion on my build. Before I push the changes, I wanted to ask for your help by testing the new feature.

To do so, download and unzip this [Attachment](https://github.com/aceisace/Inky-Calendar/files/4124310/auto-expansion-test.zip) (contains 2 files).

* Place `configuration.py` in `/home/pi/Inky-Calendar/settings/` (rename your old file)
* Place `inkycal.py` in `/home/pi/Inky-Calendar/modules/` (rename your old file)

Reboot to apply the changes. Then, you can leave the top and/or bottom section empty and the middle section will (hopefully) expand itself correctly, leaving no empty space.

Ribitsch, 2020-01-28:
> OK @Ribitsch . I've implemented the auto-expansion on my build. Before I push the changes, I wanted to ask for your help by testing the new feature.
>
> To do so, download and unzip this [...]

Sure! I probably can test it tomorrow.

Ribitsch, 2020-01-29:
> Sure! I probably can test it tomorrow.

Hi,

your files worked perfectly!

aceisace, 2020-01-30:
Thank you very much for testing the new files. If it works on your build as well, that's a good sign.
I'm still working on the black-red separation, but it'll be done soon :)

zAm1987, 2020-02-02:
@aceisace 
`Cleanup of previous images...Done
Initialising weather... Done
Weather module: Connectivity check passed, Generating image...Done
Traceback (most recent call last):
  File "/home/pi/Inky-Calendar/modules/inkycal.py", line 29, in <module>
    middle_section_module = importlib.import_module(middle_section)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/Inky-Calendar/modules/inkycal_agenda.py", line 21, in <module>
    font = ImageFont.truetype(NotoSans+'Medium.ttf', agenda_font_size)
NameError: name 'agenda_font_size' is not defined`

aceisace, 2020-02-16:
@Ribitsch Have you received the news about the new software update? Please let me know if this specific issue persists on the latest release. If it doesn't, please don't forget to close this issue. Thanks for understanding.

closed 2020-02-16 by aceisace