aceinnolab/Inkycal #71: Improve translation

ahobitt, 2020-01-25:
Currently the modules are mainly in English, although I indicate DE in the settings. 
For the weather it says "now" instead of "jetzt"
In the calendar only the English is displayed for "today_in_your_language", "tomorrow_in_your_language" and "at_in_your_language" and no 24 hour view.

aceisace, 2020-01-25:
Hi @ahobitt , sorry for the inconvenience. It seems I forgot to update the settings file, leaving some unused entries. Please use the web-ui to generate a new settings file.

The string translations have moved to the module that uses it. For example, 'now', used by the weather module checks up what to use in the weather module, not the settings file. 

The wiki can help you set some additional settings used by each module:
[Weather module documentation](https://github.com/aceisace/Inky-Calendar/wiki/Weather-module-(inkycal_weather))
[Calendar module documentation](https://github.com/aceisace/Inky-Calendar/wiki/Calendar-module-(inkycal_calendar))

ahobitt, 2020-01-25:
Thank you for the documentation in the wiki. I think it would be much easier to have a .po-file for every supported language.

aceisace, 2020-01-27:
You might be right that having .py files for every language would make the translation easier, @ahobitt. But in the long term (new features etc.) it would be too much trouble having to change them all. So for now, this is more desired.

aceisace, 2020-01-27:
I just did a small search and test for a solution to have python translate certain things @ahobitt . It seems Google translate is pretty well suited for this kind of task. In the next release hopefully, I'll be able to implement the automatic translation of the used english words. 

What do your think?

aceisace, 2020-02-25:
@ahobitt 
I've added support for translations for the calendar module now. It will be available in the next release. You can take a look at the translations [here](https://github.com/aceisace/Inky-Calendar/blob/dev/modules/inkycal_calendar.py). If you notice something which doesn't look right, please let me know. Otherwise, you can close this issue. Thanks in advance.

ahobitt, 2020-03-02:
Hi @aceisace 
Thank you very much. That looks good!

closed 2020-03-02 by ahobitt