aceinnolab/Inkycal #72: Images not generated after the initial (first) execution

Ribitsch, 2020-01-26:
I really don't know, what I'm doing wrong, the first time when I run the inkycal.py script, the images for the calendar are created. After that all successive updates skip this step and no new image is generated. This is why at every update, the modules images are taken from the the first run.

This is my settings.py (for testing I use the standard settings from the HTML settings creator):

ical_urls = ['https://calendar.google.com/calendar/ical/en.usa%23holiday%40group.v.calendar.google.com/public/basic.ics']
rss_feeds = ['http://feeds.bbci.co.uk/news/world/rss.xml#']
update_interval = "60"
api_key = ""
location = "Stuttgart, DE"
week_starts_on = "Monday"
calibration_hours = [0,12,18]
model = "epd_7_in_5_v2_colour"
language = "en"
units = "metric"
hours = "24"
top_section = "inkycal_weather"
middle_section = "inkycal_calendar"
bottom_section = "inkycal_rss"

aceisace, 2020-01-27:
Hi @Ribitsch . Thanks for the bug report. I was able to confirm this bug myself as well. I'll work on this asap and will provide an info about the status of this bug-fix. Sorry for the inconvenience.

aceisace, 2020-01-27:
OK @Ribitsch . I've implemented the bugfix for the problem where the image would not be updated after the first run. You can re-use your current settings file. Just update using the installer and the problem should hopefully be solved.

Ribitsch, 2020-01-28:
Thanks for the fast fix!
Will try it out in the evening

Ribitsch, 2020-01-28:
> OK @Ribitsch . I've implemented the bugfix for the problem where the image would not be updated after the first run. You can re-use your current settings file. Just update using the installer and the problem should hopefully be solved.

If all three modules are activated it works :)

But if I e.g. only want to use the agenda view and leave the other two sections blank,
I now get this error:

pi@raspberrypi:~ $ python3 /home/pi/Inky-Calendar/modules/inkycal.py
Cleanup of previous images...Done
Traceback (most recent call last):
  File "/home/pi/Inky-Calendar/modules/inkycal.py", line 22, in <module>
    top_section_module = importlib.import_module(top_section)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1003, in _gcd_import
  File "<frozen importlib._bootstrap>", line 942, in _sanity_check
ValueError: Empty module name

aceisace, 2020-01-28:
Glad to hear the problem was solved @Ribitsch . Thanks for reporting that issue.
You are right that at the moment, all sections have to be active, as it would otherwise throw import errors. This is only temporary (work in progress) and will be improved soon.

One of the users mentioned that the module from the middle section should automatically expand above and below (if they are empty) to make use of the entire space.

Ribitsch, 2020-01-28:
> Glad to hear the problem was solved @Ribitsch . Thanks for reporting that issue.
> You are right that at the moment, all sections have to be active, as it would otherwise throw import errors. This is [...]

Yes that was me (:

aceisace, 2020-02-16:
@Ribitsch 
I'll be closing this issue as the bugfix has been implemented and is available in the latest release. If you feel that something isn't working as expected, please open a separate issue. Thanks!

closed 2020-02-16 by aceisace