aceinnolab/Inkycal #74: ModuleNotFoundError: No module named 'RSS'

vitasam, 2020-01-28:
Hi, some requirements are missed? It was Ok before I did an update today
```
pi@raspberrypi:~ $ python3 /home/pi/Inky-Calendar/modules/inkycal.py
Cleanup of previous images...Done
Initialising weather... Done
Calendar module: Generating image...Done
Traceback (most recent call last):
  File "/home/pi/Inky-Calendar/modules/inkycal.py", line 24, in <module>
    bottom_section_module = importlib.import_module(bottom_section)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'RSS'
```

aceisace, 2020-01-29:
Hi @vitasam . Sorry for that. It seems I forgot to update an entry in the settings file. Just change this line (in `Inky-Calendar/settings/settings.py`): 
`bottom_section = "RSS"` -> `bottom_section = "inkycal_rss"`.

I've fixed it in the settings file template. Thanks for reporting this problem :)

vitasam, 2020-01-29:
Thanks! An amazing project!

aceisace, 2020-01-29:
Glad to hear you like it @vitasam :)

aceisace, 2020-01-29:
Don't forget to close this issue if it is resolved @vitasam . Thanks in advance!

vitasam, 2020-01-29:
Yes, I re-tested it, it is Ok now. Will close it

closed 2020-01-29 by vitasam