aceinnolab/Inkycal #77: Merging with 9.7 screen E-ink repo

Crickus, 2020-02-03:
**@Aceisace**, wouldn't it be too insolent to ask you to add 9.7 E-ink screen support for your project.
I'll promise to take care of testing part :)

aceisace, 2020-02-03:
@Crickus Sure, no problem. I think it'll take a week or two to have it integrated. Also, thanks for offering that you'll help with the testing :)
I'll update you on this post when some progress is made.

aceisace, 2020-02-16:
@Crickus Just a quick update. I've just now published a new release (software update). If there aren't too many new issues in the next coming days, I can start working on the support for the 9.7" E-Paper.

Crickus, 2020-05-28:
**@aceisace**  Any chance what you make 9.7 screen support in your mew beta?
Parsing private calendars stop working for me in any way. Some time ago repeating actions was shown, but new actions didn't appear. Now it shows only national holiday calendar. I think new parsing method you proposed here may help, but I'm not sure it can be applied to 9.7 version of script.
Also calendar began glitch (python process multiplies and eat whole memory) when month has more than 5 weeks to display ;(

aceisace, 2020-05-28:
@Crickus 
Currently, the BETA of the new release does not support other displays than mentioned in the README, but I'll be working on this once BETA is launched. It will take some weeks to get inkycal running on the 9.7" ePaper display since I do not have the means to test it myself (counting on your help with this one). 

But the support for this ePaper is something that will be available in the next (production ready) release.

If you want to help out with BETA, you can join [inkycal on Discord](https://discord.gg/sHYKeSM). That would be more helpful with giving feedback etc.

Crickus, 2020-05-28:
**@aceisace** Sure I'll try to help in any possible way.
Chat is awesome idea.

aceisace, 2020-05-29:
@Crickus Thanks for your help. BETA is now officially launched. Please take a look at the #75 to find out more how to get started.

aceisace, 2020-06-19:
@Crickus Hi Dimka, I've satrted working on the support for the 9.7" repo. Please check the channel of the 9.7" epaper display on Discord. Thanks in advance

aceisace, 2020-11-10:
@Crickus Closing this issue in favor of the Discord channel specifically for the integration of this E-Paper display. Please use the 9.7" E-Paper channel in the discord server for faster development. Thanks in advance.

closed 2020-11-10 by aceisace