aceinnolab/Inkycal #84: Suggestion: font sizes belong to settings.py

surak, 2020-02-29:
I have the calendar in a corner, and my vision is not that sharp as it used to be. Plus, our font rendering is not optimal.

Because of that, I use bigger fonts... And this configuration belongs to the settings, no?

aceisace, 2020-03-01:
@surak  Hi surak, thanks for this sugegstion.

The idea was that:
* Required settings belong to settings.py (change via web-ui)
* Optional settings belong to configuration.py (change manually, for advanced users only)

As the fontisze for each module isn't strictly required, fontsizes are found and set in configuration.py.

But I'd like to know the opinion of @vitasam and  @Atrejoe on this regard.

aceisace, 2020-04-03:
Will work on this once the situation gets better, apologies for the delay.

aceisace, 2020-05-18:
@surak 
Hi, the new release will have classes with configurable options (`inkycal_weather.fontsize = 13`). The configuration.py will not exist anymore. Any configuration belonging to a module should be set by that module. This will be a step to a more maintainable code.

A few improvements regarding the rendering of text are also underway, so hopefully there will be some more improvements to look forward to.

aceisace, 2020-05-18:
Closing this issue due to refactoring of the software (see /dev_v2 branch). If you feel a need to re-open this issue, please feel free to do so.

closed 2020-05-18 by aceisace