aceinnolab/Inkycal #89: Timezone error

DevelopAlex, 2020-03-21:
**Bug description**
Exception when parsing recurring events:

    TypeError: can't compare offset-naive and offset-aware datetimes
    Something went wrong while parsing recurring events:can't compare offset-naive and offset-aware datetimes
    Traceback (most recent call last):
    File "/home/pi/Inky-Calendar/modules/inkycal_icalendar.py", line 58, in fetch_events
    dates = rrulestr(rule, dtstart= events.begin.datetime).between(after= now.datetime, before = further_future.datetime)

Issue seems to be related to `pytz` package.
When comparing dates, both  dates must be either naive or not-naive (including timezone or not).
Also see [https://stackoverflow.com/questions/15307623/cant-compare-naive-and-aware-datetime-now-challenge-datetime-end](https://stackoverflow.com/questions/15307623/cant-compare-naive-and-aware-datetime-now-challenge-datetime-end).

pytz Version: `pytz==2019.3`

aceisace, 2020-04-03:
Will work on this once the situation gets better, apologies for the delay.

aceisace, 2020-05-19:
@DevelopAlex Hi, you might have gotten the news about the refacotoring of the inkycal software. On that note, can I ask you to test the the [ical_parser](https://github.com/aceisace/Inky-Calendar/blob/dev_ver2_0/inkycal/modules/ical_parser.py) from the /dev_v2 branch? Do you see a problem there as well in regards to the timezone? Thanks in advance!

aceisace, 2020-05-29:
@DevelopAlex 
The new release is now available as BETA. To find out more, please check here: #75 
It should fix most of the current issues, but further testing is required. If you check out the BETA, please let me know if this issue has been resolved. Thanks in advance!

aceisace, 2020-11-10:
Closing this issue as the timezone issue has been fixed in BETA. If you feel a need to re-open this issue, please feel free to do so.

closed 2020-11-10 by aceisace