aceinnolab/Inkycal #91: Error in Calendar module

MaRe64, 2020-04-03:
Hi all,

using Inky-Calendar since 3 weeks and it works fine until March 31.
Since April 1. i get an empty Calendar Screen show only "31 is not in list"
reboot etc don't help.
I can't find that anyone other has the same problem, so it might be an configuration problem?
I'm  not a programmer and new to python, and would be glad if anyone here have an idea.

i'm using v1.7.2 of Inky-Calendar

logfile.log entry:
"...Calibration.. not required. Continuing...
Weather module: Connectivity check passed, Generating image...Done
Calendar module: Generating image...Failed!
Error in Calendar module!
Reason:  31 is not in list
name 'bottom_section_module' is not defined
Initialising E-Paper...Done
Sending image data and refreshing display..."

Thanks

aceisace, 2020-04-03:
@MaRe64 
Hi, thanks for your bug report and your interest in the Inky-Calendar software.
About the crash, it seems that there's something (still) not working as expected with the parsing of events on the last day of the month.

If my hunch is correct, you are using the Calendar view and have a full day event on the 30th of April. The software shifts the end of an event to one minute later (should be 1st May, but for some reason is 31st Arpil), leading to this error.

You can try bypassing this error by either:
* switching from calendar to agenda view
* removing the full day event on the last day of month (or converting the full day event to a timed event)

In the meantime, I'll try re-creating this issue in the next upcoming days, though it might take a bit longer than expected.

MaRe64, 2020-04-04:
Hi, thanks for your quick response.
As you recommend, I switched to the agenda view, and it's fine.
About the failure, i can say that there is no event on April 30.

aceisace, 2020-04-11:
@MaRe64 Thanks for your feedback. As the problem at hand has been fixed, I'll be closing this issue. I'll also take a deeper look in the calendar module to find the cause for this error.

closed 2020-04-11 by aceisace