aceinnolab/Inkycal #95: JSON configuration file instead of python

Atrejoe, 2020-04-12:


aceisace, 2020-04-13:
@Atrejoe Hi; thanks for this Issue. I've been behind shedule so I'll try to catch up as much as possible.
I was working on the configuration file to use the json settings file (link from your comment not found?) but there are a few things that I think need to be improved in the web-ui first:

- [ ] E-Paper panel and required config should appear first on the web-ui (e-paper model, update interval, language, calibration hours). On that note, is it possible to use a dropdown list instead of checkboxes?

- [ ] One drop-down list for each section containing all modules (regardless if they are compatible, respecting the idea of 'dumb modules'). After selecting a module, module specific config options should expand automatically (like the image module config).

In the meantime, I'll work on the parsing and handling variables from the (current) json-settings file.

Atrejoe, 2020-04-13:
Hi @ace,I think we could work together on this in the same feature branch.I think your comments make sense, but are very much desired improvements, not blockers.I'll see if I can make the time tonight.

aceisace, 2020-04-13:
@Atrejoe Yes, sure. Let's work together on this feature. You are right that some of these improvements are partially desired, but they do help in grouping the variables while parsing.

I've more or less managed to finish the parsing progress of the **json** (not jsonc) settings file. Attached below is an example settings.py file and the python module for parsing the json data:

[json-test.zip](https://github.com/aceisace/Inky-Calendar/files/4471000/json-test.zip)

It's still work in progress so it might change, but at least, this is the right direction :)

Atrejoe, 2020-04-13:
For parsing jsonc, you can use `jsmin`, see: https://stackoverflow.com/a/60661844/201019

I already re-arranged the panels in the feature branch (`feature/json-settings`), [see here](https://htmlpreview.github.io/?https://github.com/aceisace/Inky-Calendar/blob/feature/json-settings/settings/settings-UI.html).

Changing the checkboxlist to dropdownlist is possible (and desired) too, I'll get it functionally working, but may have some issues with the UI framework. It currently uses `tocas-ui`, for which I have not found the docs nor unminimized css.

Dynamically adding one or more panels and their configs is possible too of course, but will take more work.
I know it would be very nice to have, when I think of a method to do this quickly (say a few hours), I will, but for now I will postpone that.

aceisace, 2020-04-13:
@Atrejoe 
Thanks for your help with the re-arrangement of the settings web-ui :). No need to rush with the implementation of the other features in the web-ui since there still is a lot of back-end work to do before json can be officially supported.

In regards to jsonc, yes it's possible to use it, but at this point, I'd like to avoid having additional requirements, if possible. Parsing json can be achieved with built-in libraries so it would make things easier. But that's just my opinion.

Atrejoe, 2020-04-14:
Okay, that too bad: now we can't keep documentation inside a functional config file, but let's focus on getting it to work in the first place.

Atrejoe, 2020-04-14:
I just had to give this a go: it appears `jsmin` is more permissive than `json` and there was a construction in my sample config that it deemed to be invalid. (`,` followed by `}`).

After removing it all seemed well:
https://repl.it/repls/TestyJitteryCarriers

aceisace, 2020-05-18:
@Atrejoe 
>Make InkCalender (inkycal.py) (prefer to) read settings.jsonc instead of settings.py

I've ticked point 1 since /dev_v2 no longer looks for a settings.py file and the web-ui no longer generates a settings.py file.

> Adjust instructions
Could you please elaborate a bit on this? Does this refer to the main file or the web-ui? Thanks in advance.

aceisace, 2020-05-29:
@Atrejoe About the migration tool from .py to .json, I don't think it's worth the work. The previous setting.py file does not support some of the parameters of the current .json file. And with the new web-ui, it is already pretty simple to add personal config (thanks to you) :)

aceisace, 2020-06-14:
I'm clsoing this issue since this feature has been implemented in BETA. The BETA is no longer dependent on a settings.py file, the WEB-UI does not generate a .py file and apart from some minor issues, the WEB-UI seems to be working well. Thanks for your hard work @Atrejoe !

closed 2020-06-14 by aceisace