aceinnolab/Inkycal #97: Multiple Event Prints for same events

Ribitsch, 2020-04-14:
As can be seen in the image, the same events are printed multiple times (even on the false date)

Happens for:
-) Recurring All Day Events (e.g. Birthdays)
-) All Day Events (e.g. Holiday)

[image]

aceisace, 2020-05-18:
@Ribitsch For the new release, I've switched from the ics library (currently used to parse icalendars) to the icalendar library. This will hopefully fix all currently related issues to the (correct) parsing of events. 

On that note, I'd like to ask you to test the new icalendar parser from here:
[new ical parser](https://github.com/aceisace/Inky-Calendar/blob/dev_ver2_0/inkycal/modules/ical_parser.py)

After copying this file to the rpi, please run the file and set it up:
```python
#  init class
a = icalendar()
# url can be a string or list of strings (containing urls): 'URL1' or ['URL1', 'URL2', 'URL3']
a.load_url('url')
# parse events from now until the next 4 weeks (applies to recurring events as well)
a.get_events(arrow.now(), arrow.now().shift(weeks=4))
a.show_events()
```

Do the events match up with your calendar?

Ribitsch, 2020-05-18:
Thanks for the info, I’ll try it asap

aceisace, 2020-05-19:
@Ribitsch Thanks for testing the new parser. It seems that this specific bug has been solved in the /dev_v2 branch. Once the new release hits beta, I'll notify you once again so you can test it :)

aceisace, 2020-05-29:
@Ribitsch 
The new release is now available as BETA. To find out more, please check here: #75 
It should fix most of the current issues, but further testing is required. If you check out the BETA, please let me know if this issue has been resolved. Thanks in advance!

aceisace, 2020-11-10:
Closing this issue as the bugfix has been implemented in the development branch. If this issue happens again, feel free to re-open this issue.

closed 2020-11-10 by aceisace