aceinnolab/Inkycal #98: Development Independent of hardware

Atrejoe, 2020-04-14:
In order to speed up development, most development tasks (apart from actual rending) can be developed on more powerful machines and in richer environments than running this on a Pi zero.

Think of IDEs like VS Code, PyCharm and various git GUIs.

# Optional side-quest
It would be nice if we can have some development options (with sensible defaults) like:
- [ ] Skip actual rending / sending to E-Ink panel
- [ ] Logging verbosity
- [ ] Show error details on screens (stack trace, besides message)

vitasam, 2020-04-17:
Hi @aceisace , should `feature/issue98_image_only_rendering` be merged in to `dev` branch?

aceisace, 2020-04-17:
@vitasam Yes, it can and should be merged. Merging now...

vitasam, 2020-04-17:
@aceisace I merged yet, could you double check that everything is Ok?

vitasam, 2020-04-24:
Tested it in `dev` brunch - works as expected. Closing the issue.

closed 2020-04-24 by vitasam