aceinnolab/Inkycal #104: Gallery with makers

surak, 2020-04-26:
Hi, how about a Gallery with the picture of the implementation in the wild?

aceisace, 2020-04-28:
@surak 
You mean something like a hall of fame containing pictures of the Inky-Calendar from different users?

surak, 2020-04-28:
Genau!

aceisace, 2020-04-29:
@surak 
OK, added a new page 'Hall of Fame' in the wiki: [Hall-of-Fame](https://github.com/aceisace/Inky-Calendar/wiki/hall-of-fame).

aceisace, 2020-05-05:
@surak Thanks for your mail. I've added the content to the Hall of fame page!

aceisace, 2020-05-18:
Closing this issue as the hall of fame has been created. If you feel a need to re-open this issue, please feel free to do so.

closed 2020-05-18 by aceisace

surak, 2020-05-18:
I will! The link to that image is broken :-)

https://github.com/aceisace/Inky-Calendar/wiki/hall-of-fame

aceisace, 2020-05-18:
@surak Fixed!