aceinnolab/Inkycal #105: Does not launch on fresh install of Raspbian (with desktop) - Syntax Error

clone545, 2020-05-02:
`python3 Inky-Calendar/modules/inkycal.py` does not launch - Gives below error:

`Traceback (most recent call last):
  File "Inky-Calendar/modules/inkycal.py", line 12, in <module>
    from configuration import *
  File "/home/pi/Inky-Calendar/modules/configuration.py", line 14, in <module>
    from settings import *
  File "/home/pi/Inky-Calendar/modules/settings.py", line 1
    ical_urls = [https://calendar.google.com/calendar/ical/<account redacted>/basic.ics]
                      ^
SyntaxError: invalid syntax`

**To Reproduce**
Steps to reproduce the behavior:
Install BCM2835 libraries as per Waveshare setup guide:  [Link](https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT)
Install from `bash -c "$(curl -sL https://raw.githubusercontent.com/aceisace/Inky-Calendar/master/Installer.sh)"` as per guide
Occurs on each run of `python3 Inky-Calendar/modules/inkycal.py`

**Expected behavior**
Expect some activity on e-paper display
Expect output to indicate running.

**Screenshots**
See above error output

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]  Raspbian (current latest, with Desktop)

**Additional context**
settings.py:

`ical_urls = [https://calendar.google.com/calendar/ical/<account redacted>/basic.ics]
rss_feeds = [http://feeds.bbci.co.uk/news/rss.xml]
update_interval = "20"
api_key = ""
location = "London, GB"
week_starts_on = "Monday"
calibration_hours = [0,12,18]
model = "epd_7_in_5_v2_colour"
language = "en"
units = "imperial"
hours = "24"
top_section = "inkycal_weather"
middle_section = "inkycal_calendar"
bottom_section = "inkycal_rss"
inkycal_image_path = "https://github.com/aceisace/Inky-Calendar/blob/master/Gallery/Inky-Calendar-logo.png?raw=true"
inkycal_image_path_body = ""
`

Any help to get this working would be appreciated.

aceisace, 2020-05-02:
@clone545 Hi, thanks for your interest in the Inky-Calendar software :).
Had a look at the bug-report. Thanks for using the provided tempate!

To help you with debugging, please follow the steps below:
### settings file:
- `ical_urls = [https://calendar.google.com/calendar/ical/<account redacted>/basic.ics]`
ical_urls is a list that should contain urls as strings -> `"like this"`. Add (`"` or `'` on either side of the url:
fix: `ical_urls = ["https://calendar.google.com/calendar/ical/<account redacted>/basic.ics"]`

- `rss_feeds = [http://feeds.bbci.co.uk/news/rss.xml]`
Same as above.
Fix: `rss_feeds = ["http://feeds.bbci.co.uk/news/rss.xml"]`

### Installing drivers as per waveshare instructions
- The bash installer usually installs everything required to run the software. This also includes drivers and additional libraries to drive the currently supported ePaper displays. There are a few incompatabilites between waveshare's code and the Inky-Calendar software (which is why the installer should be run on a fresh install). If the solutions for the settings file do not fix your issue, please re-run the installer on a fresh install of Raspbian Stretch (with Desktop)

clone545, 2020-05-03:
Thanks for getting back to me so quickly!

I've added the quotes around the ical_urls and rss_feeds and that seems to be working and the script now seems to run:

` python3 Inky-Calendar/modules/inkycal.py
Cleanup of previous images...Done
Initialising weather... Done
Calendar module: Generating image...Done
RSS module: Connectivity check passed. Generating image...Done
Current Date: 3 May 2020
Current Time: 14:08
-----------Main programm started now----------
Calibration.. not required. Continuing...
[Errno 2] No such file or directory: '/home/pi/Inky-Calendar/images/inkycal_weather.png'
Calendar module: Generating image...Done
RSS module: Connectivity check passed. Generating image...Done
Initialising E-Paper...Done
Sending image data and refreshing display...Done
Sending E-Paper to deep sleep...Done
12 Minutes left until next refresh
Cleanup of previous images...Done`

That said, when it says `Initialising E-Paper...Done Sending image data and refreshing display...Done Sending E-Paper to deep sleep...Done` there's still nothing showing on the display.  Is there something i'm doing wrong here?

clone545, 2020-05-03:
Ok, never mind, I spotted that I'd still got the display set to "Colour" in the settings.

I'm still missing the weather at the top as i'm getting this as part of the program run:
`[Errno 2] No such file or directory: '/home/pi/Inky-Calendar/images/inkycal_weather.png'`

[image]

clone545, 2020-05-03:
Ok nevermind, I had the display set to "colour" and the API key has been generated but isn't activated yet which is why I assume the weather wasn't generating properly.

I'll give it another try in a few hours once the API key has hopefully been activated.

aceisace, 2020-05-03:
@clone545 Good idea. The activation of the api-key does take some time (usually a few hours), but if the display still doesn't show anything, double-check your display model and set the correct model in the settings.py file:
`model = "epd_7_in_5_v2_colour"`

clone545, 2020-05-03:
Yep, I've just tried it now and it seems to work when I manually run `python3 Inky-Calendar/modules/inkycal.py`

All I need to do now is make sure that it's running on boot and I think I'm done!  Still need to find a frame though :)

Many thanks for your help in getting it going!  Looking forward to see how this project evolves in future too!

clone545, 2020-05-03:
FYI the install script didn't install `supervisor` and manually copying the code into the expected config and running the supervisor reload & restart hasn't triggered the script to start, either immediately or after a reboot.

I've change the $USER parameters in the config to hard-coded match my current user account and still no joy.

I've managed to get it to run on boot though with an `@reboot` trigger in cron which triggers `python3 /home/$USER/Inky-Calendar/modules/inkycal.py`

aceisace, 2020-05-03:
@clone545 Glad to hear it's working now :)
If the installer didn't install supervisor, it means the option for auto-start was not activated. I'd recommend renaming your current Inky-Calendar folder and re-run the installer. You can skip the updating part, just be sure to activate the auto-start on boot option at the end. Once done, you can replace the default settings file with the one from the previous Inky-Calendar folder. The installer normally sets up everything for the current user, under normal circumstances, you don't need to change the supervisor setup-file.

About the frame, I'd recommend a deep case where you can hide all the electronics. Alternatively, you can check out the following two 3d-files for integrating it into a frame:
* [Inky-Calendar frame holder](https://www.thingiverse.com/thing:4159363) (Credit to Ribitsch)
* [Inky-Calendar sleeve](https://www.thingiverse.com/thing:4256591) (Credit to eboda)

aceisace, 2020-05-18:
@clone545 
Has your issue been solved? If yes, please don't forget to close this issue. Thanks in advance.

clone545, 2020-05-18:
Ah yes, this was resolve (sorry, forgot to close the issue)  Thanks again with your help with this one.

closed 2020-05-18 by clone545