aceinnolab/Inkycal #111: Text alignment of agenda module

AlessandroMandelli, 2020-05-09:
Hi, I'm a very beginner with python and coding. I tried the agenda module on a 4.2 three colour eink. It works fine, but there are some problems in the text alignment. As you see from the attached jpg, I read on the 1st line of the agensa: Sab 9 Ma, then on the 2nd Dom 10, and then on the 3rd, Lun 11 M.
Dom 10 is not aligned on the left, but there is like a space and the month does not appear.
Is it possible to fix it and read, for example: Sabato 9 Maggio, Domenica 10 Maggio, Lunedì 11 Maggio, and so on?
Thanks.
[image]

aceisace, 2020-05-09:
@AlessandroMandelli 
The agenda module splits part of the total-width into a certain proportion for dates and event names. 
Ideally, the date should fit in the allocated width, making more room for event names, but this can be changed manually if you like. 

To do so, please edit: 
`/home/pi/Inky-Calendar/modules/inkycal_agenda.py`
scroll down to line 26/27:
```python3
date_col_width = int(line_width * 0.20)  # sets the width for date
time_col_width = int(line_width * 0.15)  # sets the width for time of event
```
`0.20` is 20% of the total width (0.2*100=20%), you can increase this value to allocate more space for the date coloumn.

This will only work with the current release. Updating to a new version or re-installing will remove these changes.

Hope this helps :)

Atrejoe, 2020-05-13:
It may be worth mentioning that adjustments to `inkycal_agenda.py` will not survive the next update.
In the future any config should be first enables in the component and then performed in `settings.json`.

As an improvement, I think the panel itself could determine what it nééds to show in the left column, determine the minimum width for it, then allocates the remainder of the space to the right column.

aceisace, 2020-05-14:
@Atrejoe Yep, that's on my to-do list.
With the custom config part of each module's init, the date-format can be specified. The agenda module will then compute what width is required to show the date-format without truncating, leaving the remaining width for event names.

aceisace, 2020-05-19:
@Atrejoe @AlessandroMandelli 
OK, this problem has been fixed in the /dev_v2 branch. In the next release, the agenda and calendar module will measure the max width of the date, then assign this width. This leads to automatic alignment of dates and times. 👍 

I'll be closing this issue as it has been resolved. If you feel a need to re-open this issue, please feel free to do so.

closed 2020-05-19 by aceisace