aceinnolab/Inkycal #112: What do you think about adding a Clock Module?

AlessandroMandelli, 2020-05-09:
It would be great if you can think about adding a clock module in the top section, showing the name of the day, the number, the name of the month and the year. And the possibility to move the weather in the bottom section, so you will have: the clock, the calendar with printed events and the weather.

Atrejoe, 2020-05-09:
Hi @AlessandroMandelli , a true clock including the time would be not so useful, as the refresh rate of the panel is quite low.

However a current date panel would be a useful (and simple) addition. (even if only to see if the app is still alive, when dead it would show an old date)

Currently a setup where panels can be set at any position, with custom heights is in the making, and would be possible then.

aceisace, 2020-05-09:
@AlessandroMandelli Just as @Atrejoe has pointed out, any update frequency below than once every 10 mins causes damage (partially permanent one) on the ePaper display (see ghosting). ePaper displays are meant to show static content, so a clock would not fit in this criteria.

I'm currently refactoring the software. A short (banner-like) section to show time of update, release version (software update notice) and a few other infos, right at the bottom of the display is in planning.

aceisace, 2020-11-11:
@AlessandroMandelli Closing this issue since a true clock would not be suitable for a E-Paper display made for static content. For very small E-Paper displays, this could potentially be possible, but none of these are compatible with Inkycal. Thank you for understanding.

closed 2020-11-11 by aceisace