aceinnolab/Inkycal #115: Error in Calendar Module

clone545, 2020-05-18:
**Describe the bug**
`pi@pical:~ $ python3 Inky-Calendar/modules/inkycal.py
Cleanup of previous images...Done
Initialising weather... Done
Weather module: Connectivity check passed, Generating image...Done
Calendar module: Generating image...Failed!
Error in Calendar module!
Reason:  No ':' in line '<meta name="viewport" content="width=device-width, initial-scale=1">'
RSS module: Connectivity check passed. Generating image...Done
Current Date: 18 May 2020
Current Time: 12:36`

Eink display shows weather module as normal, then a large gap where calendar used to be with `No ':' in line '<meta name="viewport" content="width=device-width, initial-scale=` before being cut off by the edge of the screen, then the RSS module as normal.

**To Reproduce**
Unsure;  No changes were made (by me at least) to the calendar or any of the modules.  Calendar stopped updating at some undetermined point yesterday. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image](https://user-images.githubusercontent.com/8034367/82209892-fb67ee00-9905-11ea-8112-166e6e1bb3bc.png)

**Desktop (please complete the following information):**
Pi Zero W with Raspbian Full

aceisace, 2020-05-18:
@clone545 
Hi, thanks for the report. From the error message, it seems to be a bug, however the full traceback is required to pinpoint the cause. I've attached a modified file for the calendar module. Can you try copying this file in the modules folder (`Inky-Calendar/modules/`) and then try running it? With this, you should be able to see the full traceback.

[inkycal_calendar_mod.zip](https://github.com/aceisace/Inky-Calendar/files/4645290/inkycal_calendar_mod.zip)

clone545, 2020-05-18:
I've tried running it and get this:

`pi@pical:~ $ python3 Inky-Calendar/modules/inkycal_calendar_mod.py
Calendar module: Generating image...Traceback (most recent call last):
  File "Inky-Calendar/modules/inkycal_calendar_mod.py", line 128, in <module>
    upcoming_events = fetch_events()
  File "/home/pi/Inky-Calendar/modules/inkycal_icalendar.py", line 31, in fetch_events
    calendars = [Calendar(fix_ical(url)) for url in ical_urls]
  File "/home/pi/Inky-Calendar/modules/inkycal_icalendar.py", line 31, in <listcomp>
    calendars = [Calendar(fix_ical(url)) for url in ical_urls]
  File "/home/pi/.local/lib/python3.7/site-packages/ics/icalendar.py", line 58, in __init__
    container = string_to_container(imports)
  File "/home/pi/.local/lib/python3.7/site-packages/ics/parse.py", line 175, in string_to_container
    return lines_to_container(txt.splitlines())
  File "/home/pi/.local/lib/python3.7/site-packages/ics/parse.py", line 171, in lines_to_container
    return parse(tokenize_line(unfold_lines(lines)))
  File "/home/pi/.local/lib/python3.7/site-packages/ics/parse.py", line 162, in parse
    for line in tokenized_lines:
  File "/home/pi/.local/lib/python3.7/site-packages/ics/parse.py", line 157, in tokenize_line
    yield ContentLine.parse(line)
  File "/home/pi/.local/lib/python3.7/site-packages/ics/parse.py", line 65, in parse
    raise ParseError("No ':' in line '{}'".format(line))
ics.parse.ParseError: No ':' in line '<meta name="viewport" content="width=device-width, initial-scale=1">'
pi@pical:~ $`

I'm not sure if that's the expected behaviour or not, but hopefully it helps.

aceisace, 2020-05-18:
@clone545 
Thanks for testing the modification and pasting the full traceback. Seems as though it's related to the parsing of the calendar you are trying to use. My hunch is that one or more of your iCalendar urls are not pointing at the .ics file (actual iCalendar file) but rather to the link showing your calendar in html (or similar).

For starters, please open `settings.py` and validate each iCalendar url you have via this [iCalendar validator](https://icalendar.org/validator.html).

clone545, 2020-05-19:
I think you've hit the nail on the head.

I had 2 calendars it was pulling down, my personal calendar (which was in .ics format) and a national holiday calendar (https://calendar.google.com/calendar/embed?src=en.uk%23holiday%40group.v.calendar.google.com&ctz=Europe%2FLondon) that doesn't seem to have a .ics extension.  I've removed the second calendar and now it seems to be rendering the calendar normally again.

I'm not quite sure what would have changed and caused it to fail as it was rendering fine up until Sunday.  I've not made any changes to either calendar, nor any other configs.

aceisace, 2020-05-19:
@clone545 
Yes, the provided url of your iCalendar has an `embed` tag, it's meant to be embedded on websites, so it doesn't contain the actual .ics file.

As for why the rendering was working fine up to some point, it might be because inkycal needs to be restarted before any changes take effect (e.g. via reboot). Simply modifiying the settings file does not mean the changes are applied immediately. It might be that your rpi restarted on Sunday, but that's also an hunch :)

Ribitsch, 2020-05-19:
if you want to use the google holiday calendar, change it to the following syntax (works for Austrian version)

"https://calendar.google.com/calendar/ical/de.austrian%23holiday%40group.v.calendar.google.com/public/basic.ics"

clone545, 2020-05-19:
I've swapped out the Google Calendar to just another ics calendar hosted elsewhere and it seems to be working fine now.  Thanks for the pointers and the help to get it going again :)

closed 2020-05-19 by clone545