aceinnolab/Inkycal #117: Feature: Horizontal mode

GaryBoone, 2020-05-26:
The Inky-Calendar code isn't usable for those of use with horizontally mounted displays. Adding a horizontal or landscape mode would allow different arrangements provide additional flexibility. 

Looking through past feature requests and updates, it appears that horizontal mode was a planned feature. What's the current status? Is it included in the new development branch? 

Looking at the code, I see that write_text has a rotation parameter. So at least the base code is already there. 

Has there been any planning for how to do both vertical and horizontal modes? For example, each widget could have (x, y, orientation) parameters. Or alternatively, the current vertical division scheme could be retained, but an orientation and number of columns could be added. Then the vertical layout would apply within a column.

Atrejoe, 2020-05-26:
Hi @GaryBoone,

Yes horizontal mode (or rotated mode) in likely to be included in v2 (see https://github.com/aceisace/Inky-Calendar/commit/17b0f610b05ce9442a25cf5f637b6c121574ab83#commitcomment-39368928, note: this only talks about 180 degrees rotation, but we are talking about multiple of 90 so far), which is nearing the beta stage.

We have not yet unambiguously defined the rules for rotated mode. You suggest that the division should switch from horizontal splits to vertical splits, which makes sense.
The direction we are currently going would make it possible to introduce a grid-system. See the discussion at: https://github.com/aceisace/Inky-Calendar/issues/94#issuecomment-626403219

`shameless-plus mode`
The is also a server-based panel generation solution (made by me), which has panel-rotation and section resizing. All features which will be included in v2 of the client-side solution.
You can give it a try: https://inkycal.robertsirre.nl/
`/shameless-plus mode`

aceisace, 2020-05-26:
@GaryBoone There has been a past attempt at supporting horizontal mode (9.7" ePaper display)
[image]
The problem with the horizontal mode however, is the need of re-allocating sections. As @Atrejoe has pointed out, it is already possible to set relative heights of each module in the /dev_ver2_0 branch. But supporting  horizontal mode without adding a lot of extra code would require a 2-coloumn grid instead of one (like now).

The `rotation` parameter in `write_text` is not required at all for the horizontal mode. Just adjusting the image size will do the job (-> `image_width, image_height = image_height, image_width`).

So in short, yes, this feature is still in planning, but there are a bunch of other things I have to work on first.

Atrejoe, 2020-05-26:
Off topic: I'm amazed by the anti-aliassing on this panel

aceisace, 2020-05-28:
@Atrejoe Yes, you are right. The rendering is amazing, but it lies partially with the fact that the 9.7" support several levels of grey, while the others only support pure-black and white.

aceisace, 2021-02-12:
As this is a feature request which will require re-writing parts of the software, I've added this feature request on the project page:
[Horizontal mode](https://github.com/aceisace/Inkycal/projects/3#card-54823296)
Once there has been some progress, it will be notified on Discord and the corresponding card will be moved at the correct place.

I'll be closing this issue then. If you feel a need to re-open this issue, please feel free to do so.

closed 2021-02-12 by aceisace